zephyrproject-rtos/zephyr #65528: flash_stm32_ospi driver falls back to default erase cmd because of wrong logic

mkunzecw, 2023-11-21:
**Describe the bug**

When ([flash_stm32_ospi_erase] (https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v3.5.0/drivers/flash/flash_stm32_ospi.c#L964) of  flash_stm32_ospi is called it always falls back to 4096byte erases even if the external flash supports larger erases and a better erase command would fit.

This are 2 related bugs:
A. In for loop the best solution is searched but one not aligned solution forces fall back to default size
https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v3.5.0/drivers/flash/flash_stm32_ospi.c#L1071

replacing line 1071 with } else if (bet == NULL) { fixes this problems

Logic explained:

1. It is looped over all available erase types
2. An erase type is chosen if it is aligned and no best erase type is found or the current is better
3. If the erase type is not aligned **or another one was found that is better, a fallback to default is done**

The fall back is wrong and should only be done if no best erase type was found prior, that's the else clause needs a check.

```
drivers/flash/flash_stm32_ospi.c#L1066
--  } else {
++ } else if (bet == NULL) {
```

B. related to #60904
only the remaining erase size is considered, same fix can be applied here

```
drivers/flash/flash_stm32_ospi.c#L1066
--     && SPI_NOR_IS_ALIGNED(size, etp->exp)
++   && (size >= BIT(etp->exp))
```

mkunzecw, 2023-11-22:
added more information, corrected solution

FRASTM, 2023-12-06:
@mkunzecw can you please check th PR https://github.com/zephyrproject-rtos/zephyr/pull/66229 and see if it match ?

FRASTM, 2023-12-14:
If you intend to erase  blocks of 64KB), the command to send has one of the following instruction : 
- SPI_NOR_CMD_BE_4B
- SPI_NOR_CMD_BE
- SPI_NOR_OCMD_BE

I guess this is another case, depending on the size to erase : to be evaluated in nb of blocks + nb of sectors.

closed 2024-01-02 by carlescufi