zephyrproject-rtos/zephyr #65559: subsys: modem: backends: uart: modem.backends.uart.isr and modem.backends.uart.async fail

katgiadla, 2023-11-22:
**Describe the bug**
The tests `modem.backends.uart.isr` and `modem.backends.uart.async` fail.

Observed for `nrf5340dk_nrf5340_cpuapp`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf5340dk_nrf5340_cpuapp` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/subsys/modem/backends/uart -p nrf5340dk_nrf5340_cpuapp --device-testing --device-serial /dev/ttyACM1 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.5.0-1863-g23324fc5bae4 ***
Running TESTSUITE modem_backend_uart_suite
===================================================================
START - test_transmit_receive
TX: 2,2
TX ACC: 2
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.1
- Commit SHA or Version used: zephyr-v3.5.0-1863-g23324fc5bae4

bjarki-trackunit, 2023-11-22:
Hi, did you bridge the UART TX and RX pins as described in the dts .overlay? it designates which pins need to be bridged :)

PerMac, 2023-11-22:
If a test requires some extra modification of a board, e.g. pins connected, using some shield etc. it has to use keyword "fixture" https://docs.zephyrproject.org/latest/develop/test/twister.html#fixtures so the test is auto-picked only for setups which supports it

bjarki-trackunit, 2023-11-22:
I will look into adding the fixture :) have not seen it before

PR #65668 (2023-11-27): tests: modem: backend: Add fixture to test suite
The tests in this testcase require physical modification of the boards to be performed. This is now documented using the `fixture: gpio_loopback` and `notes:` properties.

Fixes: #65559