zephyrproject-rtos/zephyr #65571: [Coverity CID: 333393] Out-of-bounds access in subsys/bluetooth/mesh/crypto_tc.c

zephyrbot, 2023-11-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/bluetooth/mesh/crypto_tc.c#L72

Category: Memory - corruptions
Function: `bt_mesh_aes_cmac_raw_key`
Component: Bluetooth
CID: [333393](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=333393)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/mesh/crypto_tc.c

```
56     int bt_mesh_aes_cmac_raw_key(const uint8_t key[16], struct bt_mesh_sg *sg, size_t sg_len,
57                            uint8_t mac[16])
58     {
59       struct tc_aes_key_sched_struct sched;
60       struct tc_cmac_struct state;
61     
>>>     CID 333393:    (OVERRUN)
>>>     Overrunning struct type tc_cmac_struct of 84 bytes by passing it to a function which accesses it at byte offset 87.
62       if (tc_cmac_setup(&state, key, &sched) == TC_CRYPTO_FAIL) {
63               return -EIO;
64       }
65     
66       for (; sg_len; sg_len--, sg++) {
67               if (tc_cmac_update(&state, sg->data, sg->len) == TC_CRYPTO_FAIL) {
66       for (; sg_len; sg_len--, sg++) {
67               if (tc_cmac_update(&state, sg->data, sg->len) == TC_CRYPTO_FAIL) {
68                       return -EIO;
69               }
70       }
71     
>>>     CID 333393:    (OVERRUN)
>>>     Overrunning struct type tc_cmac_struct of 84 bytes by passing it to a function which accesses it at byte offset 87.
72       if (tc_cmac_final(mac, &state) == TC_CRYPTO_FAIL) {
73               return -EIO;
74       }
75     
76       return 0;
77     }
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

PavelVPV, 2023-11-23:
I think this is the same issue as this one: https://github.com/zephyrproject-rtos/zephyr/issues/65580#issuecomment-1822762754

ceolin, 2024-01-31:
This is still open. I have changed the priority accordingly with Coverity analysis. Please, properly triage it in coverity if something needs to be changed.

PavelVPV, 2024-02-08:
Closing as this is a misconfiguration of Coverity, not the stack issue.

closed 2024-02-08 by PavelVPV