zephyrproject-rtos/zephyr #65578: [Coverity CID: 333376] Unsigned compared against 0 in subsys/bluetooth/shell/bt.c

zephyrbot, 2023-11-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/bluetooth/shell/bt.c

Category: Control flow issues
Function: `cmd_adv_data`
Component: Bluetooth
CID: [333376](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=333376)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/shell/bt.c#L1975

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_NO_EFFECT). ([CWE-570](http://cwe.mitre.org/data/definitions/570.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

jhedberg, 2023-11-22:
@Fanhang32 could you try to fix this? It looks like the code was added by you in commit 3fb434a6530897772e40922e780446b2f6ed4a24

Fanhang32, 2023-12-23:
> @Fanhang32 could you try to fix this? It looks like the code was added by you in commit [3fb434a](https://github.com/zephyrproject-rtos/zephyr/commit/3fb434a6530897772e40922e780446b2f6ed4a24)

@jhedberg I'm sorry for the late reply.

 It looks like the  type of `len`  should be `ssize_t`  as it return by `ad_init`

```
	len = ad_init(&data[*data_len], AD_SIZE - *data_len, adv_set_opt[selected_adv]);
	if (len < 0) {
		shell_error(sh, "Failed to initialize stack advertising data");

		return -ENOEXEC;
	}

```

closed 2023-12-27 by jhedberg