zephyrproject-rtos/zephyr #65580: [Coverity CID: 333374] Out-of-bounds access in subsys/bluetooth/crypto/bt_crypto.c

zephyrbot, 2023-11-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/bluetooth/crypto/bt_crypto.c

Category: Memory - corruptions
Function: `bt_crypto_aes_cmac`
Component: Bluetooth
CID: [333374](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=333374)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/crypto/bt_crypto.c

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

jhedberg, 2023-11-22:
@nashif how come some Coverity reports are able to give specific lines in files, however this one is referencing an entire c-file? (making it impossible to have even some initial idea of the issue without opening the report in Coverity)

jhedberg, 2023-11-22:
Here's the relevant part from the Coverity report:
```
 33
   	
CID 333374 (#2 of 2): Out-of-bounds access (OVERRUN)
6. overrun-buffer-val: Overrunning struct type tc_cmac_struct of 84 bytes by passing it to a function which accesses it at byte offset 87.
 34        if (tc_cmac_final(out, &state) == TC_CRYPTO_FAIL) {
 35                return -EIO;
 36        }
```

jhedberg, 2023-11-22:
This looks either like a false positive, or then a TinyCrypt bug. The API usage looks correct, i.e. the pointer type passed is as expected, and even exactly the same as used by Zephyr's own TinyCrypt test code (e.g. `tests/crypto/tinycrypt/src/cmac_mode.c`).

jhedberg, 2023-11-22:
Here's the struct definition:
```c
/* struct tc_cmac_struct represents the state of a CMAC computation */
typedef struct tc_cmac_struct {
/* initialization vector */
	uint8_t iv[TC_AES_BLOCK_SIZE];
/* used if message length is a multiple of block_size bytes */
	uint8_t K1[TC_AES_BLOCK_SIZE];
/* used if message length isn't a multiple block_size bytes */
	uint8_t K2[TC_AES_BLOCK_SIZE];
/* where to put bytes that didn't fill a block */
	uint8_t leftover[TC_AES_BLOCK_SIZE];
/* identifies the encryption key */
	unsigned int keyid;
/* next available leftover location */
	unsigned int leftover_offset;
/* AES key schedule */
	TCAesKeySched_t sched;
/* calls to tc_cmac_update left before re-key */
	uint64_t countdown;
} *TCCmacState_t;
```

@nashif could it be that for some reason Coverity sees one value for TC_AES_BLOCK_SIZE when looking at the TinyCrypt module code, and another (different) value when looking at Zephyr code, and because of that thinks that there is memory overflow happening?

closed 2024-01-22 by zephyrbot