zephyrproject-rtos/zephyr #65584: [Coverity CID: 333367] Out-of-bounds access in subsys/mgmt/mcumgr/transport/src/smp_udp.c

zephyrbot, 2023-11-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/mgmt/mcumgr/transport/src/smp_udp.c

Category: Memory - corruptions
Function: `create_thread`
Component: Other
CID: [333367](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=333367)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/mgmt/mcumgr/transport/src/smp_udp.c#L295

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

nordicjm, 2023-11-22:
@andyross this looks to be a bogus report, can you confirm?

nordicjm, 2023-12-12:
Ping @andyross and @peter-mitsis

andyross, 2023-12-12:
Sorry, missed the pings earlier.  I can't access the coverity report, but for sure I don't see anything particularly wrong with the line fingered.  Maybe something is passing a bad config pointer to that routine elsewhere?

nordicjm, 2023-12-12:
Should be able to login using SAML. This is what it shows:

> 293static void create_thread(struct config *conf, const char *name)
> 294{
>    CID 333367 (#1 of 1): Out-of-bounds access (OVERRUN)1. overrun-buffer-val: Overrunning array conf->stack of 576 bytes by passing it to a function which accesses it at byte offset 591. [[show details](https://scan9.scan.coverity.com/eventId=9520487-0&modelId=9520487-0&fileInstanceId=124550179&filePath=%2Ftwister-out%2Fnrf52840dk_nrf52840%2Ftests%2Fsubsys%2Fmgmt%2Fmcumgr%2Fall_options%2Fmgmt.mcumgr.all.options%2Fzephyr%2Finclude%2Fgenerated%2Fsyscalls%2Fkernel.h&fileStart=59&fileEnd=86)]
> 295        k_thread_create(&(conf->thread), conf->stack,
> 296                        K_KERNEL_STACK_SIZEOF(conf->stack),
> 297                        smp_udp_receive_thread, conf, NULL, NULL,
> 298                        CONFIG_MCUMGR_TRANSPORT_UDP_THREAD_PRIO, 0, K_FOREVER);

Which then eventually goes to this:

> 286static inline char *Z_KERNEL_STACK_BUFFER(k_thread_stack_t *sym)
> 287{
>    1. element_addr: Forming the address of the element at index 592U requires buffer (char *)sym to have at least 592 elements.
> 288        return (char *)sym + K_KERNEL_STACK_RESERVED;
> 289}

andyross, 2023-12-12:
Oh.  Then that seems like a bug!   It's a tiny toy stack and this arch+config environment presumably needs more for a default stack setup.   Do we know what config is being built?

nordicjm, 2023-12-13:
Since these are just using the normal thread macros, the bug is in those or does something need changing in this module's code?

nordicjm, 2023-12-18:
ping @andyross

andyross, 2024-01-02:
Saw the general bug come by in #67106 and realized there was a question here.  Looking back again, it seems like the proximate issue is that the thread creation in smp_udp.c is just too small.  CONFIG_MCUMGR_TRANSPORT_UDP_STACK_SIZE isn't big enough to hold a working stack with the expected setting for CONFIG_ARCH_KERNEL_STACK_RESERVED (exactly which configuration it's building isn't clear to me).  And that can result in a stack overflow at initialization time, or so coverity says.

So a fix for the proximate failure would be to just make that setting larger, or adjust stack size based on the reserved value, etc...

But the root problem here is that the kernel needs to detect this.  What should have happened is a runtime failure in k_thread_create() and not an overflow!

ceolin, 2024-01-31:
Priority changed based on Coverity analysis. Please feel free to properly triage it there.

PR #68432 (2024-02-04): kernel/thread: Detect in-kernel "reserved" stack overflow
Traditionally, k_thread_create() has required that the application size the stack correctly.  Zephyr doesn't detect or return errors and treats stack overflow as an application bug (though obviously some architectures have runtime features to trap on overflows).

At this one spot though, it's possible for the kernel to adjust the stack for K_THREAD_STACK_RESERVED in such a way that the arch layer's own stack initialization overflows.  That failure can be seen by static analysis, so we can't just sweep it under the rug as an application  [... cut]
closed 2024-02-04 by nashif