zephyrproject-rtos/zephyr #65589: [Coverity CID: 333357] Out-of-bounds access in subsys/bluetooth/mesh/crypto_tc.c

zephyrbot, 2023-11-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/bluetooth/mesh/crypto_tc.c

Category: Memory - corruptions
Function: `bt_mesh_sha256_hmac_raw_key`
Component: Bluetooth
CID: [333357](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=333357)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/mesh/crypto_tc.c#L104

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

PavelVPV, 2023-11-23:
I think this is the same issue as this one: https://github.com/zephyrproject-rtos/zephyr/issues/65580#issuecomment-1822762754

ceolin, 2024-01-31:
It was detected in the latest snapshot. Have changed the priority to match with Coverity.

PavelVPV, 2024-02-08:
Closing as this is a misconfiguration of Coverity, not the stack issue.

closed 2024-02-08 by PavelVPV