zephyrproject-rtos/zephyr #65598: [Coverity CID: 330594] Out-of-bounds access in drivers/bluetooth/hci/userchan.c

zephyrbot, 2023-11-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/drivers/bluetooth/hci/userchan.c

Category: Memory - corruptions
Function: `uc_open`
Component: Bluetooth
CID: [330594](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=330594)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/bluetooth/hci/userchan.c#L351

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

jhedberg, 2023-11-22:
Relevant part from Coverity:
```
CID 330594 (#1 of 1): Out-of-bounds access (OVERRUN)
4. overrun-buffer-val: Overrunning array rx_thread_stack of 24 bytes by passing it to a function which accesses it at byte offset 591. [[show details](https://scan9.scan.coverity.com/eventId=9495708-5&modelId=9495708-0&fileInstanceId=124160115&filePath=%2Ftwister-out%2Fnative_posix%2Fsamples%2Fbluetooth%2Fhap_ha%2Fsample.bluetooth.hap_ha.monaural_no_presets%2Fzephyr%2Finclude%2Fgenerated%2Fsyscalls%2Fkernel.h&fileStart=59&fileEnd=86)]
351        k_thread_create(&rx_thread_data, rx_thread_stack,
352                        K_KERNEL_STACK_SIZEOF(rx_thread_stack),
353                        rx_thread, NULL, NULL, NULL,
354                        K_PRIO_COOP(CONFIG_BT_DRIVER_RX_HIGH_PRIO),
355                        0, K_NO_WAIT);
356
```

@nashif this is the second report with seemingly correct `k_thread_create()` usage. Seems like a false positive (I'd expect there to be quite many of these in that case, from the latest Coverity run).

nashif, 2023-11-22:
> @nashif this is the second report with seemingly correct `k_thread_create()` usage. Seems like a false positive (I'd expect there to be quite many of these in that case, from the latest Coverity run).

do not know, do not shoot the messanger please :-)
@ceolin can you please take a look at this?

ceolin, 2023-12-11:
My impression is that Coverity is mixing different builds during the analysis. Looking the code the only way this problem happens is if we have `K_KERNEL_STACK_RESERVED` defined differently.

I am checking if it is possible to this happen depending on how headers are included.

henrikbrixandersen, 2024-02-05:
Should be fixed by #68432.

closed 2024-02-05 by henrikbrixandersen