zephyrproject-rtos/zephyr #65640: BLE Audio not heard from nRF5340 Audio DK using Zephyr Shell application.

ravitest12, 2023-11-23:
I have flashed the Zephyr shell Test application on nRF5340 Audio DK. Gave the below commands in the shell of nRF5340.

bt init
tmap init
bap init
cap_acceptor init
mcc init
vcp_vol_rend init

bt advertise on 

Pair and connect with nRF5340 from Pixel 7 Phone. BLE Audio connected successfully.
Now i played a youtube video on the Pixel 7 Phone, BLE Audio is received on nRF5340 as shown in the shell below, but audio is not heard from the headphone aux jack of nRF5340. Am i missing anything here ?

uart:~$ bt init
Bluetooth initialized
Settings Loaded
[00:00:29.657,531] <inf> fs_nvs: 6 Sectors of 4096 bytes
[00:00:29.657,531] <inf> fs_nvs: alloc wra: 2, dc0
[00:00:29.657,562] <inf> fs_nvs: data wra: 2, 53c
[00:00:29.671,691] <wrn> bt_hci_core: Controller to host flow control not supported
[00:00:29.676,055] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:29.692,840] <inf> bt_hci_core: Identity: D0:28:B4:E4:DA:37 (random)
[00:00:29.692,871] <inf> bt_hci_core: HCI: version 5.2 (0x0b) revision 0x0d41, manufacturer 0x07e8
[00:00:29.692,901] <inf> bt_hci_core: LMP: version 5.2 (0x0b) subver 0x0d41
uart:~$ tmap init
Registering TMAS with role: 63
uart:~$ bap init
[00:00:41.761,810] <dbg> bt_pacs: bt_pacs_cap_register: cap 0x200012cc dir sink codec_cap id 0x06 codec_cap cid 0x0000 codec_cap vid 0x0000
[00:00:41.761,932] <dbg> bt_pacs: bt_pacs_cap_register: cap 0x200012c4 dir source codec_cap id 0x06 codec_cap cid 0x0000 codec_cap vid 0x0000
uart:~$ cap_acceptor init
uart:~$ mcc init
[00:00:51.069,763] <dbg> bt_mcc: bt_mcc_init: Object selected callback: 0x4aa51
[00:00:51.069,793] <dbg> bt_mcc: bt_mcc_init: Object content callback: 0x4aab9
[00:00:51.069,793] <dbg> bt_mcc: bt_mcc_init: Object metadata callback: 0x4aea9
uart:~$ vcp_vol_rend init
[00:00:58.225,555] <dbg> bt_vcp_vol_rend: prepare_aics_inst: AICS P 0x200018c4
uart:~$ bt advertise on
Advertising started
Connected: 79:70:C5:19:52:36 (random)
[0]: Connected to 79:70:C5:19:52:36 (random)
Member[0] connected
LE data len updated: TX (len: 251 time: 2120) RX (len: 27 time: 328)
[00:01:12.931,945] <dbg> bt_has: connected: conn 0x20009448 err 0
Security changed: 79:70:C5:19:52:36 (random) level 2
Identity resolved 79:70:C5:19:52:36 (random) -> E8:D5:2B:02:89:56 (public)
Bonded with E8:D5:2B:02:89:56 (public)
[00:01:15.677,947] <dbg> bt_pacs: pacs_security_changed: 79:70:C5:19:52:36 (random) changed security level to 2
[00:01:15.685,668] <dbg> bt_has: security_changed: conn 0x20009448 level 2 err 0
[00:01:15.847,045] <dbg> bt_pacs: pacs_auth_pairing_complete: E8:D5:2B:02:89:56 (public) paired (bonded)
[00:01:15.847,198] <dbg> bt_csip_set_member: auth_pairing_complete: E8:D5:2B:02:89:56 (public) paired (bonded)
[00:01:15.857,543] <wrn> bt_gatt: MTU Exchange failed (err -120)
[00:01:15.867,828] <wrn> bt_gatt: MTU Exchange failed (err -120)
[00:01:15.878,082] <wrn> bt_gatt: MTU Exchange failed (err -120)
[00:01:15.878,143] <dbg> bt_pacs: notify_cb:
[00:01:15.878,173] <dbg> bt_pacs: notify_cb:
[00:01:16.516,174] <dbg> bt_csip_set_member: set_sirk_cfg_changed: value 0x0001
Client E8:D5:2B:02:89:56 (public) requested to read the sirk. Responding with Accept
[00:01:16.560,058] <dbg> bt_vcp_vol_rend: volume_state_cfg_changed: value 0x0001
[00:01:16.605,133] <dbg> bt_csip_set_member: set_size_cfg_changed: value 0x0001
[00:01:16.650,085] <dbg> bt_vocs: offset_state_cfg_changed: value 0x0001
[00:01:16.651,397] <dbg> bt_pacs: snk_cfg_changed: attr 0x95390 value 0x0001
[00:01:16.695,220] <dbg> bt_csip_set_member: set_lock_cfg_changed: value 0x0001
[00:01:16.698,608] <dbg> bt_vocs: read_offset_state: offset 0, counter 0
[00:01:16.740,753] <dbg> bt_pacs: snk_read: conn 0x20009448 attr 0x9537c buf 0x200254dc len 63 offset 0
[00:01:16.750,152] <dbg> bt_csip_set_member: read_set_sirk: Set sirk not encrypted
[00:01:16.750,183] <dbg> bt_csip_set_member: read_set_sirk: Set SIRK
                                             cd cc 72 dd 86 8c cd ce  22 fd a1 21 09 7d 7d 45 |..r..... "..!.}}E
[00:01:16.785,583] <dbg> bt_vcp_vol_rend: read_vol_state: Volume 100, mute 0, counter 0
LE PHY updated: TX PHY LE 2M, RX PHY LE 2M
[... 101 lines cut ...]
[00:01:32.655,609] <dbg> bt_ascs: ascs_iso_accept: iso_chan 0x20017368
[00:01:32.791,656] <dbg> bt_ascs: ascs_ep_iso_connected: stream 0x2000c230 ep 0x2000fd0c dir sink
[00:01:32.791,748] <dbg> bt_ascs: ase_status_changed: ase 0x2000fd08 id 0x01 enabling -> streaming
[00:01:32.791,809] <dbg> bt_ascs: ascs_ep_get_status: ep 0x2000fd0c id 0x01 state streaming
[00:01:32.791,809] <dbg> bt_ascs: ascs_ep_get_status_enable: dir sink cig 0x01 cis 0x00
Stream 0x2000c230 started
[100]: Incoming audio on stream 0x2000c230 len 120 ts 93607338 seq_num 100 flags 9 (dup ts 50; dup psn 0, err_pkts 0, lost_pkts 0)
[200]: Incoming audio on stream 0x2000c230 len 120 ts 94607326 seq_num 200 flags 9 (dup ts 100; dup psn 0, err_pkts 0, lost_pkts 0)
[300]: Incoming audio on stream 0x2000c230 len 120 ts 95607315 seq_num 300 flags 9 (dup ts 150; dup psn 0, err_pkts 0, lost_pkts 0)
[400]: Incoming audio on stream 0x2000c230 len 120 ts 96627293 seq_num 401 flags 9 (dup ts 200; dup psn 0, err_pkts 0, lost_pkts 0)
[500]: Incoming audio on stream 0x2000c230 len 120 ts 97627272 seq_num 501 flags 9 (dup ts 250; dup psn 0, err_pkts 0, lost_pkts 0)
[600]: Incoming audio on stream 0x2000c230 len 120 ts 98627265 seq_num 601 flags 9 (dup ts 300; dup psn 0, err_pkts 0, lost_pkts 0)
[700]: Incoming audio on stream 0x2000c230 len 120 ts 99627247 seq_num 701 flags 9 (dup ts 350; dup psn 0, err_pkts 0, lost_pkts 0)
[800]: Incoming audio on stream 0x2000c230 len 120 ts 100627238 seq_num 801 flags 9 (dup ts 400; dup psn 0, err_pkts 0, lost_pkts 0)
[900]: Incoming audio on stream 0x2000c230 len 120 ts 101627220 seq_num 901 flags 9 (dup ts 450; dup psn 0, err_pkts 0, lost_pkts 0)
[1000]: Incoming audio on stream 0x2000c230 len 120 ts 102627205 seq_num 1001 flags 9 (dup ts 500; dup psn 0, err_pkts 0, lost_pkts 0)
[1100]: Incoming audio on stream 0x2000c230 len 120 ts 103647196 seq_num 1101 flags 9 (dup ts 549; dup psn 0, err_pkts 0, lost_pkts 0)
[1200]: Incoming audio on stream 0x2000c230 len 120 ts 104647182 seq_num 1201 flags 9 (dup ts 599; dup psn 0, err_pkts 0, lost_pkts 0)
[1300]: Incoming audio on stream 0x2000c230 len 120 ts 105647171 seq_num 1301 flags 9 (dup ts 649; dup psn 0, err_pkts 0, lost_pkts 0)
[1400]: Incoming audio on stream 0x2000c230 len 120 ts 106647164 seq_num 1401 flags 9 (dup ts 699; dup psn 0, err_pkts 0, lost_pkts 0)
[1500]: Incoming audio on stream 0x2000c230 len 120 ts 107647161 seq_num 1501 flags 9 (dup ts 749; dup psn 0, err_pkts 0, lost_pkts 0)
[1600]: Incoming audio on stream 0x2000c230 len 120 ts 108647161 seq_num 1601 flags 9 (dup ts 799; dup psn 0, err_pkts 0, lost_pkts 0)
[1700]: Incoming audio on stream 0x2000c230 len 120 ts 109647167 seq_num 1701 flags 9 (dup ts 849; dup psn 0, err_pkts 0, lost_pkts 0)
[1800]: Incoming audio on stream 0x2000c230 len 120 ts 110647166 seq_num 1801 flags 9 (dup ts 899; dup psn 0, err_pkts 0, lost_pkts 0)
[1900]: Incoming audio on stream 0x2000c230 len 120 ts 111647171 seq_num 1901 flags 9 (dup ts 949; dup psn 0, err_pkts 0, lost_pkts 0)

Thalley, 2023-11-23:
@ravitest12 Please use the bug template when reporting bugs. I currently don't see what the bug you are reporting is. 

Based on the title, it sounds like you are expecting audio output (via USB or AUX?), which we simply don't have. So if that is the case, then it's not a bug, but simply a missing feature.

ravitest12, 2023-11-27:
@Thalley Thank you for updating the issue. Yes i am expecting the audio output via AUX on the nRF5340. Is there any solution or work around to fix this ?

Thalley, 2023-11-27:
> @Thalley Thank you for updating the issue. Yes i am expecting the audio output via AUX on the nRF5340. Is there any solution or work around to fix this ?

Nope, and one planned unfortunately. The driver for the DAC on the nRF5340 doesn't have (to my knowledge) an open source driver that we can use with Zephyr. @larsgk have been working towards this for a while, but not sure what the state is right now. 

It is, however, possible to add USB audio out if that is good enough for your case?

ravitest12, 2023-11-27:
@larsgk Thanks for your support. I want the USB Audio out to work using the Shell application. what config should be enabled for this in prj.conf file ? I can see there is one type C USB port on nRF5340 board. should i connect a type C to Aux adapter to it ? how does the board get power ?

larsgk, 2023-11-27:
> @larsgk Thanks for your support. I want the USB Audio out to work using the Shell application. what config should be enabled for this in prj.conf file ? I can see there is one type C USB port on nRF5340 board. should i connect a type C to Aux adapter to it ? how does the board get power ?

Ok, a few things:

1) the shell app doesn't support USB audio in/out, but maybe something to consider in the future, @Thalley ?

2) for now, I only know of the broadcast samples (mentioned above - one merged, one in PR) adding actual USB audio support (besides the echo sample for the USB subsys).

3) the board is powered using the USB-C port (e.g. connected to a PC), but the same port exposes the JLink/SEGGER flashing/logging interface AND whatever you expose from the application core (in the samples mentioned, it's a USB 'headset').

For the shell to support USB audio, we would need to either make a specific implementation for e.g. broadcast/unicast bap or maybe something more generic using USB audio (subsys) as one audio data path (of potentially multiple).

Thalley, 2023-11-28:
Changed to enhancement as it requests additional functionality in the shell, rather than it being a bug.

Thalley, 2024-03-04:
The linked pull requests will add USB stereo support for the shell, but it won't add AUX support (drivers are missing for that). 

If this is not good enough, please add a response here

larsgk, 2024-04-29:
@carlescufi @Thalley  - let's not close this just yet.  We should add the cirrus logic drivers and config files (that is now possible, license wise) to get AUX/Jack support.

Otherwise we should make a new issue specific to getting aux support in nRF5340 Audio DK