zephyrproject-rtos/zephyr #65652: nrf `sys_clock_disable` faults in secure code

JordanYates, 2023-11-23:
**Describe the bug**

The `sys_clock_disable` function added in #56528 is faulting when called in a secure context from a non-secure application.

`CONFIG_SYSTEM_TIMER_HAS_DISABLE_SUPPORT` is enabled purely based on the hardware block, but the implementation does not recognise that the `NRF_RTC1` peripheral has been assigned to the non-secure domain.

When `sys_clock_disable` is eventually called, it predictably triggers a data bus fault:
```
[00:02:15.458,435] <err> os: Exception occurred in Secure State
[00:02:15.484,008] <err> os: ***** HARD FAULT *****
[00:02:15.504,455] <err> os:   Fault escalation (see below)
[00:02:15.530,029] <err> os: ***** BUS FAULT *****
[00:02:15.550,506] <err> os:   Precise data bus error
[00:02:15.576,019] <err> os:   BFAR Address: 0x50015004
```
Where `0x50015000` is the secure address of `NRF_RTC1` and `0x50015004` is the address of the `NRF_RTC_TASK_STOP` task for that instance.

**Expected behavior**

Rebooting from a secure application should not unconditionally trigger a fault.

**Environment (please complete the following information):**

 - Zephyr v3.5

**Additional context**

The secure application is one I have written, which runs the following logic to assign all peripherals to the non-secure domain before jumping to the nonsecure application:
```
	/* Configure all peripherals for non-secure access */
	for (int i = 0; i < ARRAY_SIZE(NRF_SPU->PERIPHID); i++) {
		NVIC_DisableIRQ(i);
		NRF_SPU->PERIPHID[i].PERM = 0;
		NVIC_SetTargetState(i);
	}
```

anangl, 2023-11-29:
Is this actually a bug in the system timer? You are configuring as non-secure a peripheral that your secure application is configured to use (RTC1). I think you should rather use `CONFIG_SYS_CLOCK_EXISTS=n` in your application.

JordanYates, 2023-11-29:
That's a fair point, to be honest I wasn't even aware you could disable the system timer in an application.
The one thing it does break is any timeouts that may be present before the jump to the NS application.

closed 2024-04-13 by github-actions[bot]