zephyrproject-rtos/zephyr #65674: boards: rzt2m_starterkit: Issue with serial output

joeker64, 2023-11-23:
**Describe the bug**
I am currently trying to get serial output from the "hello-world" application with the 'rzt2m_starterkit' board but I only seem to get junk from the USB serial connection (CN16). I have the board-rate set to "115200" which is correct according to this:

https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/arm/renesas/rzt2m.dtsi#L95

I have also used a logic analyser to try and debug this issue attaching a wire to the resistor for the IC39 line driver (can be found on schematic) on the back of the board which carries the signal "UART_USB_TX". From the signal I get the estimated board-rate of "152439". If the board-rate is set to this value you will still get some junk but it will show the "hello-world". This can be seen with the attached image below.

I'm unsure what's going wrong it could be that the some clocks are set incorrectly and are creating a non-standard board-rate in which the chip cannot read.  

[image]

**Expected behavior**
Working USB serial console  

**Environment**
 - OS: MacOS (have also tried on Linux)
 - Toolchain: Zephyr SDK 0.16.3

tgorochowik, 2023-11-24:
Hi, thank you for the report. We just did a quick test and it seems to work perfectly for us. We'll double check that Zephyr explicitly sets all the relevant clocks to make sure if it's not a matter of different defaults or something like that.

joeker64, 2023-11-28:
Have you updated any firmware on the board? I'm currently using it fresh out of the box?

jbylicki, 2023-12-06:
Hi, could you provide your DIP switch settings? Specifically for `SW3`, `SW4`, `SW5`, and `SW6`? Additionally, if possible, could you attach the problematic `zephyr.elf` file for analysis?

joeker64, 2023-12-07:
SW3:
| 1   | 2   | 3   | 4   | 5   | 6   | 7   | 8   |
|-----|-----|-----|-----|-----|-----|-----|-----|
| off | off | off | off | off | off | off | off |

SW4:
| 1  | 2  | 3  | 4  | 5   | 6   | 7   | 8   |
|----|----|----|----|-----|-----|-----|-----|
| on | on | on | on | off | off | off | off |

SW5:
| 1   | 2   | 3  | 4   | 5  | 6   | 7   | 8   | 9  | 0   |
|-----|-----|----|-----|----|-----|-----|-----|----|-----|
| off | off | on | off | on | off | off | off | on | off |

SW6:
| 1   | 2   | 3  | 4   | 5   | 6  | 7   | 8  | 9   | 0  |
|-----|-----|----|-----|-----|----|-----|----|-----|----|
| off | off | on | off | off | on | off | on | off | on |

Attached zephyr.elf:
[zephyr.elf.tar.gz](https://github.com/zephyrproject-rtos/zephyr/files/13598983/zephyr.elf.tar.gz)

jbylicki, 2023-12-28:
The issue was that Zephyr never actually set the defaults for clock control registers. I suspect either your default environment / bootloader or some example you've loaded previously on that `debugserver` session was messing up the clock control registers. I have made a PR linked to this issue containing the fix.

PR #67045 (2024-03-04): soc/arm/renesas_rzt2m: set default System Clock Control register values
This PR implements the verbose setting of default clock control register values for the RZ/T2M board. Previously, those registers were not set and if on the same `debugserver` session one has launched another binary and said registers were modified, it wouldn't re-adjust later in Zephyr.

Fixes #65674
closed 2024-03-04 by henrikbrixandersen