zephyrproject-rtos/zephyr #65711: drivers: adc: Add driver for MCP3914 8ch AFE on SPI

dean-at-evos, 2023-11-23:
We are developing a product using this analogue front end and could contribute the driver back to zephyr. 

There is a draft PR with the current state of the driver to get the conversation started around minimal functionality and features needed to merge in to the zephyr tree. It is functional in async mode with interrupt driven sampling. Only the basic features of the chip are implemented at this stage, with some more to come.

If there are refactors or a different/better approach we can get these in early.

#65646 

Thanks for a great product!

henrikbrixandersen, 2023-11-24:
@dean-at-evos Great! Since you've already opened a PR, there's really no need for this issue. Feel free to close it.
Just a heads-up: As long as your PR is in draft state, it likely wont receive any reviews unless explicitly requested.

dean-at-evos, 2023-11-26:
Thanks, so the protocol would be to assign someone from maintainers as a draft, or just submit as a pr and let the bot do it?

henrikbrixandersen, 2023-11-26:
> Thanks, so the protocol would be to assign someone from maintainers as a draft, or just submit as a pr and let the bot do it?

Typically, you'd just move the PR out of draft state once ready and let the bot do the assignments.

closed 2023-11-27 by dean-at-evos