zephyrproject-rtos/zephyr #65722: There should be a way to declare a minimum value for CONFIG_HEAP_MEM_POOL_SIZE

jhedberg, 2023-11-24:
With the recent introduction of the ACPICA library into Zephyr, the minimum required HEAP_MEM_POOL_SIZE (used by `k_malloc()`) can be up to several megabytes on certain platforms. This causes issues when samples or test apps explicitly set their own value for this Kconfig option:
```shell
❯ git grep CONFIG_HEAP_MEM_POOL_SIZE= tests samples|wc -l
     155
```

Almost all of these are too small for some ACPI-using platforms, resulting in runtime failures.

Additionally, there are lots of boards & SoCs that set conditional default values for the Kconfig option:
```
boards/x86/intel_rpl/Kconfig.defconfig:config HEAP_MEM_POOL_SIZE
boards/x86/intel_rpl/Kconfig.defconfig- default 64000000
boards/x86/intel_rpl/Kconfig.defconfig-config MAIN_STACK_SIZE
boards/x86/intel_rpl/Kconfig.defconfig- default 320000
--
boards/x86/qemu_x86/Kconfig.defconfig:config HEAP_MEM_POOL_SIZE
boards/x86/qemu_x86/Kconfig.defconfig-  default 1048576 if ACPI
boards/x86/qemu_x86/Kconfig.defconfig-  depends on KERNEL_MEM_POOL
boards/x86/qemu_x86/Kconfig.defconfig-
--
boards/x86/qemu_x86/Kconfig.defconfig:config HEAP_MEM_POOL_SIZE
boards/x86/qemu_x86/Kconfig.defconfig-  default 1048576 if ACPI
boards/x86/qemu_x86/Kconfig.defconfig-  depends on KERNEL_MEM_POOL
boards/x86/qemu_x86/Kconfig.defconfig-
--
boards/x86/up_squared/Kconfig.defconfig:config HEAP_MEM_POOL_SIZE
boards/x86/up_squared/Kconfig.defconfig-        default 1048576 if ACPI
boards/x86/up_squared/Kconfig.defconfig-        depends on KERNEL_MEM_POOL
boards/x86/up_squared/Kconfig.defconfig-
--
boards/xtensa/esp32_devkitc_wroom/Kconfig.defconfig:config HEAP_MEM_POOL_SIZE
boards/xtensa/esp32_devkitc_wroom/Kconfig.defconfig-    default 98304 if WIFI
boards/xtensa/esp32_devkitc_wroom/Kconfig.defconfig-    default 40960 if BT
boards/xtensa/esp32_devkitc_wroom/Kconfig.defconfig-    default 4096
```

What all of these applications and board configurations really want, is to declare a *minimum* value for the heap size, rather than a default value - setting a bigger value is ok, but a smaller value will likely break a board, feature or application.

One potential solution could be to add a new `HEAP_MEM_POOL_MIN_SIZE` option that all the boards would set a default to. This would then be combined with a `range` for the original option, as well as setting its default value to the new minimum option value:
```
config HEAP_MEM_POOL_MIN_SIZE
	int
	default 0

config HEAP_MEM_POOL_SIZE
	int "Heap memory pool size (in bytes)"
	range HEAP_MEM_POOL_MIN_SIZE 104857600 # upper limit is just a "big" value, 100MB in this case
	default HEAP_MEM_POOL_MIN_SIZE
```

jhedberg, 2023-11-24:
@tbursztyka @finikorg @nashif @dcpleung @andyross would appreciate your thoughts on how to best solve this issue. Even better if each board/feature/subsystem/sample could "vote" on what they need as the minimum value and then the actual minimum would be set as the maximum of the voted values, but I don't think Zephyr has any existing solution for this kind of a mechanism.

jhedberg, 2023-11-24:
This issue was brought up to me by @nashif pointing out that `tests/kernel/threads/thread_stack/` fails to run on `up_squared`. It's not a surprise, since the sample has `CONFIG_HEAP_MEM_POOL_SIZE=192` in its `prj.conf`, while `up_squared` needs roughly 1MB for ACPI to initialize and work correctly. The test command I used was the following, and the same failure can probably be observed on other ACPI-using boards as well (even qemu_x86_64 has a requirement of several hundred kilobytes):

```
west build -p -b up_squared -T tests/kernel/threads/thread_stack/kernel.threads.thread_stack
```

jhedberg, 2023-11-24:
I just tried my proposed solution, but the Kconfig warning actually ends up failing the build, so it seems we can't use this or then all apps will need to change to some other mechanism  (maybe Kconfig overlay) to set their desired minimum:
```
warning: HEAP_MEM_POOL_SIZE (defined at kernel/Kconfig:841) was assigned the value '192' but got the
value '1048576'. See http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_HEAP_MEM_POOL_SIZE
and/or look up HEAP_MEM_POOL_SIZE in the menuconfig/guiconfig interface. The Application Development
Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
might be helpful too.

warning: user value 192 on the int symbol HEAP_MEM_POOL_SIZE (defined at kernel/Kconfig:841) ignored due to being outside the active range ([1048576, 1048576000]) -- falling back on defaults
Parsing /Users/jhedberg/src/zephyr/zephyr/Kconfig
Loaded configuration '/Users/jhedberg/src/zephyr/zephyr/boards/x86/up_squared/up_squared_defconfig'
Merged configuration '/Users/jhedberg/src/zephyr/zephyr/tests/kernel/threads/thread_stack/prj.conf'

error: Aborting due to Kconfig warnings
```

jhedberg, 2023-11-24:
Another solution, though not a general one, would be for ACPI to stop using the system heap and define its own instead. It kind of defeats the purpose of having shared system heap (`k_malloc()`) at all though.

finikorg, 2023-11-24:
Something like this?
```
diff --git a/tests/kernel/threads/thread_stack/Kconfig b/tests/kernel/threads/thread_stack/Kconfig
new file mode 100644
index 0000000000..0fcf4f27af
--- /dev/null
+++ b/tests/kernel/threads/thread_stack/Kconfig
@@ -0,0 +1,8 @@
+if !X86_64
+
+config HEAP_MEM_POOL_SIZE
+	int
+	default 192
+endif
+
+source "Kconfig.zephyr"
diff --git a/tests/kernel/threads/thread_stack/prj.conf b/tests/kernel/threads/thread_stack/prj.conf
index b6b5951635..f289eb1954 100644
--- a/tests/kernel/threads/thread_stack/prj.conf
+++ b/tests/kernel/threads/thread_stack/prj.conf
@@ -3,4 +3,3 @@ CONFIG_INIT_STACKS=y
 CONFIG_THREAD_STACK_INFO=y
 CONFIG_APPLICATION_DEFINED_SYSCALL=y
 CONFIG_TEST_USERSPACE=y
-CONFIG_HEAP_MEM_POOL_SIZE=192
```

jhedberg, 2023-11-24:
> Something like this?

Could work, except that the dependency should be `!ACPI` instead of `!X86_64`. Two potential concerns however:
 - There are some 155 apps (under `tests/` and `samples/`) that explicitly set `HEAP_MEM_POOL_SIZE`. So lots of places to update.
 - It's not a general solution. When the next subsystem comes along that has special requirements from the system heap size, we'll have the same issue again.

jhedberg, 2023-11-25:
@finikorg this works, but is quite cumbersome if it needs to be done to every single app:
```patch
--- /dev/null
+++ b/tests/kernel/threads/thread_stack/Kconfig
@@ -0,0 +1,6 @@
+if !ACPI
+config HEAP_MEM_POOL_SIZE
+       default 192
+endif # !ACPI
+
+source "Kconfig.zephyr"
diff --git a/tests/kernel/threads/thread_stack/prj.conf b/tests/kernel/threads/thread_stack/prj.conf
index b6b5951635..f289eb1954 100644
--- a/tests/kernel/threads/thread_stack/prj.conf
+++ b/tests/kernel/threads/thread_stack/prj.conf
@@ -3,4 +3,3 @@ CONFIG_INIT_STACKS=y
 CONFIG_THREAD_STACK_INFO=y
 CONFIG_APPLICATION_DEFINED_SYSCALL=y
 CONFIG_TEST_USERSPACE=y
-CONFIG_HEAP_MEM_POOL_SIZE=192
```

dcpleung, 2023-11-27:
I think having a minimum size kconfig is better. It would allow anyone using the board to automatically get the minimum size. Or else they would need to specify the size in their app.

jhedberg, 2023-11-28:
> I think having a minimum size kconfig is better. It would allow anyone using the board to automatically get the minimum size. Or else they would need to specify the size in their app.

Sure, I'm also leaning toward a new `HEAP_MEM_POOL_MIN_SIZE` Kconfig option, however I'm thinking that in the first phase I'd deal with this in the heap definition rather than e.g. setting a Kconfig `range`, so that the code just rounds up to the higher value of 'MEM_POOL_SIZE' and 'MEM_POOL_MIN_SIZE'. This way we don't have to touch the 155 apps that'd start generating build errors due to trying to set a too small value for some boards.

closed 2023-12-20 by carlescufi