zephyrproject-rtos/zephyr #65735: CI: tests/subsys/logging/log_output/logging.output.thread_id fails on qemu_cortex_m0

henrikbrixandersen, 2023-11-24:
**Describe the bug**
The `tests/subsys/logging/log_output/logging.output.thread_id` test suite fails on `qemu_cortex_m0` in CI on `main`. Git bisect points to commit 7a5fcb8c60e2e6ba26704fabc28fe2d4e0106513.

**To Reproduce**
Steps to reproduce the behavior:
1. `west twister -p qemu_cortex_m0 -s tests/subsys/logging/log_output/logging.output.thread_id`
2. See error

**Expected behavior**
Test case passes.

**Impact**
CI failure on `main`.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.5.0-1847-g7a5fcb8c60e2 ***
Running TESTSUITE test_log_output
===================================================================
START - test_colors
 PASS - test_colors in 0.002 seconds
===================================================================
START - test_format_ts
[00:00:01.000,000] domain/src: test
 PASS - test_format_ts in 0.001 seconds
===================================================================
START - test_level_flag
 PASS - test_level_flag in 0.001 seconds
===================================================================
START - test_levels
 PASS - test_levels in 0.001 seconds
===================================================================
START - test_no_flags
 PASS - test_no_flags in 0.001 seconds
===================================================================
START - test_no_flags_dname
 PASS - test_no_flags_dname in 0.001 seconds
===================================================================
START - test_raw
 PASS - test_raw in 0.001 seconds
===================================================================
START - test_thread_id
<err> [test_thread_id] src: Test
 PASS - test_thread_id in 0.001 seconds
===================================================================
START - test_ts_flag
 PASS - test_ts_flag in 0.001 seconds
===================================================================
START - test_ts_to_us
 PASS - test_ts_to_us in 0.001 seconds
===================================================================
TESTSUITE test_log_output succeeded
[00:00:00.447,265] <err> [irq] os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.458,679] <err> [irq] os: Current thread: 0x200000f0 (logging)
--- 5 messages dropped ---
[00:00:00.602,203] <err> [irq] os: Halting system
```

**Environment (please complete the following information):**
 - Commit SHA: 10a167f6c1f1ab4850f3bf1f8007826bff6f5285

henrikbrixandersen, 2023-11-28:
@keith-packard Any idea what is going on here?

keith-packard, 2023-11-28:
Yup, you're getting the 'large' version of printf which is causing something to overflow. Updating to SDK 0.16.4 "solves" the problem. It's overflowing the ztest stack. Fixed in #65893

PR #65893 (2023-11-28): tests/log_output: Avoid printf stack overflow with old picolibc
Before picolibc version 1.8.5, the only version of printf available with long long output support was the version including full support for floating point numbers and positional parameters. That version uses more stack space than the limited version causing an overflow.

Increase the ztest stack space by 128 bytes.

Closes: #65735
closed 2023-11-28 by nashif