zephyrproject-rtos/zephyr #65744: ESP32C3: BLE Peripheral advertisement fails to start when Central already scanning [zephyr-v3.5.0]

abdulayubzoi, 2023-11-24:
**Describe the bug**
It seems that on Espressif `esp32c3` with Zephyr `v3.5.0`, starting BLE peripheral advertisement via `bt_le_adv_start()` fails with error code `-EIO`, if the BLE central is already scanning. This was working fine with Zephyr `v3.4.0` for my BLE application. 

**To Reproduce**
The bug can be reproduced by slightly modifying the [Scan & Advertise](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/scan_adv) sample as follows:

- Add advertisement params
```
static struct bt_le_adv_param adv_param = BT_LE_ADV_PARAM_INIT(BT_LE_ADV_OPT_CONNECTABLE, 
                                                               BT_GAP_ADV_FAST_INT_MIN_2, 
                                                               BT_GAP_ADV_FAST_INT_MAX_2, 
                                                               NULL);
```

- Use `adv_param` to start advertisement
```
err = bt_le_adv_start(&adv_param, ad, ARRAY_SIZE(ad), NULL, 0);
```
- Build the application for `esp32c3_devkitm` and flash it to the board.
- Observe the debug logs.

**Expected behavior**
No error on starting advertisement while the client is scanning.

**Impact**
Showstopper, since I'm unable to run my application properly.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.5.0 ***
Starting Scanner/Advertiser Demo
[00:00:00.000,000] <inf> esp32_bt_adapter: BT controller compile version [d913766]
[00:00:00.162,000] <inf> bt_hci_core: Identity: FF:2C:50:7E:A3:98 (random)
[00:00:00.162,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x0060
[00:00:00.162,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
Bluetooth initialized
[00:00:00.566,000] <wrn> bt_hci_core: opcode 0x2005 status 0x0c
Advertising failed to start (err -5)
```

**Environment:**
- Devcontainer based on image: zephyrprojectrtos/zephyr-build:v0.26.6
- Toolchain: Zephyr SDK 0.16.4
- Zephyr branch: zephyr-v3.5.0
- Target: esp32c3 (DevkitM-1)

**Additional context**
For reference, the modified `main.c` from the sample is attached as [main.txt](https://github.com/zephyrproject-rtos/zephyr/files/13459607/main.txt)

henrikbrixandersen, 2023-11-28:
@abdulayubzoi Can you reproduce this on current `main`?

fabiobaltieri, 2023-11-28:
@abdulayubzoi this may be a limitation of the controller, but if it used to work in 3.4.0 there may be something else there. Would you be able to `git bisect` the issue and find what specific commit broke it?

abdulayubzoi, 2023-11-29:
> @abdulayubzoi Can you reproduce this on current `main`?
Yes, I've been able to reproduce it on main.

carlescufi, 2024-04-11:
> We shall submit the new hal_v5.1 base PR in a couple of weeks.

@sylvioalves this was done, right? Can you close the issue then?

sylvioalves, 2024-04-11:
> > We shall submit the new hal_v5.1 base PR in a couple of weeks.
> 
> @sylvioalves this was done, right? Can you close the issue then?

Yes.

closed 2024-04-11 by sylvioalves

ilanchezh1yan, 2025-02-01:
Hello All,
I am a young fellowman started to work with BLE communication establishment in esp32_devikitc_wroom with zephyr. I built the "peripheral_gatt_write" in sample directory. As per the code logic the terminal logs for "Advertising started" but nrf_Connect application can't scan for the device. 

There is no hardware issue I have verified with ArduinoIDE. The code works fine with nrf52832. I am blindfolded without any leads. Requesting for clarification. 

It would be better if you discuss about the HCI interface between the RADIO and the controller. In Zephyr devicetree "bt_hci_bus=ipm". Is that inter-processor mailbox is the bus between the RADIO and the controller.

Serial port COM10
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM10 instead...
--- idf_monitor on \\.\COM10 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffbdb60,len:10712
load:0x40080000,len:91040
0x40080000: _WindowOverflow4 at D:/zephyrproject/zephyr/arch/xtensa/core/window_vectors.S:56

1150 mmu set 00010000, pos 00010000
entry 0x40090048
0x40090048: __start at D:/zephyrproject/zephyr/soc/espressif/common/loader.c:253

I (58) soc_init: ESP Simple boot
I (58) soc_init: compile time Feb  1 2025 11:11:01
W (59) soc_init: Unicore bootloader
I (59) spi_flash: detected chip: generic
I (62) spi_flash: flash io: dio
I (65) soc_init: chip revision: v3.1
I (69) flash_init: SPI Speed      : 40MHz
I (72) flash_init: SPI Mode       : DIO
I (76) flash_init: SPI Flash Size : 4MB
I (80) boot: DRAM: lma 0x00001020 vma 0x3ffbdb60 len 0x29d8   (10712)
I (86) boot: IRAM: lma 0x00003a00 vma 0x40080000 len 0x163a0  (91040)
0x40080000: _WindowOverflow4 at D:/zephyrproject/zephyr/arch/xtensa/core/window_vectors.S:56

I (92) boot: IRAM: lma 0x00019db8 vma 0x00000000 len 0x6240   (25152)
I (98) boot: IMAP: lma 0x00020000 vma 0x400d0000 len 0x33428  (209960)
0x400d0000: _stext at ??:?

I (104) boot: IRAM: lma 0x00053430 vma 0x00000000 len 0xcbc8   (52168)
I (110) boot: DMAP: lma 0x00060000 vma 0x3f400000 len 0x6b2c   (27436)
I (117) boot: Image with 6 segments
I (120) boot: DROM segment: paddr=00060000h, vaddr=3f400000h, size=06B2Ch ( 27436) map
I (128) boot: IROM segment: paddr=00020000h, vaddr=400d0000h, size=33428h (209960) map
I (146) boot: Jumping to the main image...
I (174) heap_runtime: ESP heap runtime init at 0x3ffc9e34 size 88 kB.
*** Booting Zephyr OS build v4.0.0-3819-gddff91fef6ac ***
[00:00:00.175,000] <inf> esp32_bt_adapter: BT controller compile version [241c96c]
[00:00:00.397,000] <wrn> net_buf: Timeout discarded. No blocking in syswq
[00:00:00.397,000] <inf> bt_hci_core: Identity: 88:13:BF:70:8D:90 (public)
[00:00:00.397,000] <inf> bt_hci_core: HCI: version 4.2 (0x08) revision 0x030e, manufacturer 0x0060
[00:00:00.397,000] <inf> bt_hci_core: LMP: version 4.2 (0x08) subver 0x030e
Bluetooth initialized
Advertising successfully started

Thanking you for your valuable time in advance.