zephyrproject-rtos/zephyr #65748: echo-server net sample crash in Atmel sam_e70_xplained if usermode is enabled

jukkar, 2023-11-24:
**Describe the bug**
The `samples/net/sockets/echo-server` crashes if userspace is enabled in Atmel `sam_e70_xplained` board.

**To Reproduce**
```
mkdir build
west build -p -b sam_e70_xplained samples/net/sockets/echo_server/ -d build/echo-server -- -DCONFIG_USERSPACE=y
west flash -d build/echo-server
```

**Expected behavior**
No crash

**Impact**
Not sure if this has ever worked so the crash might be expected. Anyway, the userspace can be enabled and it does something in this platform so expectation is it to work. Verified that the sample app works ok in `qemu_x86` with userspace enabled.

nandojve, 2023-11-26:
Hi @jukkar ,

I made a git bisect to get an idea about this issue and this was the result:

```
533aaed7a0c9eb180913c1b7b802bf4df397a276 is the first bad commit
commit 533aaed7a0c9eb180913c1b7b802bf4df397a276
Author: Jukka Rissanen <jukka.rissanen@linux.intel.com>
Date:   Tue Nov 10 15:00:49 2020 +0200

    samples: net: echo-server: Print receive statistics periodically
    
    Print receive statistics of how many KiB we receive / sec.
    
    Signed-off-by: Jukka Rissanen <jukka.rissanen@linux.intel.com>

 samples/net/sockets/echo_server/src/common.h |  5 +++++
 samples/net/sockets/echo_server/src/tcp.c    | 30 ++++++++++++++++++++++++++++
 samples/net/sockets/echo_server/src/udp.c    | 30 ++++++++++++++++++++++++++++
 3 files changed, 65 insertions(+)
```

This is the last good
```
commit 526328963113fd51a140ffd6d54bf44e9d9b4a9f
Author: Peter Bigot <peter.bigot@nordicsemi.no>
Date:   Sun Nov 22 06:31:34 2020 -0600

    drivers: regulator: avoid improper access to internal structures
    
    Nothing in the API description the delayed work structure sanctions
    direct reference to internal fields.  Do not assume that a delayed
    work item can be submitted without delay by invoking k_work_submit()
    with a reference to the contained work item.  Instead submit with the
    delayed API and no wait.
    
    Signed-off-by: Peter Bigot <peter.bigot@nordicsemi.no>

```

where

```
uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-1776-g533aaed7a0c9  ***
[00:00:00.001,000] <inf> eth_sam: MAC: fc:c2:3d:08:90:35
[00:00:00.001,000] <inf> eth_sam: Queue 0 activated
[00:00:00.001,000] <inf> eth_sam: Queue 1 set to idle
[00:00:00.001,000] <inf> eth_sam: Queue 2 set to idle
[00:00:00.001,000] <inf> eth_sam_phy: Soft Reset of ETH PHY
[00:00:00.091,000] <inf> eth_sam_phy: PHYID: 0x221573 at addr: 0
[00:00:00.097,000] <inf> net_config: Initializing network
[00:00:00.097,000] <inf> net_config: Waiting interface 1 (0x20418404) to be up...
[00:00:02.112,000] <inf> eth_sam: Link up
[00:00:03.695,000] <inf> eth_sam_phy: common abilities: speed 100 Mb, full duplex
[00:00:03.695,000] <inf> net_config: Interface 1 (0x20418404) coming up
[00:00:03.695,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:03.795,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:03.795,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:03.795,000] <inf> net_echo_server_sample: Run echo server
[00:00:03.795,000] <inf> net_echo_server_sample: Network connected
[00:00:03.795,000] <inf> net_echo_server_sample: Starting...
[00:00:03.795,000] <err> os: ***** MPU FAULT *****
[00:00:03.795,000] <err> os:   Data Access Violation
[00:00:03.795,000] <err> os:   MMFAR Address: 0x2040624c
[00:00:03.795,000] <err> os: r0/a1:  0x20402fe8  r1/a2:  0x0042278b  r2/a3:  0x000927c0
[00:00:03.795,000] <err> os: r3/a4:  0x2040624c r12/ip:  0x00401165 r14/lr:  0x004196a3
[00:00:03.795,000] <err> os:  xpsr:  0xa1000000
[00:00:03.795,000] <err> os: Faulting instruction address (r15/pc): 0x0041955e
[00:00:03.795,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:03.795,000] <err> os: Current thread: 0x204041b0 (tcp6_thread_id)
[00:00:03.865,000] <err> os: Halting system
```

and

```
uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-1775-g0ed6a3cd6969  ***
[00:00:00.001,000] <inf> eth_sam: MAC: fc:c2:3d:08:90:35
[00:00:00.001,000] <inf> eth_sam: Queue 0 activated
[00:00:00.001,000] <inf> eth_sam: Queue 1 set to idle
[00:00:00.001,000] <inf> eth_sam: Queue 2 set to idle
[00:00:00.001,000] <inf> eth_sam_phy: Soft Reset of ETH PHY
[00:00:00.091,000] <inf> eth_sam_phy: PHYID: 0x221573 at addr: 0
[00:00:00.097,000] <inf> net_config: Initializing network
[00:00:00.097,000] <inf> net_config: Waiting interface 1 (0x20418404) to be up...
[00:00:02.112,000] <inf> eth_sam: Link up
[00:00:03.695,000] <inf> eth_sam_phy: common abilities: speed 100 Mb, full duplex
[00:00:03.695,000] <inf> net_config: Interface 1 (0x20418404) coming up
[00:00:03.695,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:03.823,000] <inf> net_config: IPv6 address: fd00::fec2:3dff:fe08:9035
[00:00:03.823,000] <inf> net_echo_server_sample: Run echo server
[00:00:03.823,000] <inf> net_echo_server_sample: Network connected
[00:00:03.823,000] <inf> net_echo_server_sample: Starting...
[00:00:03.823,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv6~
[00:00:03.824,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv4~
[00:00:03.824,000] <inf> net_echo_server_sample: <log_strdup alloc failed>
[00:00:03.824,000] <inf> net_echo_server_sample: <log_strdup alloc failed>
```

By the commit dates it has been long since this worked with SAM family. I was wondering if so you have another point closer to mainline where this worked for this particular hardware.

When I debug the application in the mainline 20979f8 I know that when 
https://github.com/zephyrproject-rtos/zephyr/blob/ecd2961160644823d9924c9360030a9f712f561b/samples/net/sockets/echo_server/src/tcp.c#L389 or https://github.com/zephyrproject-rtos/zephyr/blob/ecd2961160644823d9924c9360030a9f712f561b/samples/net/sockets/echo_server/src/tcp.c#L404 execute the system crashes.

The below point is never reached

https://github.com/zephyrproject-rtos/zephyr/blob/ecd2961160644823d9924c9360030a9f712f561b/samples/net/sockets/echo_server/src/tcp.c#L130

I ran samples/userspace/[[hello_world_user | prod_consumer | shared_mem] without issues.
I know that gmac ethernet driver define special non cached regions and I'm not following what is going on with MPU regions so I'm not sure if it could be related.

jukkar, 2023-11-26:
> By the commit dates it has been long since this worked with SAM family. I was wondering if so you have another point closer to mainline where this worked for this particular hardware.

Unfortunately I just tried usermode first time with this hw so I was just guessing if it should work or not. As you said it might be related to Ethernet driver issues. I reported it so that the issue is known and not forgotten. Not sure if this is worth investigating as the hw is kind of old already. Perhaps we could set the configs so that selecting this hw with networking and usermode enabled would not be allowed.

nandojve, 2023-12-03:
Hi @jukkar ,

I open the #66067 with a few changes to get opinion from others that know ARM and how to handle MPU Regions. Using the patch I got same error. So, my question is there are any arm board that runs this example correctly using usermode?

For me it is not clear if it is missing configuration in the SoC/Driver, sample, or possible how ARM MPU is handled. So, if we can isolate maybe we can solve the issue.

The current mainline version of the driver define pure __no_cache and the patch is a try to isolate the buffers on a NoCache RAM region. The sample/userspace examples continue to work and driver seems to continue operational using other network examples.

CC: @carlocaione

closed 2024-02-17 by github-actions[bot]