zephyrproject-rtos/zephyr #65762: Receive ping reply from ::: when Border Router pings a multicast group (NRF52840/OpenThread)

qnquangg, 2023-11-25:
Hi team, I'm new to OpenThread and Zephyr. Could you please provide some guidance? Thank you.

### SCENARIO
- I'm currently developing an application on NRF52840 with OpenThread. My goal is to test multicast for Packet Delivery Rate (PDR) using ping.
- Here is my setup:
Border Router: Raspberry Pi with NRF52840 Dongle.
Node: NRF52840 DK, Zephyr, version 2.5.0
- Both devices can join the same OpenThread network, where the Border Router acts as the leader, and the node as a router. I can see it on web UI of Border Router.

### ISSUE
The problem I'm encountering is that when the border router pings the multicast group using a command, it also receives a packet from `:::`. I don't want to see `:::` in the output as it causes duplicate packets in my log. Could you please review my setup and code, which I've attached below, and let me know if everything is correct?

```
$ ping -6 -t 1 -I wpan0 ff06::abcd
PING ff06::abcd(ff06::abcd) from fdaa:fb3:4d49:1:8a41:7df0:abb3:3811 wpan0: 56 data bytes
64 bytes from fd31:9326:2ca1:b79e:e776:b4ee:5770:d7b9: icmp_seq=1 ttl=64 time=31.3 ms
64 bytes from ::: icmp_seq=1 ttl=64 time=37.9 ms
64 bytes from fd31:9326:2ca1:b79e:e776:b4ee:5770:d7b9: icmp_seq=2 ttl=64 time=18.3 ms
64 bytes from ::: icmp_seq=2 ttl=64 time=44.9 ms
64 bytes from fd31:9326:2ca1:b79e:e776:b4ee:5770:d7b9: icmp_seq=3 ttl=64 time=17.2 ms
64 bytes from ::: icmp_seq=3 ttl=64 time=32.2 ms
64 bytes from fd31:9326:2ca1:b79e:e776:b4ee:5770:d7b9: icmp_seq=4 ttl=64 time=15.6 ms
64 bytes from ::: icmp_seq=4 ttl=64 time=20.8 ms
^C
--- ff06::abcd ping statistics ---
4 packets transmitted, 4 received, +4 duplicates, 0% packet loss, time 3003ms
rtt min/avg/max/mdev = 15.552/27.266/44.900/10.151 ms
```

--------------
My code to subscribe multicast IP address:

prj.conf
```
# nRF board library
CONFIG_DK_LIBRARY=y

# Generic networking options
CONFIG_NETWORKING=y

# Network shell
CONFIG_SHELL=y
CONFIG_SHELL_ARGC_MAX=26
CONFIG_SHELL_CMD_BUFF_SIZE=416

# Network sockets
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_SOCKETS_POLL_MAX=4

# Enable Openthread FTD features set
CONFIG_OPENTHREAD=y
CONFIG_OPENTHREAD_THREAD_VERSION_1_3_1=y
CONFIG_OPENTHREAD_FTD=y

# Openthread settings
CONFIG_OPENTHREAD_PANID=4660
CONFIG_OPENTHREAD_CHANNEL=15
CONFIG_OPENTHREAD_NETWORK_NAME="OpenThreadDemo"
CONFIG_OPENTHREAD_NETWORKKEY="00:11:22:33:44:55:66:77:88:99:aa:bb:cc:dd:ee:ff"
CONFIG_OPENTHREAD_XPANID="1111111122222222"

# L2 OpenThread enabling
CONFIG_NET_L2_OPENTHREAD=y
CONFIG_OPENTHREAD_SHELL=y
# CONFIG_IEEE802154_NET_IF_NO_AUTO_START=y

# CONFIG_MBEDTLS_SHA1_C=n
# CONFIG_FPU=y
```

main.c
```
static void on_thread_state_changed(otChangedFlags flags, struct openthread_context *ot_context,
									void *user_data)
{
	if (flags & OT_CHANGED_THREAD_ROLE)
	{
		switch (otThreadGetDeviceRole(ot_context->instance))
		{

		case OT_DEVICE_ROLE_ROUTER:
			LOG_INF("OT_DEVICE_ROLE_ROUTER");
			struct otIp6Address addr;
			inet_pton(AF_INET6, "ff06::abcd", &addr);
			openthread_api_mutex_lock(ot_context);
			otIp6SubscribeMulticastAddress(ot_context->instance, &addr);
			openthread_api_mutex_unlock(ot_context);
			break;

		default:
			LOG_INF("OT DEFAULT STATE");
			break;
		}
	}
}
```

I also have workaround solution, but I don't think it is good. My solution is to subscribe 2 multicast groups at the same time.
```
                         inet_pton(AF_INET6, "ff06::abcd", &addr);
			openthread_api_mutex_lock(ot_context);
			otIp6SubscribeMulticastAddress(ot_context->instance, &addr);
			openthread_api_mutex_unlock(ot_context);
                         inet_pton(AF_INET6, "ff06::abcd", &addr);
			openthread_api_mutex_lock(ot_context);
			otIp6SubscribeMulticastAddress(ot_context->instance, &addr);
			openthread_api_mutex_unlock(ot_context);
```

Please advise if I am doing correctly or not. Thanks for your time.

rlubos, 2023-11-28:
The duplicate reply seems to be caused by both, OT and Zephyr stack processing the ping request. Try the following code to disable ICMP Echo Request processing int the OT stack:
```
otIcmp6SetEchoMode(openthread_get_default_instance(), OT_ICMP6_ECHO_HANDLER_DISABLED);
```
It's not possible to disable ICMP Echo Requests in the Zehpyr side.

As for the missing source address in replies (I assume from Zephyr), it looks like some IP address misconfiguration. From the Zephyr perspective, you'd need a global address to reply to a multicast of scope larger than mesh-local (> ff03). Can you paste the output of the following commands?
```
net iface
ot ipaddr
ot ipmaddr
```

rlubos, 2023-11-29:
Thanks for the details. I see two issues here

1. Sending ping reply with `::` source address is clearly wrong and is a bug on the Zephyr side. If Zephyr cannot match the source address for the reply, it should drop the packet instead of sending it. I'll send a fix shortly.

2. But for the reason **why** there is no source address match:
- OTBR sends ping from `fdaa:fb3:4d49:1:8a41:7df0:abb3:3811` which has a different prefix than on mesh prefix
- Thread device has only `fd31:9326:2ca1:b79e:c357:29:54a3:d428` assigned, which is mesh-local ULA. Such address is not routable outside of Thread mesh network, therefore is not selected as a match for OTBR address.

I believe Border Router should assign Supplemantal ULA prefix for the mesh, so that devices can configure an address that can be used for external configuration? @konradderda @edmont @canisLupus1313 Thoughts?

canisLupus1313, 2023-11-29:
@qnquangg @rlubos The Thread border router should advertise ULA prefix based on config. Please refer to:
https://openthread.io/guides/border-router/docker/test-connectivity#form-the-thread-network

This should do the trick.

edmont, 2023-11-29:
On the node device, `ot prefix` command can be used to show the configured prefixes.

rlubos, 2023-11-29:
As for the Zephyr bug, I've opened https://github.com/zephyrproject-rtos/zephyr/pull/65907 with at fix, Zephry should no longer send a reply with unspecified IPv6 source address.

rlubos, 2023-12-12:
@qnquangg Ping, any updates? Did you manage to resolve the issue with the suggestions above?

carlescufi, 2024-01-10:
Closing due to inactivity. Please reopen if still an issue.

closed 2024-01-10 by carlescufi