zephyrproject-rtos/zephyr #65763: GD32 flash(v2) driver bank0 writing problem.

serhang, 2023-11-25:
Hello
Nowadays, I'm trying to integrate GD32F103 to Zephyr 3.5 and I have encountered a problem about bank0 writing function.

I believe the code at flash_gd32_v2.c (gd32_fmc_v2_bank0_write) -> line 80
	flash_prg_t *prg_flash = (flash_prg_t *)((uint8_t *)SOC_NV_FLASH_SIZE + offset);
should be
	flash_prg_t *prg_flash = (flash_prg_t *)((uint8_t *)SOC_NV_FLASH_ADDR + offset);

Thanks.

serhang, 2023-11-25:
I also found some problems about erase function.

in my case these changes work:

in flash_gd32_v2.c
**function:** gd32_fmc_v2_bank0_erase_block

**line 161 and 162:**
size -= GD32_NV_FLASH_V2_BANK0_PAGE_SIZE;
page_addr += GD32_NV_FLASH_V2_BANK0_PAGE_SIZE;
instead of
size -= GD32_NV_FLASH_V2_BANK0_SIZE;
page_addr += GD32_NV_FLASH_V2_BANK0_SIZE;

and

**function:** flash_gd32_valid_range

**line 300 and 305:**
if (offset % GD32_NV_FLASH_V2_BANK0_PAGE_SIZE) {
(len % GD32_NV_FLASH_V2_BANK0_PAGE_SIZE)) {

instead of

if (offset % GD32_NV_FLASH_V2_BANK0_SIZE) {
(len % GD32_NV_FLASH_V2_BANK0_SIZE)) {

**Notes:**
I created some partitions on main flash of GD32F103 and checked only functions of bank0.

cameled, 2023-11-27:
@serhang Thanks for look into gd32 flash driver.

You are right, there have some problem in gd32 flash v2 driver. I will create a new PR to fix it later in today.

cameled, 2023-11-27:
@serhang  I have create a new PR to fix it. Could you please help to check if there still have problem in here? Thanks!

serhang, 2023-11-27:
@cameled Thank you. The partitions on bank0 work well with the changes. We also designed a board based on gd32f207rgt6. After your commit I'll check the bank1 functions too.

PR #65791 (2023-12-01): drivers: flash: update gd32 flash v2
There have some incorrect usage of bank0/1 size and bank0/1 page size.

Fix: #65763
closed 2023-12-01 by fabiobaltieri