zephyrproject-rtos/zephyr #65771: qemu-system-riscv64 high CPU usage with hello-world sample

yf13, 2023-11-26:
**Describe the bug**
When trying hello world sample with `qemu_riscv64` target on x86_64 Ubuntu Jammy host, the CPU usage of `qemu-sysem-riscv64` is too high.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr
2. west build -b qemu_riscv64 samples/hello_world
3. west build -t run
5. check CPU usage with `top` command

**Expected behavior**
A simple hello world program shouldn't cause high CPU usage.

**Impact**
People may feel bad

**Logs and console output**
None

**Environment (please complete the following information):**
 - OS: Linux Ubuntu Jammy
 - Toolchain: Zephyr SDK v0.16.3
 - Commit SHA or Version used: Zephyr OS build v3.5.0-rc3-9-gb272760f062b

**Additional context**
N/A

henrikbrixandersen, 2023-11-28:
CC: @kartben I believe this is to be expected, but perhaps needs to be documented better?

yf13, 2023-11-29:
@henrikbrixandersen can you kindly teach why it is expected?  

I guess that  after `hello_world` app thread exists its `main()` function, other Zephyr threads are running for `qemu_riscv64`  board.  But I couldn't understand why CPU usage is so high. I set debug break at `arch_cpu_idle()` function and found it has been repeated hit after app main thread exits. After delete the break point and continue, I got high CPU usage again.

I saw similar [high cpu issue](https://github.com/mit-pdos/xv6-riscv/issues/15) in other projects, which seems to be resolved with proper use of `wfi` instruction. Not sure if Zephyr has similar situation.

fkokosinski, 2023-11-29:
> which seems to be resolved with `wfi` instruction under proper

From my investigation of this issue, this can explain at least part of the problem. I've noticed that on the `qemu_riscv64` board, the `sys_clock_set_timeout` function from the RISC-V machine time driver is called multiple times after the main function finishes. This doesn't happen on the `qemu_cortex_a53` (or any other arm or arm64 QEMU boards). Setting `MTIE` to 0 helps, but of course causes any timeout-related code to not function properly. So, IMO, the cause of this high CPU usage is the way the RISC-V machine timer driver handles `MTIE` when tickless kernel is enabled.

On the other hand, I've noticed the high cpu usage on other architectures as well: ARC, MIPS and SPARC. So I think that this should be treated more as a feature request rather than a bug.

yf13, 2023-11-29:
Forgot to mention that my QEMU console shows a warning:

```
*** Booting Zephyr OS build zephyr-v3.5.0-2239-ga51bd53cef2b ***
qemu-system-riscv64: warning: icount sleep disabled and no active timers
Hello World! qemu_riscv64
```

With debugger I saw the call to `sys_clock_set_timeout()` are with` (ticks=2147483647, idle=false)` after app main() returns.  are these incoming parameters correct?

yf13, 2024-09-03:
When trying with the stock QEMU on Ubuntu 22.04, the issue doesn't occur:

```shell
$ west build -b  qemu_riscv64 samples/hello_world --pristine
$ qemu-system-riscv64 -M virt -m 256 -nographic  -bios none -kernel build/zephyr/zephyr.elf
```
So likely the issue is related to the QEMU used by `west build -t run`?