zephyrproject-rtos/zephyr #65795: flash_stm32_qspi.c fails to compile with CONFIG_MISRA_SANE=y

benner, 2023-11-27:
**Application using QSPI and CONFIG_MISRA_SANE=y fails to compile**

If I enable `CONFIG_MISRA_SANE=y` in `prj.conf` application fails to compile.

Target platform: STM32G4

**To Reproduce**
On STM32, configure QSPI device (`st,stm32-qspi`), enable  `CONFIG_MISRA_SANE=y` in `prj.conf`.

**Expected behavior**
I'm able to compile the project

**Impact**
Cannot enforce  `CONFIG_MISRA_SANE` for whole project.

**Logs and console output**

```
/tmp/zephyrworkspace/zephyr/drivers/flash/flash_stm32_qspi.c: In function 'flash_stm32_qspi_init':
/tmp/zephyrworkspace/zephyr/drivers/flash/flash_stm32_qspi.c:1278:95: error: ISO C90 forbids array 'raw' whose size cannot be evaluated [-Werror=vla]
 1278 |                 uint8_t raw[JESD216_SFDP_SIZE(decl_nph)];
      |                                                                                               ^                   
cc1: some warnings being treated as errors
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.16.3
 - Using older but tested currently newest commit 5da13ea30a6a1b1beef19466a4f2211f34b4d61c too.

**Additional context**
DT:
```
		quadspi1: quadspi@a0001000 {
			compatible = "st,stm32-qspi";
			#address-cells = <0x1>;
			#size-cells = <0x0>;
			reg = <0xa0001000 0x400>;

			interrupts = <95 0>;

			clocks = <&rcc STM32_CLOCK_BUS_AHB3 0x00000100>;
			status = "disabled";
		};
```

FRASTM, 2023-11-27:
not sure this is a stm32 issue ... but a jesd216 one

FRASTM, 2023-12-08:
Note that when CONFIG_MISRA_SANE=y
build error occur when compiling  samples/drivers/jesd216/
```
~/samples/drivers/jesd216/src/main.c: In function 'main':
~/zephyr/samples/drivers/jesd216/src/main.c:325:17: error: ISO C90 forbids variable length array 'dw' [-Werror=vla]
  325 |                 uint32_t dw[php->len_dw];
      |                 ^~~~~~~~
cc1: some warnings being treated as errors
```

closed 2024-05-12 by github-actions[bot]