zephyrproject-rtos/zephyr #65805: Thread network / get DNS servers from OTBR

typepub, 2023-11-27:
**Is your enhancement proposal related to a problem? Please describe.**
Thread device need to get dynamically the list of DNS servers from OTBR, same way it gets the IPv6 prefix.

**Describe the solution you'd like**
When Thread connection is made, get IPv6 prefix and provision (using DNS Resolve Init ?) automatically a context with the DNS servers proposed by the OTBR

**Describe alternatives you've considered**
So far, static configuration of DNS servers in the Kconfig is working well but do not comply with generic configuration requirements

**Additional context**
N/A

rlubos, 2023-11-27:
Do I understand correctly, that what's needed here is to implement processing of `DNS_SERVERS` option in DHCPv6 client, as described in [RFC3646](https://www.rfc-editor.org/rfc/rfc3646.txt)?

typepub, 2023-11-27:
Hi @rlubos 

It is more related to [rfc8106](https://datatracker.ietf.org/doc/html/rfc8106). OpenThread Border Router use RA to inform the Thread network. The idea is, for the device, to catch this information (as it receives from RA the routing information) and provision the DNS server list according to the rfc8106.

By the way, as rfc8106 states that RA can be used concurrently to DHCPv6, support for DHCPv6 DNS option is not mutually exclusive with RA.

rlubos, 2023-11-27:
Hmm, if it's about Router Advertisement processing, we already have RDNSS option handling implemented (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/Kconfig.ipv6#L134), please make sure you have the option enabled.

> OpenThread Border Router use RA to inform the Thread network.
However, I probably have some outdated Thread information, but I'm slightly confused here. I thought that Neighbor Discovery protocol is not used within Thread networks (on-mesh)?

typepub, 2023-11-27:
Agree on both points. RDNSS is activated but seems not working as, when asking for DNS server, the answer is not correct. DNS server from device point of view:
`dns config
Server: [2001:4860:4860:0:0:0:0:8888]:53
ResponseTimeout: 6000 ms
MaxTxAttempts: 3
RecursionDesired: yes
ServiceMode: srv_txt_opt
Nat64Mode: allow
Done
`

rlubos, 2024-04-03:
Closing due to lack of activity an answers for the questions asked.