zephyrproject-rtos/zephyr #65817: Confusing warning: "No ID address. App must call settings_load"

mondalaci, 2023-11-27:
I get the following warning in the console:

`I: No ID address. App must call settings_load()`

According to https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/zephyr/connectivity/bluetooth/bluetooth-arch.html#persistent-storage :

> Once enabled, it is the responsibility of the application to call settings_load() after having initialized Bluetooth (using the bt_enable() API).

My code works as suggested:

```
bt_enable(NULL);
settings_load();
```

If I reverse the order of the above function calls, I get:

```
Starting Bluetooth Peripheral HIDS keyboard
I: 6 Sectors of 4096 bytes
I: alloc wra: 4, d08
I: data wra: 4, 5c0
I: SoftDevice Controller build revision: 
I: e0 7e 2e c1 5e 05 85 23 |.~..^..#
I: 46 15 dc fa 8e 29 7d 70 |F....)}p
I: 10 93 a5 fc             |....    
I: HW Platform: Nordic Semiconductor (0x0002)
I: HW Variant: nRF52x (0x0002)
I: Firmware: Standard Bluetooth controller (0x00) Version 224.11902 Build 2231721665
I: No ID address. App must call settings_load()
I: Identity: F0:98:11:CE:A4:92 (random)
I: HCI: version 5.4 (0x0d) revision 0x1077, manufacturer 0x0059
I: LMP: version 5.4 (0x0d) subver 0x1077
```

This warning message always appears, even if I use the API as suggested, so I think it's rather disturbing and should be removed.

jhedberg, 2023-11-27:
Hi @mondalaci, what makes you think the message is a warning? It's an "info" message, since it's critical that the app calls `settings_load()`  when `CONFIG_BT_SETTIINGS` is enabled. Some functions of the stack might work without an identity address, but many will not work. The message is basically a reminder to app-developers that if they build with such a configuration they need to remember to call settings_load(). The stack cannot know if the stack will or will not call settings_load() after bt_enabled() so it always prints this reminder message. How would you propose to change the message to make it look less than a warning? If you would completely remove it, how would you instead give an indication to app-developers when their app exhibits mysterious behaviour because the Bluetooth stack is only partially initialised (because settings_load() was never called)?

mondalaci, 2023-11-27:
Hi @jhedberg, sorry, I mean an info message, not a warning. But I think it's a good idea only to display messages when really justified, regardless.

Alternatively, how about displaying this info message when the functions that only work with an identity address are called and when previously, `settings_load()` wasn't called?

thedjnK, 2023-11-27:
> Hi @jhedberg, sorry, I mean an info message, not a warning. But I think it's a good idea only to display messages when really justified, regardless.
>
> Alternatively, how about displaying this info [...]

Why should overhead be added to **every** function that uses settings?

mondalaci, 2023-11-27:
Any way to implement my suggestion with some kind of preprocessor magic? That'd involve no runtime overhead.

thedjnK, 2023-11-27:
> Any way to implement my suggestion with some kind of preprocessor magic? That'd involve no runtime overhead.

No

mondalaci, 2023-11-27:
While I find the info message that gets displayed even if the API is used correctly annoying, if no good solution exists, I don't want to push this too hard, and feel free to close this issue.