zephyrproject-rtos/zephyr #65836: modem_cellular: hl7800 PPP: IPCP never resolves

rerickson1, 2023-11-27:
**Describe the bug**
When using the modem_cellular driver with the HL7800 modem, PPP IPCP resolution never completes.

**To Reproduce**
1. Checkout zephyr main SHA c2dc103101a5d4c5153e39b3571cd2b543bf317a (date 23-11-27)
2. apply patch
[23-11-27_cellular_modem_hl7800.patch](https://github.com/zephyrproject-rtos/zephyr/files/13481379/23-11-27_cellular_modem_hl7800.patch)
4. `cd samples/net/cellular_modem`
5. `west build -p auto -b pinnacle_100_dvk`
6. `west flash`

**Expected behavior**
IPCP should complete and allow IP data to be exchanged

**Impact**
showstopper

**Logs and console output**
[23-11-27_hl7800.log](https://github.com/zephyrproject-rtos/zephyr/files/13481387/23-11-27_hl7800.log)

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain Zephyr SDK 0.16.3
 - Zephyr c2dc103101a5d4c5153e39b3571cd2b543bf317a

bjarki-trackunit, 2023-11-28:
Just looking at the log, the modem is sending PPP frames, but the PPP L2 subsystem is not handling them at all, leaving me to believe it has not been enabled, has `net_if_up()` and `net_if_carrier_on()` been called on the net_l2 interface? I will look closer into it later :)

rerickson1, 2023-11-28:
Sierra Wireless recommended I move ATD to after the device registering on the network. It made no difference.
Also, fixed starting CMUX needs to wait for OK.
[23-11-28_cellular_modem_hl7800.patch](https://github.com/zephyrproject-rtos/zephyr/files/13492735/23-11-28_cellular_modem_hl7800.patch)
[23-11-28_01_hl7800_ipcp_fail.log](https://github.com/zephyrproject-rtos/zephyr/files/13492736/23-11-28_01_hl7800_ipcp_fail.log)

rerickson1, 2023-11-28:
Correction, [23-11-28_01_hl7800_ipcp_fail.log](https://github.com/zephyrproject-rtos/zephyr/files/13492736/23-11-28_01_hl7800_ipcp_fail.log) actually succeeded in IPCP negotiation, but the DNS query failed way before the network interface was ready.

This log shows IPCP never completing:
[23-11-28_02_hl7800_ipcp_fail.log](https://github.com/zephyrproject-rtos/zephyr/files/13492927/23-11-28_02_hl7800_ipcp_fail.log)

rerickson1, 2023-11-28:
Log showing IPCP completing and then getting into a loop where the iface goes up/down forever.

[23-11-28_03_hl7800_ipcp_fail.log](https://github.com/zephyrproject-rtos/zephyr/files/13493363/23-11-28_03_hl7800_ipcp_fail.log)

rerickson1, 2023-12-01:
@bjarki-trackunit thanks for taking a look. I have been trying everything.
I disabled CMUX and just use PPP and there is no change in behavior.
I agree with what you see. The HL7800 is requesting a reconfig after the interface is brought up.

rerickson1, 2024-03-11:
@bjarki-trackunit since last discussion, something has fixed the issue! I know there were many low level UART fixes, maybe one of those solved the issue.

```
*** Booting Zephyr OS build v3.6.0-693-ga454bb9abcab ***
Powering on modem
Bring up network interface
Waiting for L4 connected
Waiting for DNS server added
Retrieving cellular info
RSSI -75
IMEI: 35461609029951
MODEL_ID: HL7800
MANUFACTURER: Sierra Wireless
SIM_IMSI: 001010123456789
FW_VERSION: HL7800.4.7.1.0
Performing DNS lookup of test-endpoint.com
Resolved to 139.177.179.5
Opening UDP socket
Socket opened
Sending echo packet
Receiving echoed packet
Echo transmit time 507ms
Sending echo packet
Receiving echoed packet
Echo transmit time 527ms
Sending echo packet
Receiving echoed packet
Echo transmit time 495ms
Sending echo packet
Receiving echoed packet
Echo transmit time 506ms
Sending echo packet
Receiving echoed packet
Echo transmit time 559ms
Sending echo packet
Receiving echoed packet
Echo transmit time 510ms
Sending echo packet
Receiving echoed packet
Echo transmit time 519ms
Sending echo packet
Receiving echoed packet
Echo transmit time 528ms
Sending echo packet
Receiving echoed packet
Echo transmit time 487ms
Sending echo packet
Receiving echoed packet
Echo transmit time 499ms
Sending echo packet
Receiving echoed packet
Echo transmit time 494ms
Sending echo packet
Receiving echoed packet
Echo transmit time 486ms
Sending echo packet
Receiving echoed packet
Echo transmit time 494ms
Sending echo packet
Receiving echoed packet
Echo transmit time 504ms
Sending echo packet
Receiving echoed packet
Echo transmit time 512ms
Sending echo packet
Receiving echoed packet
Echo transmit time 481ms
Successfully sent and received 16 of 16 packets
Average time per successful echo: 506 ms
Close UDP socket
Opening UDP socket
Socket opened
Sending 128 packets
Awaiting response from server
Server received 128 packets
Server dropped 0 packets
Time elapsed sending packets 18055ms
Throughput 7259 bytes/s
Close UDP socket
Restart modem
Waiting for L4 connected
L4 connected
Opening UDP socket
Socket opened
Sending echo packet
Receiving echoed packet
Echo transmit time 495ms
Sending echo packet
Receiving echoed packet
Echo transmit time 485ms
Sending echo packet
Receiving echoed packet
Echo transmit time 513ms
Sending echo packet
Receiving echoed packet
Echo transmit time 503ms
Sending echo packet
Receiving echoed packet
Echo transmit time 506ms
Sending echo packet
Receiving echoed packet
Echo transmit time 516ms
Sending echo packet
Receiving echoed packet
Echo transmit time 522ms
Sending echo packet
Receiving echoed packet
Echo transmit time 514ms
Sending echo packet
Receiving echoed packet
Echo transmit time 510ms
Sending echo packet
Receiving echoed packet
Echo transmit time 499ms
Sending echo packet
Receiving echoed packet
Echo transmit time 498ms
Sending echo packet
Receiving echoed packet
Echo transmit time 495ms
Sending echo packet
Receiving echoed packet
Echo transmit time 488ms
Sending echo packet
Receiving echoed packet
Echo transmit time 498ms
Sending echo packet
Receiving echoed packet
Echo transmit time 509ms
Sending echo packet
Receiving echoed packet
Echo transmit time 490ms
Successfully sent and received 16 of 16 packets
Average time per successful echo: 502 ms
Close UDP socket
Powering down modem
Sample complete
```

closed 2024-03-13 by dleach02