zephyrproject-rtos/zephyr #65863: bcm958402m2_m7 build fails with CONFIG_CACHE_MANAGEMENT

tbursztyka, 2023-11-28:
**Describe the bug**
The title says it all.
Building with CONFIG_CACHE_MANAGEMENT ends up with such errors for bcm958402m2_m7:
```
error: implicit declaration of function 'SCB_EnableDCache' 
```

cmsis module does not seem to include the relevant cache header for that specific soc/cpu.

To "fix" it, it only requires to de-select CPU_HAS_DCACHE and CPU_HAS_ICACHE which happen to be selected for arch/arm/cortex_m/Kconfig at option CPU_CORTEX_M7

I have no idea if cortex m7 has data/instruction cache support or if it's a cmsis module issue.

**To Reproduce**
Steps to reproduce the behavior:
1. scripts/twister -p bcm958402m2_m7 -T tests/kernel/cache

**Expected behavior**
It should probably be skipped by twister.

**Impact**
CI failure

henrikbrixandersen, 2023-11-28:
Tagging as medium since this is blocking other PRs.

carlocaione, 2023-12-04:
I'm a bit lost on this.

> (make sure CONFIG_CACHE_MANAGEMENT is set in test's prj.conf)

It is not already? https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/kernel/cache/prj.conf

> scripts/twister -p bcm958402m2_m7 -T tests/kernel/cache

```
INFO    - Total complete:    2/   2  100%  skipped:    2, failed:    0, error:    0
INFO    - 2 test scenarios (2 test instances) selected, 2 configurations skipped (0 by static filter, 2 at runtime).
INFO    - 0 of 2 test configurations passed (0.00%), 0 failed, 0 errored, 2 skipped with 0 warnings in 3.14 seconds
```

closed 2024-02-26 by github-actions[bot]