zephyrproject-rtos/zephyr #65892: drivers: bluetooth: hci: h4 high priority packtes ignored if CONFIG_BT_RECV_BLOCKING is not set

msmttchr, 2023-11-28:
Looking at the code below:

https://github.com/zephyrproject-rtos/zephyr/blob/617cb789c34585921c07c883d9739faa62d75ce1/drivers/bluetooth/hci/h4.c#L333-L354

and with actual test, it appears to me that packets with `evt_flags` defined as `BT_HCI_EVT_FLAG_RECV_PRIO` by are ignored/discarded when CONFIG_BT_RECV_BLOCKING is not set.
This typically happens for `BT_HCI_EVT_CMD_COMPLETE` or `BT_HCI_EVT_CMD_STATUS` events making the driver unusable.

henrikbrixandersen, 2023-11-29:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

msmttchr, 2023-11-29:
> Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

Ok, thanks for the clear indication. I updated the description according to the format.