zephyrproject-rtos/zephyr #65899: nrf5340: retained memory GPREGRET driver faults due to using wrong address

thedjnK, 2023-11-28:
**Describe the bug**
If using GPREGRET1 on an nRF5340-based board with the retained memory driver, the board will fault with an odd address - 0x551c. There appears to be something wrong in dts land because the addresses should be getting remapped by this code:
```
                peripheral@50000000 {
                        #address-cells = <1>;
                        #size-cells = <1>;
                        ranges = <0x0 0x50000000 0x10000000>;

                        /* Common nRF5340 Application MCU
                         * peripheral description
                         */
                        #include "nrf5340_cpuapp_peripherals.dtsi"
                };
```
Which has this in the included file:
```
power: power@5000 {
        compatible = "nordic,nrf-power";
        reg = <0x5000 0x1000>;
        interrupts = <5 NRF_DEFAULT_IRQ_PRIORITY>;
        status = "okay";
        #address-cells = <1>;
        #size-cells = <1>;

        gpregret1: gpregret1@551c {
                #address-cells = <1>;
                #size-cells = <1>;
                compatible = "nordic,nrf-gpregret";
                reg = <0x551c 0x1>;
                status = "okay";
        };

        gpregret2: gpregret2@5520 {
                #address-cells = <1>;
                #size-cells = <1>;
                compatible = "nordic,nrf-gpregret";
                reg = <0x5520 0x1>;
                status = "okay";
        };
};
```

However, seemingly, `DT_INST_REG_ADDR()` is returning 0x551c for this and not the remapped address of 0x5000551c

**Expected behavior**
Driver should work and address returned should be shifting up which it is not

**Impact**
Showstopper

**Environment (please complete the following information):**
 - Commit SHA or Version used: 789addb02f6cd7479601ed3b487bcf57c2a3919c

nordicjm, 2023-11-29:
@mbolivar-ampere is this a bug in the dts tools/parsing? Seems that the parent here gets the address updated (0x50005000), but it's child nodes do not (0x551c)

mbolivar-ampere, 2023-11-30:
This is not a bug; the `power` node does not have a `ranges` property, so the address space of its child nodes is a completely distinct address space.

If you want to translate `gpregret1` and `gpregret2` addresses into the parent address space, add a `ranges` property to the `power` node appropriately. See the DT spec for details.

closed 2023-11-30 by mbolivar-ampere