zephyrproject-rtos/zephyr #65902: gptp: wrong gptp_announce message length

mariopaja, 2023-11-29:
**Issue:** Device warns for invalid gptp_announce message length:

**Invalid length for ANNOUNCE packet should have 68 bytes but has 64 byte**

We found out that the issue is at [#define GPTP_ANNOUNCE_LEN(pkt)](https://github.com/zephyrproject-rtos/zephyr/blob/9521371157c23e70cfb58b6a76aa9b65c93868a9/subsys/net/l2/ethernet/gptp/gptp_messages.h#L68).

```
#define GPTP_ANNOUNCE_LEN(pkt) \
	(sizeof(struct gptp_hdr) + sizeof(struct gptp_announce) \
	 + ntohs(GPTP_ANNOUNCE(pkt)->tlv.len) \
	 - sizeof(struct gptp_path_trace_tlv) + 4)
```

Is there any particular reason why +4 is added?

**Solution**: For us removing +4 from GPTP_ANNOUNCE_LEN solved the issue. The devices then could successfully receive the announce messages

jukkar, 2023-11-29:
There is a comment above that line that says
```
For the Announce message, the TLV is variable length. The len field
indicates the length of the TLV not accounting for tlvType and lengthField
which are 4 bytes.
```
Perhaps that +4 is incorrect as you say. Would you be able to send a PR that fixes it?

mariopaja, 2023-11-29:
> There is a comment above that line that says
>
> ```
> For the Announce message, the TLV is variable length. The len field
> indicates the length of the TLV not accounting for tlvType and lengthField
> which [...]

Sure, I can send a PR that addresses this issue. 

TVL is a variable length based on the 802.1AS-2011 but in Zephyr it is fixed to  [12 bytes.](https://github.com/zephyrproject-rtos/zephyr/blob/9521371157c23e70cfb58b6a76aa9b65c93868a9/subsys/net/l2/ethernet/gptp/gptp_messages.h#L152)

```
struct gptp_path_trace_tlv {
	/** TLV type: 0x8. */
	uint16_t type;

	/** Length. Number of TLVs * 8 bytes. */
	uint16_t len;

	/** ClockIdentity array of the successive time-aware systems. */
	uint8_t  path_sequence[1][8];
} __packed;
```

[image]

closed 2023-12-01 by mariopaja