zephyrproject-rtos/zephyr #65957: [tfm][lpcxpresso55s69_ns] all tfm_integration cases failed.

hakehuang, 2023-11-30:
**Describe the bug**
all tfm_integration cases failed.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - lpcxpresso55s69_ns
 - What have you tried to diagnose or workaround this issue?
 - N/A
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
this case is never PASS in v3.5.0
 - ...

**To Reproduce**
Steps to reproduce the behavior:
cd samples/tfm_integration/psa_protected_storage

```
west build -b lpcxpresso55s69_ns
west flash --file=build/zephyr/tfm_merged.hex
```
the log is not as expected

**Expected behavior**
see the read me files in each tfm_integration

**Impact**
tfm_integration

**Logs and console output**

```
[INF] Starting bootloader
[WRN] This device was provisioned with dummy keys. This device is NOT SECURE
[INF] Image index: 1, Swap type: none
[INF] Starting bootloader
[WRN] This device was provisioned with dummy keys. This device is NOT SECURE
[INF] Image index: 1, Swap type: none
[INF] Image index: 0, Swap type: none
[INF] Bootloader chainload address offset: 0x8000
[INF] Jumping to the first image slot
[WRN] This device was provisioned with dummy keys. This device is NOT SECURE
[Sec Thread] Secure image initializing!
Booting TF-M v1.8.0
[INF][Crypto] Provisioning entropy seed... complete.

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, )
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: 915f8a10b62b24b244d2da00793e2b670f677763

butok, 2023-12-01:
> @hakehuang , have these tests passed on `lpcxpresso55s69_ns` before?

butok, 2023-12-01:
Reopened. 
Sorry, closed by accident, by pressing wrong GH button.

butok, 2023-12-01:
> > @hakehuang , have these tests passed on `lpcxpresso55s69_ns` before?
> 
> I just set up testing for this _ns board, the formal testing is just download the zephyr.elf, not the tfm_merged.hex, which is not correct.

TFM requires tfm_merged.hex.

hakehuang, 2023-12-02:
> It should be a common TFM issue, could you check it?
> BTW: Do we have somewhere regular Zephyr TFM regression tests?

Community NO. but I have set up regular testing on NXP platforms, this is why I report this issue. and lpc55s609 is the only platform that we support

hakehuang, 2023-12-06:
git bisect direct to this commit

```
c6e3bac4f20a659143a56bda0166a111629d79b1 is the first bad commit
commit c6e3bac4f20a659143a56bda0166a111629d79b1
Author: Emilio Benavente <emilio.benavente@nxp.com>
Date:   Wed Jan 11 17:40:46 2023 -0600
 
    soc: arm: lpc55xxx: Updated clock init
 
    Updated the clock init to reflect the sdk also
    updated the clock frequencies to reflect the
    respective soc clock values, this file originally
    contained unexpected clock values, updated comments
    to reflect changes and got rid of doxygen style
    comments
 
    Signed-off-by: Emilio Benavente <emilio.benavente@nxp.com>
 
soc/arm/nxp_lpc/lpc55xxx/Kconfig.soc |  10 ++-
soc/arm/nxp_lpc/lpc55xxx/soc.c       | 129 ++++++++++++++++++++++++++---------
2 files changed, 104 insertions(+), 35 deletions(-)

```

butok, 2023-12-06:
Thank you @hakehuang, saved us time.

Hi @EmilioCBen, could you have a look at your commit c6e3bac4f20a659143a56bda0166a111629d79b1. It was flagged as a bad one that broke the TFM LPC55S69. Can it be modified somehow to make TFM work again?

PR #66624 (2023-12-19): soc: lpc55xxx: Fix TFM
TFM is using flash, so sys. clock must be decreased. 
Fixes #65957
commit abcfd0cb (2023-12-19): soc: lpc55xxx: Fix TFM

TFM is using flash, so sys. clock must be decreased.
Fixes #65957
closed 2023-12-19 by carlescufi