zephyrproject-rtos/zephyr #65973: slist find without remove

Thalley, 2023-11-30:
**Is your enhancement proposal related to a problem? Please describe.**
The main issue is that if you add the same node twice a slist, then the slist iterators will end up in an infinite loop as the node will simply point to itself. 

To help prevent this, it would be nice to have a quick way to see if a node already exist in an slist before adding it. 

Currently we only have `sys_slist_find_and_remove` which technically could be used to always (attempt to) remove a node before adding it, but there's no reason to actually remove it. 

**Describe the solution you'd like**
Add `sys_slist_find` or `sys_slist_exists` that returns a value (either boolean or the node itself).

**Describe alternatives you've considered**
Alternatively `sys_slist_append` should return an error in the case that a node is added more than once. 

**Additional context**
N/A

peter-mitsis, 2023-11-30:
Thinking aloud ...

Searching for a specific node in a _slist_ requires a linear search which is an O(N) operation. This is something that we would not want as default behaviour in sys_slist_append(). Probably not for debugging behaviour either as that linear search would come with an expected high performance penalty. I would expect the better practice would be to foist the responsibility of a search onto the caller so that the penalty is not incurred by all users of sys_slist_append()--even in a debugging scenario.

If it is done, then my preference would be for naming it something like ...
`    bool sys_snode_is_linked(sys_slist_t * list, sys_snode_t *node, sys_snode_t **prev);`
If the node does need to be removed later, it may then be done more quickly and easily without necessarily incurring the penalty of another linear search.

Thalley, 2023-11-30:
> Searching for a specific node in a slist requires a linear search which is an O(N) operation. 

Agreed. An alternative would be to check that the `node->next` is NULL, which also would have the benefit of not accidentally removing it from another list, when adding to the new list.

> If the node does need to be removed later, it may then be done more quickly and easily without necessarily incurring the penalty of another linear search.

Not a bad option.

zacchaeusliang, 2023-12-11:
Hey i'm intrested in working on this ticket!

moonlight83340, 2023-12-18:
I have tried to look at the issue myself with the proposition of @peter-mitsis 
https://github.com/moonlight83340/zephyr/commit/04b6f598059ae9d7b168f0171b98dd8d64a01f93
I'm not really sure about my modification because, I haven't a big angle on that project.. 
Thinking also that unit test would be great..

Thalley, 2023-12-18:
Thanks @zacchaeusliang and @moonlight83340 

Those interesting approaches to fixing this, both satisfying the requirements I had. 

I'll leave it to @andyross and @peter-mitsis to decided which (if any or both) of the approaches should be gone with. 

@moonlight83340 Feel free to create a pull request with your commit so that we can provide review comments

Thalley, 2023-12-28:
> @Thalley Is my commit still needed?

Until a solution has been approved, I'd say that either solution is still valid :) I don't have a strong preference, as either solution solved this issue in a satisfactory way.

closed 2024-03-12 by nashif