zephyrproject-rtos/zephyr #65976: CI is broken for `qemu_arc_hs5x` board

finikorg, 2023-11-30:
I have CI failures when running thread tests. Quick test:

```
west build -p -b qemu_arc_hs5x zephyr/tests/kernel/threads/thread_apis -T kernel.threads.apis -t run
```

There is also test PR: https://github.com/zephyrproject-rtos/zephyr/pull/65966

henrikbrixandersen, 2023-12-01:
Git bisect point to commit 7a5fcb8c60e2e6ba26704fabc28fe2d4e0106513.
From https://github.com/zephyrproject-rtos/zephyr/actions/runs/7054749220/job/19204123350

```
*** Booting Zephyr OS build zephyr-v3.5.0-2432-g8f9fd239eb26 ***
Running TESTSUITE threads_lifecycle
===================================================================
START - test_abort_from_isr
 PASS - test_abort_from_isr in 0.001 seconds
===================================================================
START - test_abort_from_isr_not_self
 PASS - test_abort_from_isr_not_self in 0.001 seconds
===================================================================
START - test_essential_thread_abort
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/kernel/sched.c:1879
	aborting essential thread 0x8040d450
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x8040d90c (test_essential_thread_abort)
 PASS - test_essential_thread_abort in 0.003 seconds
===================================================================
START - test_essential_thread_operation
 PASS - test_essential_thread_operation in 0.001 seconds
===================================================================
START - test_k_busy_wait

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/threads/thread_apis/src/main.c:575: threads_lifecycle_test_k_busy_wait: dt >= k_us_to_cyc_floor64(100) is false

 FAIL - test_k_busy_wait in 0.002 seconds
===================================================================
START - test_k_thread_stack_space_get_user
 PASS - test_k_thread_stack_space_get_user in 0.001 seconds
===================================================================
START - test_resume_unsuspend_thread
 PASS - test_resume_unsuspend_thread in 0.001 seconds
===================================================================
START - test_systhreads_idle
 PASS - test_systhreads_idle in 0.105 seconds
===================================================================
START - test_systhreads_main
 PASS - test_systhreads_main in 0.001 seconds
===================================================================
START - test_thread_join
 PASS - test_thread_join in 0.118 seconds
===================================================================
START - test_k_thread_state_str
 PASS - test_k_thread_state_str in 0.001 seconds
===================================================================
START - test_threads_abort_repeat
 PASS - test_threads_abort_repeat in 0.214 seconds
===================================================================
START - test_threads_cpu_mask
 PASS - test_threads_cpu_mask in 0.002 seconds
===================================================================
START - test_threads_suspend_resume_cooperative
 PASS - test_threads_suspend_resume_cooperative in 0.217 seconds
===================================================================
TESTSUITE threads_lifecycle_1cpu succeeded

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  96.43% [threads_lifecycle]: pass = 27, fail = 1, skip = 1, total = 29 duration = 3.298 seconds
 - PASS - [threads_lifecycle.test_abort_from_isr] duration = 0.001 seconds
 - PASS - [threads_lifecycle.test_abort_from_isr_not_self] duration = 0.001 seconds
 - PASS - [threads_lifecycle.test_essential_thread_abort] duration = 0.003 seconds
 - PASS - [threads_lifecycle.test_essential_thread_operation] duration = 0.001 seconds
 - FAIL - [threads_lifecycle.test_k_busy_wait] duration = 0.002 seconds
 - PASS - [threads_lifecycle.test_k_thread_stack_space_get_user] duration = 0.001 seconds
 - PASS - [threads_lifecycle.test_resume_unsuspend_thread] duration = 0.001 seconds
 - PASS - [threads_lifecycle.test_systhreads_idle] duration = 0.105 seconds
 - PASS - [threads_lifecycle.test_systhreads_main] duration = 0.001 seconds
[... 16 lines cut ...]
 - PASS - [threads_lifecycle.test_threads_suspend] duration = 0.110 seconds
 - PASS - [threads_lifecycle.test_threads_suspend_resume_preemptible] duration = 0.219 seconds
 - PASS - [threads_lifecycle.test_threads_suspend_timeout] duration = 0.269 seconds
 - PASS - [threads_lifecycle.test_user_mode] duration = 0.001 seconds

SUITE PASS - 100.00% [threads_lifecycle_1cpu]: pass = 12, fail = 0, skip = 0, total = 12 duration = 1.663 seconds
 - PASS - [threads_lifecycle_1cpu.test_customdata_get_set_coop] duration = 0.507 seconds
 - PASS - [threads_lifecycle_1cpu.test_customdata_get_set_preempt] duration = 0.510 seconds
 - PASS - [threads_lifecycle_1cpu.test_delayed_thread_abort] duration = 0.060 seconds
 - PASS - [threads_lifecycle_1cpu.test_k_busy_wait_user] duration = 0.110 seconds
 - PASS - [threads_lifecycle_1cpu.test_k_thread_foreach] duration = 0.020 seconds
 - PASS - [threads_lifecycle_1cpu.test_k_thread_foreach_null_cb] duration = 0.002 seconds
 - PASS - [threads_lifecycle_1cpu.test_k_thread_foreach_unlocked] duration = 0.018 seconds
 - PASS - [threads_lifecycle_1cpu.test_k_thread_foreach_unlocked_null_cb] duration = 0.002 seconds
 - PASS - [threads_lifecycle_1cpu.test_k_thread_state_str] duration = 0.001 seconds
 - PASS - [threads_lifecycle_1cpu.test_threads_abort_repeat] duration = 0.214 seconds
 - PASS - [threads_lifecycle_1cpu.test_threads_cpu_mask] duration = 0.002 seconds
 - PASS - [threads_lifecycle_1cpu.test_threads_suspend_resume_cooperative] duration = 0.217 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: 468be086bf1bb038c2b90c707f3805cd
PROJECT EXECUTION FAILED
```

finikorg, 2023-12-01:
I tested older SDK 0.16.3 and the test is working there

keith-packard, 2023-12-01:
argh. this test is working for me with SDK 0.16.4 and 0.16.3. I'm running this tree:
```
commit 5174e94c91c560de8961da3e81a4b001473daa6d (HEAD -> main, origin/main)
Author: Mariusz Skamra <mariusz.skamra@codecoup.pl>
Date:   Tue Nov 21 09:15:15 2023 +0100

    Bluetooth: audio: has: Fix coverity issues
    
    This fixes issues discovered by coverity check.
    
    Fixes: #65325
    Fixes: #65326
    Signed-off-by: Mariusz Skamra <mariusz.skamra@codecoup.pl>
```
Any suggestions on what I can do to try and reproduce this?

keith-packard, 2023-12-01:
Hrm. Could this just be a timing problem? Have you tried running with the twister `--retry-failed` option to see if it is just a transient issue?

finikorg, 2023-12-04:
You can reproduce this here: https://github.com/zephyrproject-rtos/zephyr/pull/65966
Just trigger a new build.
I am able to reproduce it locally always.

keith-packard, 2023-12-04:
I managed to reproduce this by running on a much slower test machine. As I mentioned in #65966, I think this is a bug in the test which assumes that all cycles will be accounted to the thread doing the busy-wait operation and none will be consumed elsewhere. I don't understand how cycles are accounted to threads, so I could be wrong here...

PR #66119 (2023-12-05): tests/thread_apis: Allow 20µs slop in k_busy_wait test
When testing k_busy_wait(100), don't require that at least 100µs of CPU time be consumed by the thread itself; some of that time may be used elsewhere in the system. Instead, just make sure that at least 80µs is accounted to the thread and allow 20µs to get used elsewhere.

This series also has a patch to remove an open-coded time conversion in z_impl_k_busy_wait; I can pull that out separately if desired.

Closes: #65976
closed 2023-12-05 by aescolar