zephyrproject-rtos/zephyr #65980: Implement ThreadX Comparability Layer

cfriedt, 2023-11-30:
**Is your feature request related to a problem? Please describe.**

The ThreadX API is pretty lightweight and was just released under the MIT license [1]. We should implement it in Zephyr for companies who need a migration strategy.

**Describe the solution you'd like**
An implementation of the (main) ThreadX interface using Zephyr API calls under the hood.

Potentially NetX support as well (although ThreadX had some limited BSD Sockets API support).

**Describe alternatives you've considered**

We could of course not do this, it's just that it would be a relatively easy user base to pick up if people are thinking of making the switch.

**Additional context**
Microsoft opens sources ThreadX under MIT license • The Register

https://www.theregister.com/2023/11/28/microsoft_opens_sources_threadx/

nordicjm, 2023-12-01:
> We could of course not do this

Gets my vote

cfriedt, 2023-12-01:
> > We could of course not do this
> 
> Gets my vote

@nordicjm - FWIW, I did this back in 2018 or something for a previous employer as a research project and it was surprisingly easy.

Not that I'm saying I want to do it again by any means 😅

nordicjm, 2023-12-01:
If people want to move their code over to zephyr, they should do it properly. I don't see the point in a threadx/freertos/etc. porting layer that people in zephyr have to implement and then support. If something breaks, we then have to fix it. It sounds like a lot of work for no real gain.

cfriedt, 2023-12-05:
> @cfriedt How about the migration guide? I know it will become outdated sooner or later, but at least the Zephyr Project won't need to maintain that piece of code.

That's a good idea!

cfriedt, 2023-12-05:
Technically, we could also offer a FreeRTOS migration guide.

nordicjm, 2023-12-05:
A migration guide for both sounds like a great idea