zephyrproject-rtos/zephyr #66023: OTS - OLCP Indications sent before write rsp

piotrnarajowski, 2023-12-01:
**Describe the bug**
OTS test cases require that at least one object stored on the IUT is initialized as Current Object. This is achieved by Lower Tester performing procedure as desrcibed in section 4.2.4. of [OTS TS](https://www.bluetooth.org/docman/handlers/DownloadDoc.ashx?doc_id=309983). According to subsection 6. of this preamble procedure, "The IUT, after sending a Write Response to acknowledge the write to the Object List Control Point, sends an indication of the Object List Control Point characteristic.". Unfortunately, zephyr in a first place is sending indication of the OLCP characteristic followed by write response.

**To Reproduce**
Can be reproduced with AutoPTS

**Expected behavior**
According to [OTP spec](https://www.bluetooth.org/docman/handlers/DownloadDoc.ashx?doc_id=309936) section 4.4.4.: "An OLCP or OACP operation is started when the ATT Write Response is received from the Object Server as a result of the Object Client writing an Op Code to a control point to perform some desired action. The control point operation ends when the Object Server sends an indication to the Object Client with the Op Code set to Response Code.". This suggests that correct order while performing OLCP or OACP operations should be Write Response First, Indication second.

**Impact**
Unable to pass any OTS test case that require operations on objects. 

**Logs and console output**
[OTS_SR_CR_BV_02_C_2023_12_01_10_52_20.zip](https://github.com/zephyrproject-rtos/zephyr/files/13524934/OTS_SR_CR_BV_02_C_2023_12_01_10_52_20.zip)

**Environment (please complete the following information):**
 - OS: Windows 11
 - Toolchain: Zephyr SDK 0.16.4
 - [PR#66022](https://github.com/zephyrproject-rtos/zephyr/pull/66022)

Thalley, 2023-12-05:
Thanks for reporting this. @jhedberg I am unsure I and Asbjørn should be assigned to this, as we did not write the server implementation (I think that was @kapi-no ?), and this will be very low priority for us.

piotrnarajowski, 2024-02-05:
Please remove the stale label as this is the reason why over 30 OTS test cases fail.

jhedberg, 2024-04-06:
Is @kapi-no the right assignee for this? There doesn't seem to be much interest (or effort) for getting this fixed.

Thalley, 2024-04-06:
I guess not. I'll assign it back to me, as it's related to some LE Audio tests (even though the implementation is not).

sjanc, 2024-04-19:
Fixed in https://github.com/zephyrproject-rtos/zephyr/pull/71679

closed 2024-04-23 by carlescufi