zephyrproject-rtos/zephyr #66027: samples: cpp: hello: cannot use `std::cout` on xtensa platforms

cfriedt, 2023-12-01:
**Describe the bug**
The standard "Hello world" app does not work in C++ for xtensa platforms because the toolchain is unable to find POSIX symbols `write()`, `read()`, `lseek()`, `fstat()`, and `close()`.

The xtensa toolchain in the Zephyr SDK seems to use a unique implementation that differs from all other architectures for the standard `<<` operator in `<iostream>`, since `samples/cpp/hello_world` seems to work fine for all other supported platforms (aside from x86, but that is a separate bug).

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? `qemu_xtensa`
 - What have you tried to diagnose or workaround this issue? tried to build the sample app, saw the error
 - Is this a regression? no

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -p auto -b qemu_xtensa samples/cpp/hello_world`
2. See error

**Expected behavior**
The sample should work the way it does on all other architectures / platforms.

**Impact**
Annoyance

**Logs and console output**
```shell
[1/7] Linking CXX executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/cfriedt/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map 
: && ccache /opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/xtensa-dc233c_zephyr-elf-g++  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/cfriedt/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/arch/arch/xtensa/core/startup/libarch__xtensa__core__startup.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/timer/libdrivers__timer.a  modules/xtensa/libmodules_xtensa_hal.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/xtensa-dc233c_zephyr-elf/12.2.0"  -L/home/cfriedt/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lstdc++  -Wl,-no-pie  -DPICOLIBC_LONG_LONG_PRINTF_SCANF  --specs=picolibc.specs  -lc  -lgcc && cd /home/cfriedt/zephyrproject/zephyr/build/zephyr && /usr/bin/cmake -E true
/opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/xtensa-dc233c_zephyr-elf/12.2.0/../../../../xtensa-dc233c_zephyr-elf/bin/ld.bfd: /opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/../../picolibc/xtensa-dc233c_zephyr-elf/lib/libstdc++.a(basic_file.o): in function `(anonymous namespace)::xwrite(int, char const*, int)':
basic_file.cc:(.text._ZN12_GLOBAL__N_1L6xwriteEiPKci+0x15): undefined reference to `write'
/opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/xtensa-dc233c_zephyr-elf/12.2.0/../../../../xtensa-dc233c_zephyr-elf/bin/ld.bfd: /opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/../../picolibc/xtensa-dc233c_zephyr-elf/lib/libstdc++.a(basic_file.o): in function `std::__basic_file<char>::xsgetn(char*, int)':
basic_file.cc:(.text._ZNSt12__basic_fileIcE6xsgetnEPci+0x15): undefined reference to `read'
/opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/xtensa-dc233c_zephyr-elf/12.2.0/../../../../xtensa-dc233c_zephyr-elf/bin/ld.bfd: /opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/../../picolibc/xtensa-dc233c_zephyr-elf/lib/libstdc++.a(basic_file.o): in function `std::__basic_file<char>::seekoff(long long, std::_Ios_Seekdir)':
basic_file.cc:(.text._ZNSt12__basic_fileIcE7seekoffExSt12_Ios_Seekdir+0x21): undefined reference to `lseek'
/opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/xtensa-dc233c_zephyr-elf/12.2.0/../../../../xtensa-dc233c_zephyr-elf/bin/ld.bfd: /opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/../../picolibc/xtensa-dc233c_zephyr-elf/lib/libstdc++.a(basic_file.o): in function `std::__basic_file<char>::showmanyc()':
basic_file.cc:(.text._ZNSt12__basic_fileIcE9showmanycEv+0xa): undefined reference to `fstat'
/opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/xtensa-dc233c_zephyr-elf/12.2.0/../../../../xtensa-dc233c_zephyr-elf/bin/ld.bfd: basic_file.cc:(.text._ZNSt12__basic_fileIcE9showmanycEv+0x31): undefined reference to `lseek'
/opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/xtensa-dc233c_zephyr-elf/12.2.0/../../../../xtensa-dc233c_zephyr-elf/bin/ld.bfd: /opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/../../picolibc/xtensa-dc233c_zephyr-elf/lib/libc.a(libc_tinystdio_fdopen.c.o):(.literal.fdopen+0x18): undefined reference to `read'
/opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/xtensa-dc233c_zephyr-elf/12.2.0/../../../../xtensa-dc233c_zephyr-elf/bin/ld.bfd: /opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/../../picolibc/xtensa-dc233c_zephyr-elf/lib/libc.a(libc_tinystdio_fdopen.c.o):(.literal.fdopen+0x1c): undefined reference to `write'
/opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/xtensa-dc233c_zephyr-elf/12.2.0/../../../../xtensa-dc233c_zephyr-elf/bin/ld.bfd: /opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/../../picolibc/xtensa-dc233c_zephyr-elf/lib/libc.a(libc_tinystdio_fdopen.c.o):(.literal.fdopen+0x20): undefined reference to `lseek'
/opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/xtensa-dc233c_zephyr-elf/12.2.0/../../../../xtensa-dc233c_zephyr-elf/bin/ld.bfd: /opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/../../picolibc/xtensa-dc233c_zephyr-elf/lib/libc.a(libc_tinystdio_fdopen.c.o):(.literal.fdopen+0x24): undefined reference to `close'
/opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/xtensa-dc233c_zephyr-elf/12.2.0/../../../../xtensa-dc233c_zephyr-elf/bin/ld.bfd: /opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/../../picolibc/xtensa-dc233c_zephyr-elf/lib/libc.a(libc_tinystdio_fdopen.c.o): in function `fdopen':
fdopen.c:(.text.fdopen+0x1d): undefined reference to `close'
/opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/xtensa-dc233c_zephyr-elf/12.2.0/../../../../xtensa-dc233c_zephyr-elf/bin/ld.bfd: /opt/zephyr/zephyr-sdk-0.16.3/xtensa-dc233c_zephyr-elf/bin/../lib/gcc/../../picolibc/xtensa-dc233c_zephyr-elf/lib/libc.a(libc_tinystdio_fopen.c.o): in function `fopen':
fopen.c:(.text.fopen+0x19): undefined reference to `open'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK v0.16.3
 - Commit: cb7b650b923f160540943f0724b79b79fd8e51f4

**Additional context**
Discovered in the context of #65769

tejlmand, 2023-12-11:
@stephanosio is this something you may look into ?

closed 2024-04-06 by github-actions[bot]

elazar-vayyar, 2024-07-09:
This issue is still relevant on latest Zephyr code - as detailed in issue #75650. 
It is not clear if C++ is actually supported for ESP32.

sylvioalves, 2024-07-10:
> This issue is still relevant on latest Zephyr code - as detailed in issue #75650. It is not clear if C++ is actually supported for ESP32.

Hi @elazar-vayyar , Yes, it is supported. As reference, check `samples/cpp/cpp_synchronization` output:
```c
*** Booting Zephyr OS build v3.7.0-rc2-245-g67a31ef2d781 ***
Create semaphore 0x3fc8c1d4
Create semaphore 0x3fc8c1c0
main: Hello World!
coop_thread_entry: Hello World!
main: Hello World!
coop_thread_entry: Hello World!
```

keith-packard, 2024-07-11:
As I mentioned in #75650, libstdc++uses fopen/fdopen which picolibc layers on top of POSIX calls.
```
diff --git a/samples/cpp/hello_world/prj.conf b/samples/cpp/hello_world/prj.conf
index b801dee1855..5fdf8ea531e 100644
--- a/samples/cpp/hello_world/prj.conf
+++ b/samples/cpp/hello_world/prj.conf
@@ -1,2 +1,5 @@
 CONFIG_CPP=y
 CONFIG_REQUIRES_FULL_LIBCPP=y
+CONFIG_POSIX_DEVICE_IO=y
+CONFIG_POSIX_FD_MGMT=y
+CONFIG_POSIX_FILE_SYSTEM=y
```
makes this sample build.

Should picolibc using libstdc++ automatically define these?

cfriedt, 2024-07-11:
> As I mentioned in #75650, libstdc++uses fopen/fdopen which picolibc layers on top of POSIX calls.
> ```
> diff --git a/samples/cpp/hello_world/prj.conf b/samples/cpp/hello_world/prj.conf
> index [...]

I wish that we had this functionality in-tree still 😅

I guess if it works though, why not?

Some people might not like that they are using POSIX under the hood, but it is somewhat baked-in to newlib / picolibc. I guess also, some people might not care or might prefer to just not know 🤷🏼‍♂️

keith-packard, 2024-07-11:
> Some people might not like that they are using POSIX under the hood, but it is somewhat baked-in to newlib / picolibc. I guess also, some people might not care or might prefer to just not know 🤷🏼‍♂️

The alternative would be to implement picolibc's `fopen` directly in terms of Zephyr primitives instead of on top of POSIX APIs. Which would introduce a whole new set of circular dependencies, which I know you're a huge fan of.

hmihaylov-sc, 2024-07-30:
Hello!

I am hitting this issue as well (when calling std::filesystem APIs). I am using the `qemu_cortex_a53` board. Here is my `prj.conf`:

```
CONFIG_MAIN_STACK_SIZE=16384
CONFIG_HEAP_MEM_POOL_SIZE=1000000
CONFIG_SYS_HEAP_RUNTIME_STATS=y
CONFIG_COMMON_LIBC_MALLOC_ARENA_SIZE=2000000
CONFIG_LOG=y

CONFIG_CPP=y
CONFIG_REQUIRES_FULL_LIBCPP=y
CONFIG_STD_CPP20=y
CONFIG_CPP_EXCEPTIONS=y
CONFIG_CPP_RTTI=y
```

And to bypass this I attempted to implement the following functions:

```c++
int close(int) { return -1; }
ssize_t write(int, const void*, size_t) { return -1; }
ssize_t read(int, void*, size_t) { return -1; }
int lstat(const char *, struct stat *) { return -1; }
off_t lseek(int, off_t, int) { return -1; }

// even though they're defined, the linker still does not find them and fails the build
int stat(const char *, struct stat *) { return -1; }
int fstat(int, struct stat *) { return -1; }
int open(const char *, int, ...) { return -1; }
int gettimeofday(struct timeval *, void *) { return -1; }
```

What's odd is the top 4 function resolve fine, but the bottom 4 do not. Do you have an idea how I can work around this issue? Thanks.

keith-packard, 2024-07-30:
> Hello!
> 
> I am hitting this issue as well (when calling std::filesystem APIs). I am using the `qemu_cortex_a53` board. Here is my `prj.conf`:

Did you try adding this to get it to work?
```
CONFIG_POSIX_DEVICE_IO=y
CONFIG_POSIX_FD_MGMT=y
CONFIG_POSIX_FILE_SYSTEM=y
````

> What's odd is the top 4 function resolve fine, but the bottom 4 do not. Do you have an idea how I can work around this issue? Thanks.

That seems weird to me, but as the above should make the file API actually work, I think that's the right solution?

hmihaylov-sc, 2024-07-31:
@keith-packard what fixed it for me was wrapping my C++ code in an `extern "C" {}` block :man_facepalming: When I enable those options the CONFIG_POSIX_* symbols do not exist. I'm building for `qemu_cortex_a53`.

naymore, 2025-02-01:
Adding

```
CONFIG_POSIX_DEVICE_IO=y
CONFIG_POSIX_FD_MGMT=y
CONFIG_POSIX_FILE_SYSTEM=y
```

to my prj.conf didn't change anything for me. Cleaned + pristine build. Still the same linker issues.

```
(.venv) C:\Users\subject9399\zephyrproject\zephyr>west build -b esp32_devkitc_wroom/esp32/procpu samples/cpp/hello_world --pristine
```

Is it supposed to work? Is it just me?

bastienjauny, 2025-04-03:
> prj.conf:
> 
> ```
> CONFIG_CPP=y
> CONFIG_REQUIRES_FULL_LIBCPP=y
> CONFIG_POSIX_DEVICE_IO=y
> CONFIG_POSIX_FD_MGMT=y
> CONFIG_POSIX_FILE_SYSTEM=y
> ```
> 
> Adding the POSIX related config settings didn't change anything for me. Cleaned + pristine build. Still the same linker issue.
> 
> ```
> (.venv) C:\Users\subject9399\zephyrproject\zephyr>west build -b esp32_devkitc_wroom/esp32/procpu samples/cpp/hello_world --pristine
> ```
> 
> Is it supposed to work? Is it just me?

I just tested the proposed solution on v4.0.0 and it works on both target `qemu_xtensa` and `esp32_devkitc_wroom/esp32/procpu`.

tejlmand, 2025-04-28:
@stephanosio can this be closed ?

cfriedt, 2025-05-01:
This line should be removed so that we include xtensa for test coverage. I'll spin up a PR.

https://github.com/zephyrproject-rtos/zephyr/blob/main/samples%2Fcpp%2Fhello_world%2Fsample.yaml#L18

cfriedt, 2025-06-03:
Just following up. It's definitely still a bug that is present in the 0.17.0 SDK.

I'm not sure if I will be able to patch the SDK for xtensa in the near term