zephyrproject-rtos/zephyr #66070: hifive1: tests/kernel/fatal/exception/kernel.common.stack_sentinel timeout

manuargue, 2023-12-04:
**Describe the bug**
`tests/kernel/fatal/exception/kernel.common.stack_sentinel` consistently fails with timeout for `hifive1` board, after 8 runs in CI.

**To Reproduce**
`west twister -p hifive1 -s tests/kernel/fatal/exception/kernel.common.stack_sentinel`
See https://github.com/zephyrproject-rtos/zephyr/actions/runs/7056456377/job/19270115567?pr=66007

**Expected behavior**
Test pass.

**Impact**
Blocking PR merge.

**Logs and console output**
`INFO    - 679/679 hifive1                   tests/kernel/fatal/exception/kernel.common.stack_sentinel  FAILED Timeout (renode 60.604s)`
See https://github.com/zephyrproject-rtos/zephyr/actions/runs/7056456377/job/19270115567?pr=66007

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA or Version used #66007

**Additional context**
N/A

fkokosinski, 2023-12-14:
Hi @manuargue! Thanks for reporting this issue.

This problem is now fixed in upstream Renode. Note that Zephyr, in the image that it runs Twister on, pins the version of Renode to its latest release version. This means that this issue will be fully addressed when:
1) Renode v1.15.0 is released
2) Renode version is bumped here: https://github.com/zephyrproject-rtos/docker-image/blob/main/Dockerfile.ci#L10

Let's keep this issue open until that happens.

nashif, 2024-08-27:
@fkokosinski  this started failing in CI recently (probably after renode update):

https://github.com/zephyrproject-rtos/zephyr/actions/runs/10567555603/job/29283118820

```
23:33:21.2777 [INFO] uart: [host: 1.68s (+1.88ms)|virt:   81.2ms (+1ms)] E: >>> ZEPHYR FATAL ERROR 2147483647: Unknown error on CPU 0
23:33:21.2791 [INFO] uart: [host:  1.68s (+1.5ms)|virt: 81.7ms (+0.5ms)] E: Current thread: 0x80040098 (unknown)
23:33:21.2823 [INFO] uart: [host: 1.68s (+3.15ms)|virt: 82.3ms (+0.6ms)] Caught system error -- reason 2147483647
23:33:21.2842 [INFO] uart: [host:  1.69s (+1.9ms)|virt: 83.2ms (+0.9ms)] test alt thread 6: initiate arbitrary SW exception negative
23:33:21.2857 [INFO] uart: [host: 1.69s (+1.53ms)|virt: 83.9ms (+0.7ms)] E:      a0: fffffffe    t0: 00000000
23:33:21.2866 [INFO] uart: [host: 1.69s (+0.93ms)|virt: 84.4ms (+0.5ms)] E:      a1: 00000000    t1: 00000000
23:33:21.2877 [INFO] uart: [host: 1.69s (+0.99ms)|virt: 84.9ms (+0.5ms)] E:      a2: 00000000    t2: 00000000
23:33:21.2886 [INFO] uart: [host:    1.69s (+1ms)|virt: 85.5ms (+0.6ms)] E:      a3: 00000000    t3: 800008fc
23:33:21.2897 [INFO] uart: [host: 1.69s (+1.09ms)|virt:   86ms (+0.5ms)] E:      a4: fffffffe    t4: 00009880
23:33:21.2908 [INFO] uart: [host: 1.69s (+1.03ms)|virt: 86.6ms (+0.6ms)] E:      a5: 80043000    t5: 00000000
23:33:21.2918 [INFO] uart: [host: 1.69s (+1.06ms)|virt: 87.1ms (+0.5ms)] E:      a6: 00000000    t6: 0000002d
23:33:21.2927 [INFO] uart: [host: 1.69s (+0.87ms)|virt: 87.4ms (+0.3ms)] E:      a7: 00000000
23:33:21.2933 [INFO] uart: [host:  1.7s (+0.68ms)|virt: 87.7ms (+0.3ms)] E:      sp: 80041230
23:33:21.2958 [INFO] uart: [host:   1.7s (+2.5ms)|virt: 88.1ms (+0.4ms)] E:      ra: 800013ec
23:33:21.2964 [INFO] uart: [host:  1.7s (+0.59ms)|virt: 88.4ms (+0.3ms)] E:    mepc: 80000890
23:33:21.2970 [INFO] uart: [host:  1.7s (+0.63ms)|virt: 88.7ms (+0.3ms)] E: mstatus: 00009880
23:33:21.2973 [INFO] uart: [host:  1.7s (+0.26ms)|virt: 88.8ms (+0.1ms)] E: 
23:33:21.2984 [INFO] uart: [host:  1.7s (+1.01ms)|virt: 89.3ms (+0.5ms)] E:      s0: 8004044c    s6: 00000000
23:33:21.2994 [INFO] uart: [host:  1.7s (+1.06ms)|virt: 89.9ms (+0.6ms)] E:      s1: 8000086c    s7: 00000000
23:33:21.3005 [INFO] uart: [host:  1.7s (+1.09ms)|virt: 90.4ms (+0.5ms)] E:      s2: 00000000    s8: 00000000
23:33:21.3014 [INFO] uart: [host:  1.7s (+0.89ms)|virt: 90.9ms (+0.5ms)] E:      s3: 00000000    s9: 00000000
23:33:21.3024 [INFO] uart: [host:     1.7s (+1ms)|virt: 91.5ms (+0.6ms)] E:      s4: 80040098   s10: 00000000
23:33:21.3035 [INFO] uart: [host: 1.71s (+1.13ms)|virt:   92ms (+0.5ms)] E:      s5: 00000000   s11: 00000000
23:33:21.3037 [INFO] uart: [host:  1.71s (+0.2ms)|virt: 92.1ms (+0.1ms)] E: 
23:33:21.3043 [INFO] uart: [host: 1.71s (+0.52ms)|virt: 92.3ms (+0.2ms)] E: call trace:
23:33:21.3054 [INFO] uart: [host: 1.71s (+1.16ms)|virt: 92.9ms (+0.6ms)] E:       0: sp: 80041230 ra: 80000890
23:33:21.3079 [INFO] uart: [host: 1.71s (+2.43ms)|virt: 93.5ms (+0.6ms)] E:       1: sp: 80041234 ra: 8000086c
23:33:21.3091 [INFO] uart: [host: 1.71s (+1.25ms)|virt: 94.2ms (+0.7ms)] E:       2: sp: 8004123c ra: 8000086c
23:33:21.3103 [INFO] uart: [host: 1.71s (+1.15ms)|virt: 94.8ms (+0.6ms)] E:       3: sp: 80041240 ra: 800013ec
23:33:21.3105 [INFO] uart: [host: 1.71s (+0.32ms)|virt:   95ms (+0.2ms)] E: 
23:33:21.3121 [INFO] uart: [host: 1.71s (+1.48ms)|virt: 95.8ms (+0.8ms)] E: >>> ZEPHYR FATAL ERROR -2: Unknown error on CPU 0
23:33:21.3131 [INFO] uart: [host: 1.72s (+1.07ms)|virt: 96.3ms (+0.5ms)] E: Current thread: 0x80040098 (unknown)
23:33:21.3139 [INFO] uart: [host:  1.72s (+0.8ms)|virt: 96.7ms (+0.4ms)] Caught system error -- reason -2
23:33:21.3151 [INFO] uart: [host: 1.72s (+1.21ms)|virt: 97.4ms (+0.7ms)] test stack sentinel overflow - timer irq
23:33:21.3164 [INFO] uart: [host: 1.72s (+1.33ms)|virt: 98.1ms (+0.7ms)] posting 1792 bytes of junk to stack...

```

nashif, 2024-08-27:
ok, this is not the same, this fails on `m2gl025_miv`

nashif, 2024-08-27:
locally I can't reproduce this btw:
```
INFO    - 1/2 m2gl025_miv               tests/kernel/fatal/exception/kernel.common.stack_sentinel PASSED (renode 6.031s)
INFO    - 2/2 hifive1                   tests/kernel/fatal/exception/kernel.common.stack_sentinel PASSED (renode 6.004s)
```

fkokosinski, 2024-08-27:
Hi @nashif, thanks for reporting this. I believe this is because the CI image version used in the twister workflow is still `v0.26.13`, which includes Renode 1.14.0 (so it doesn't include the recent bump).

It looks like we're experiencing the same issue with `m2gl025_miv` as we did with `hifive1` in the past. It's just  that twister didn't pick `m2gl025_miv` up to run this test on for so long.

I couldn't reproduce this issue on `v0.26.14`, which now has Renode 1.15.2. This release includes the fix for this issue for both `hifive1` and `m2gl025_miv` boards. On `v0.26.13` I can reproduce the issue that the CI reported just fine.

So IMO we should:
1) add `m2gl025_miv` to the `platform_exclude` list - I'll prepare a PR for that in a moment
2) after `v0.26.14` is used in the twister workflow - remove both of them from the list and close this issue

PR #77635 (2024-08-29): ci: use latest docker image: v0.26.14
new docker with an updated renode version and rust toolchains.

Fixes #66070
closed 2024-08-29 by carlescufi