zephyrproject-rtos/zephyr #66087: x86 VTD does not work anymore

tbursztyka, 2023-12-04:
**Describe the bug**

VTD is used to enable multi-msi based device. As it allows allocating multiple x86 vectors in a non-contiguous way.
But it takes over all interrupts: uart, hpet, etc... so if it does not work, nothing will.

**To Reproduce**

Build samples/subsys/shell/shell_module for intel_ehl_crb with CONFIG_PCIE_MSI=y and CONFIG_PCIE_MSI_MULTI_VECTOR=y
and run it on target: it will hang after the EFI output. 

**Expected behavior**

The sample should run without any problem

**Impact**

multi-msi based device are unusable on x86. 

**Notes**

This bug has a dependency on #66076 since it requires x86 early console to work to be able to debug anything (VTD does not use the LOG subsystem but prink() at it enables at a very early stage at boot time).

jhedberg, 2024-02-14:
> PR #66272 should fix this.

@tbursztyka it didn't have any reference to this issue. Should this issue be closed now?

jhedberg, 2024-04-15:
> > PR #66272 should fix this.
> 
> @tbursztyka it didn't have any reference to this issue. Should this issue be closed now?

@tbursztyka ping

closed 2024-05-18 by nashif