zephyrproject-rtos/zephyr #66094: rzt2m_starterkit: Doc tags are wrong

nordicjm, 2023-12-04:
**Describe the bug**
The doc for `rzt2m_starterkit` on the ARM boards page does not list this board, instead it lists `Supported Features`, `Connections and IOs` and `Debugging`. This should be fixed. https://docs.zephyrproject.org/latest/boards/arm/index.html

**Expected behavior**
Board to be listed properly

**Impact**
Annoyance

andrzej-kaczmarek, 2023-12-04:
@nordicjm I'm not the proper assignee for this issue, it's not a SmartBond platform and I don't know anything about it.

nordicjm, 2023-12-04:
> @nordicjm I'm not the proper assignee for this issue, it's not a SmartBond platform and I don't know anything about it.

Will search git history and reassign

PR #66105 (2023-12-04): boards: arm: rzt2m_starterkit: fix documentation rendering
As reported in #66094, the documentation used section headers wrongly.

According to the template here: https://github.com/zephyrproject-rtos/zephyr/blob/main/doc/templates/board.tmpl
the top section should be using `#` headline, and no other section should be using `#`.
Fixes #66094.

FYI @nordicjm @tgorochowik
commit e13c2cd6 (2023-12-04): boards: arm: rzt2m_starterkit: fix documentation rendering

As reported in #66094, the documentation used section headers wrongly.

Fixes #66094.
closed 2023-12-04 by fabiobaltieri