zephyrproject-rtos/zephyr #66133: ASSERT due to iso NULL reference after releasing bidirectional streams

fredrikdanebjer, 2023-12-05:
**Description**
When ending a phone call using a bidirectional stream, rarely we are hitting one out of two asserts during teardown of the ISO.

1. in ascs_disconnect_stream_work_handler

2. in bt_iso_chan_disconnected

**Reproduce**

I mention that its hit rarely, as I have not myself been able to reproduce the issue at all, but I have had these asserts reported to me from mass-testing.

1. Create a bidirectional stream, phone call
2. End the call on the client side

**Expected behavior**
Rarely we hit one of two asserts:

1. in ascs_disconnect_stream_work_handler
we hit
__ASSERT(ep != NULL && ep->iso && stream != NULL,
		 "Invalid endpoint %p, iso %p or stream %p",
		 ep, ep == NULL ? NULL : ep->iso, stream);

2. in bt_iso_chan_disconnected
we hit
__ASSERT(chan->iso != NULL, "NULL conn for iso chan %p", chan);

**Impact**
TBD

**Logs and console output**

1. in ascs_disconnect_stream_work_handler
we hit
__ASSERT(ep != NULL && ep->iso && stream != NULL,
		 "Invalid endpoint %p, iso %p or stream %p",
		 ep, ep == NULL ? NULL : ep->iso, stream);

2. in bt_iso_chan_disconnected
we hit
__ASSERT(chan->iso != NULL, "NULL conn for iso chan %p", chan);
**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

A subset of our LE Audio Kconfigs
CONFIG_BT_BAP_UNICAST_SERVER=y
CONFIG_BT_ASCS_MAX_ACTIVE_ASES=2
CONFIG_BT_ASCS_ASE_SNK_COUNT=1
CONFIG_BT_ASCS_ASE_SRC_COUNT=1
CONFIG_BT_CAP_ACCEPTOR=y
CONFIG_BT_CSIP_SET_MEMBER=y
CONFIG_BT_CAP_ACCEPTOR_SET_MEMBER=y
CONFIG_BT_TBS_CLIENT_MINIMAL=y
CONFIG_BT_TBS_CLIENT_ACCEPT_CALL=y
CONFIG_BT_TBS_CLIENT_TERMINATE_CALL=y
CONFIG_BT_TBS_CLIENT_GTBS=y
CONFIG_BT_TBS_CLIENT_CCID=y

Thalley, 2023-12-05:
@fredrikdanebjer please provide the commit SHA, which controller you are using and more logging. If you can, please also provide an HCI log.

fredrikdanebjer, 2023-12-05:
@Thalley I added the commit SHA. However I'm afraid that I cannot provide neither HCI log, not any more logging. These ASSERTs and the circumstances described are the only thing I can provide.
The controller in use is the Demant one, so there is nothing more I can share I'm afraid.

fredrikdanebjer, 2023-12-06:
@MariuszSkamra It sounds like a good start :) as long as it supports CCP. Again I'm very sorry that I can't provide more information.

fredrikdanebjer, 2023-12-11:
@MariuszSkamra we have only tried reproducing this through manual tests. We will also look into running automatic mass/stress on our platform right now.

fredrikdanebjer, 2024-02-05:
We have not been able to reproduce this issue since December, I'm thus closing this.

closed 2024-02-05 by Thalley