zephyrproject-rtos/zephyr #66142: up_squared: tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap fail

KamilxPaszkiet, 2023-12-05:
**Describe the bug**
`tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap`
fail.

**To Reproduce**
twister --device-testing --hardware-map up2.map -s tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap -vv

**Logs and console output**
```
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.5.0-2629-gd70e8d5a8f91 ***
DEBUG   - DEVICE: E: ACPI get table failed: 5
DEBUG   - DEVICE: Running TESTSUITE mheap_api
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_k_aligned_alloc
DEBUG   - DEVICE: PASS - test_k_aligned_alloc in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_malloc_in_isr
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mem_heap/mheap_api_concept/src/test_mheap_api.c:37: tIsr_malloc_and_free: (ptr is NULL)
DEBUG   - DEVICE: bytes allocation failed from system pool
DEBUG   - DEVICE: FAIL - test_malloc_in_isr in 0.019 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_malloc_in_thread
DEBUG   - DEVICE: PASS - test_malloc_in_thread in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_mheap_calloc
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mem_heap/mheap_api_concept/src/test_mheap_api.c:132: mheap_api_test_mheap_calloc: (mem is NULL)
DEBUG   - DEVICE: calloc operation failed
DEBUG   - DEVICE: FAIL - test_mheap_calloc in 0.018 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_mheap_malloc_align4
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mem_heap/mheap_api_concept/src/test_mheap_concept.c:44: mheap_api_test_mheap_malloc_align4: block[i] is NULL
DEBUG   - DEVICE: 
DEBUG   - DEVICE: FAIL - test_mheap_malloc_align4 in 0.017 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_mheap_malloc_free
DEBUG   - DEVICE: PASS - test_mheap_malloc_free in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_mheap_threadsafe
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mem_heap/mheap_api_concept/src/test_mheap_concept.c:60: tmheap_handler: (pool_blocks[thread_id] is NULL)
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mem_heap/mheap_api_concept/src/test_mheap_concept.c:60: tmheap_handler: (pool_blocks[thread_id] is NULL)
DEBUG   - DEVICE: memory is not allocatedmemory is not allocated
DEBUG   - DEVICE: 
DEBUG   - DEVICE: FAIL - test_mheap_threadsafe in 0.037 seconds
DEBUG   - DEVICE: ERROR: cannot fail in test phase 'framework()', bailing
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mem_heap/mheap_api_concept/src/test_mheap_concept.c:60: tmheap_handler: (pool_blocks[thread_id] is NULL)
DEBUG   - DEVICE: TESTSUITE mheap_api failed.
DEBUG   - DEVICE: memory is not allocated
DEBUG   - DEVICE: ------ TESTSUITE SUMMARY START ------
DEBUG   - DEVICE: 
DEBUG   - DEVICE: 
DEBUG   - DEVICE: SUITE FAIL -  42.86% [mheap_api]: pass = 3, fail = 4, skip = 2, total = 9 duration = 0.094 seconds
DEBUG   - DEVICE: ERROR: cannot fail in test phase 'teardown()', bailing
DEBUG   - DEVICE: - PASS - [mheap_api.test_k_aligned_alloc] duration = 0.001 seconds
DEBUG   - DEVICE: - FAIL - [mheap_api.test_malloc_in_isr] duration = 0.019 seconds
DEBUG   - DEVICE: - PASS - [mheap_api.test_malloc_in_thread] duration = 0.001 seconds
DEBUG   - DEVICE: - FAIL - [mheap_api.test_mheap_calloc] duration = 0.018 seconds
DEBUG   - DEVICE: - FAIL - [mheap_api.test_mheap_malloc_align4] duration = 0.017 seconds
DEBUG   - DEVICE: - PASS - [mheap_api.test_mheap_malloc_free] duration = 0.001 seconds
DEBUG   - DEVICE: - FAIL - [mheap_api.test_mheap_threadsafe] duration = 0.037 seconds
DEBUG   - DEVICE: - SKIP - [mheap_api.test_multi_heap] duration = 0.000 seconds
DEBUG   - DEVICE: - SKIP - [mheap_api.test_sys_heap_mem_pool_assign] duration = 0.000 seconds
DEBUG   - DEVICE: 
DEBUG   - DEVICE: ------ TESTSUITE SUMMARY END ------
DEBUG   - DEVICE: 
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: RunID: 497da24b1196b92df88dba66c644470e
DEBUG   - DEVICE: PROJECT EXECUTION FAILED
```
**Environment:**
- OS: Linux
- Toolchain: zephyr-sdk-0.16.3
- zephyr-v3.5.0-2629-gd70e8d5a8f91

jhedberg, 2023-12-05:
@KamilxPaszkiet this is just a consequence of the app explicitly setting the following in its `prj.conf`:
```
CONFIG_HEAP_MEM_POOL_SIZE=256
```

There's just no way that's going to work with any platform that uses ACPI, since ACPI requires a heap size of at least a few hundred kilobytes, or in the case of `up_squared` roughly a megabyte. So this is really a bug with the application overriding the heap size, not something with `up_squared`. Since there's already an issue tracking this (#65722) and a PR where possible fixes are being discussed (#65908), I don't think it makes sense to open these separate issues. Do you agree?

jhedberg, 2023-12-05:
@KamilxPaszkiet feel free to reopen if this has some other root cause (e.g. try the test by removing that line from `prj.conf` - if the issue still appears then its about something else than the heap size)