zephyrproject-rtos/zephyr #66154: FRDM_KW41Z seems to crash early on any samples

tbursztyka, 2023-12-05:
**Describe the bug**

Sure, this is an old board, but it is still meaningful for testing since it embeds quite a few devices.
Anyway, build hello_world sample and run it on the board. Nothing will shop up. In-situ debugging shows a crash.

Unless my board is dead... (whoever proves me this is the case will close this issue then)

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b frdm_kw41z samples/hello_world
2. west flash

Or, to actually "see" the crash:

1. west build -b frdm_kw41z samples/hello_world
2. west debug

```
Start address 0x000009d0, load size 18032
Transfer rate: 3 KB/sec, 1060 bytes/write.
(gdb) step
z_arm_platform_init () at /home/tburszty/Projects/zephyrproject/zephyr/soc/arm/nxp_kinetis/kwx/soc_kw4xz.c:94
94		SystemInit();
(gdb) continue
Continuing.

Program received signal SIGSEGV, Segmentation fault.
z_arm_hard_fault () at /home/tburszty/Projects/zephyrproject/zephyr/arch/arm/core/cortex_m/fault_s.S:80
80		mrs r0, MSP

```

**Expected behavior**

Console should output the hello world message, no crash should happen.

**Impact**

Board is useless as it is.

dleach02, 2024-01-16:
@hakehuang does this still occur with the recent ram size changes that merged?

hakehuang, 2024-01-17:
> @hakehuang does this still occur with the recent ram size changes that merged?

it has been fixed. it is ok in my weekly testing

dleach02, 2024-01-26:
@tbursztyka, can you confirm if this is fixed on your side? If so, please close the ticket.

dleach02, 2024-02-02:
issue should be closed.

closed 2024-02-02 by dleach02