zephyrproject-rtos/zephyr #66177: PPP interface LCP negotiation failure

RobMeades, 2023-12-05:
[I tried raising this issue](https://discord.com/channels/720317445772017664/733037635194585148/1179480111993602118) in the `networking` sub-group for zephyr on Discord but did not get a response, hence raising it here instead.

This is with an nRF5340DK board using nRFConnect SDK (2.5.0)/Zephyr 3.4.99ncs: we are tied to the nRFConnect Zephyr versions but I think this should be OK, PPP-wise, as the only change in the PPP code since Zephyr 3.4.99ncs is to remove the "experimental" flag.

To be clear, this issue is with PPP, not with the modem stuff; we are not using that, we are JUST using PPP to a UART, on the other side of which a peer PPP entity is running.

THE PROBLEM

PPP will not pass LCP negotiation with a PPP-peer that I know to be well behaved (a production that passes LCP negotiation v LWIP's PPP interface).

I've [attached a full log](https://github.com/zephyrproject-rtos/zephyr/files/13562183/ppp_lcp.txt) with the debug output from Zephyr PPP but, in summary, the Zephyr-side PPP sends an LCP Configure-Request (`c0 21 01 01 00 04`) and the debug prints shows that the Zephyr-side PPP receives back an LCP Configure-Acknowledge for this from the device (`ff 03 c0 21 02 01 00 04 1c 90`), which looks well formed to me (given the Configure-Request), but this response is ignored and Zephyr-side PPP repeats the LCP Configure-Request.  I know that Zephyr-side PPP really is receiving/processing packets (i.e. it is not some weird data-routing or FCS issue) because the device sends its LCP Configure-Request a little while later and Zephyr-side PPP responds to that with a correctly-formatted Configure-Reject.

I am building with the following `KConfig` flags:

```
CONFIG_NETWORKING=y
CONFIG_NET_DRIVERS=y
CONFIG_PPP_NET_IF_NO_AUTO_START=y
CONFIG_NET_PPP=y
CONFIG_NET_L2_PPP=y
CONFIG_NET_PPP_ASYNC_UART=y
CONFIG_UART_ASYNC_API=y
CONFIG_NET_PPP_UART_BUF_LEN=512
CONFIG_NET_PPP_ASYNC_UART_TX_BUF_LEN=512
```
...and my code is currently not much more than:

```C
pNetIf = net_if_get_first_by_type(&NET_L2_GET_NAME(PPP));
if (pNetIf != NULL) {
    net_if_carrier_on(pNetIf);
    if (net_if_up(pNetIf) == 0) {
        ...
```

...after which I just wait for a long time (no events hooked in or the like),

jukkar, 2023-12-05:
I suppose this is not easy to reproduce as it would require your hw etc.
Do you any log file with PPP L2 debugging enabled `CONFIG_NET_L2_PPP_LOG_LEVEL_DBG=y`

RobMeades, 2023-12-05:
One additional comment: I may be wrong on this but I have tried hand-decoding the `c0 21 01 01 00 04` LCP frame that Zephyr PPP is sending and have not been able to understand it properly.

The octet `0x04`, I believe, means Quality-Protocol (from [page 40 of RFC1661](https://datatracker.ietf.org/doc/html/rfc1661#page-40)), but this is only the _type_ Quality-Protocol, which is meant to be followed by a length etc. related to what the PPP entity wants to negotiate concerning Quality-Protocol.  Could this be an issue, i.e. it is why the response is being ignored?

As I say, I'm no PPP/LCP expert so this may not be an issue.

jukkar, 2023-12-05:
I was hoping to enable PPP_L2 debugging as it might able to immediately tell why the message is rejected. Decoding the bit stream manually is quite painful.

You are probably just missing some Kconfig dependecy config option. It is possible to see what you need to enable by
`ninja menuconfig`
and going where that symbol is located and pressing `?` (which means help).

jukkar, 2023-12-05:
Do you have `CONFIG_NET_LOG=y` in your prj.conf, it is required for the debug logging.

RobMeades, 2023-12-05:
Aha, yes, that was it, log with L2 in it attached.

[lcp_ppp_l2.txt](https://github.com/zephyrproject-rtos/zephyr/files/13564324/lcp_ppp_l2.txt)

RobMeades, 2023-12-05:
Missed a bit out at the start of time (trying to disentangle the Zephyr debug from mine):

```
16:49:18,548 D: [0x20001a90] dev 0x91984
16:49:18,558 D: [0x20001a90] iface 0x20030684
16:49:18,569 D: (main): Initializing PPP L2 0x200018a0 for iface 0x20030684
16:49:18,580 D: (main): proto IPV6CP (0x8057) fsm 0x200019a8
16:49:18,591 D: (main): proto LCP (0xc021) fsm 0x200018d8
16:49:18,602 *** Booting nRF Connect SDK v2.5.0 ***
16:49:24,189 D: (tx_handler_thread): PPP TX started
```

jukkar, 2023-12-05:
> FYI, noticing `IPV6CP` above and no sign of anything saying IPV4 I tried adding `CONFIG_NET_IPV4=y` to the build but that did not change the pattern or the outcome.

For IPCP to work you indeed need to enable IPv4. The LCP happens before IPCP so missing IPCP is not affecting the issue you are seeing.

jukkar, 2023-12-05:
Instructions to all interested parties who want to run zephyr qemu_x86 against pppd in Linux. This worked fine and I did not see issues. Just wondering now how this (ppp config / setup) differs from your setup.

Remove the nomp (depends on pppd version so might not be needed) from net-tools pppd script
```
zephyr/tools/net-tools$ (master *%)$ git diff
diff --git a/loop-pppd.sh b/loop-pppd.sh
index 3d39169..ce3c5a6 100755
--- a/loop-pppd.sh
+++ b/loop-pppd.sh
@@ -33,7 +33,7 @@ ctrl_c() {
 
 while [ $STOPPED -eq 0  ]; do
     pppd $PTS_DEV +ipv6 local \
-        noauth novj noccp noaccomp nodefaultroute nomp nodetach \
+        noauth novj noccp noaccomp nodefaultroute nodetach \
         debug silent 192.0.2.2:192.0.2.1
     sleep 2
 done
```
In net-tools run these scripts (two different terminals)
```
./loop-ppp-dev.sh
sudo ./loop-pppd.sh 
```
In zephyr directory do this:

Create `samples/net/sockets/echo_server/ppp-overlay.conf`
```
CONFIG_NET_PPP=y
CONFIG_NET_L2_PPP=y
CONFIG_NET_L2_PPP_LOG_LEVEL_DBG=y
CONFIG_NET_LOG=y
CONFIG_LOG=y
CONFIG_UART_PIPE=y
```
Create `samples/net/sockets/echo_server/ppp-qemu.overlay`
```
/ {
     chosen {
             zephyr,ppp-uart = &uart1;
     };
};
```

Run echo-server sample against pppd
```
west build -b qemu_x86 -p \
    -d build/ppp-echo-server samples/net/sockets/echo_server -t run -- \
    -DOVERLAY_CONFIG=ppp-overlay.conf -DCONFIG_NET_QEMU_PPP=y -DDTC_OVERLAY_FILE=ppp-qemu.overlay
```

RobMeades, 2023-12-05:
Trying to get my head around what this is, I believe you are describing something like this?

[image]
Is there a Zephyr log of this running with the same logging on as I am using?  That might show us differences

From the point of view of "configuration variables with PPP in the name" the differences are that I specify no auto-start ('cos my PPP interface is not present when PPP is initialised, PPP has to be started only when the network is brought up), that I use the asynchronous UART (which I have to use as I don't want to be given one character at a time); but that shouldn't matter as, from a PPP perspective, there is no asychronicity, it is message-out, reply-back.  And I use larger buffer sizes because this will be passing lots of user data when it is working.

```
CONFIG_PPP_NET_IF_NO_AUTO_START=y
CONFIG_NET_PPP=y
CONFIG_NET_L2_PPP=y
CONFIG_NET_PPP_ASYNC_UART=y
CONFIG_NET_PPP_UART_BUF_LEN=512
CONFIG_NET_PPP_ASYNC_UART_TX_BUF_LEN=512
```

```
CONFIG_NET_PPP=y
CONFIG_NET_L2_PPP=y
CONFIG_NET_L2_PPP_LOG_LEVEL_DBG=y
```

Difficult to see how any of these could affect the LCP phase.

In terms of `pppd` I, of course, am not using `pppd` but I note that `noauth` is specified, which my device will certainly negotiate to "on", but we haven't got that far yet so I doubt that it is relevant.

Looking at `fsm.c`, when it receives a Configure Acknowledge it does seem to call the FSM to send a Configure Request:

[image]

...which is what we are seeing, but I guess this is meant to be the _next_ Configure-Request?

RobMeades, 2024-01-02:
Any thoughts on this?  Just that PPP simply doesn't work at all, as far as I can tell.

RobMeades, 2024-01-10:
Understood: this is just connecting via PPP to a standard modem, I think the only difference might be that in my setup it so happens that Zephyr PPP gets to send the first thing, do the first LCP negotiation request, which the modem at the other end simply acks; we never seem to get past that.

It is possible, of course, that I've configured something wrong in Zephyr PPP but I can't see anything to configure at run-time, only those compile-time `CONFIG_` settings.

jukkar, 2024-01-10:
I just used similar config options that you had above. I am using my test setup using qemu and pppd in Linux.
This is the Linux output when the connection is established:
```
Connect: ppp0 <--> /dev/pts/3
rcvd [LCP ConfReq id=0x1]
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <magic 0xbd5fe2b> <pcomp>]
sent [LCP ConfAck id=0x1]
rcvd [LCP ConfRej id=0x1 <asyncmap 0x0> <magic 0xbd5fe2b> <pcomp>]
sent [LCP ConfReq id=0x2]
rcvd [LCP ConfAck id=0x2]
sent [LCP EchoReq id=0x0 magic=0x0]
sent [IPCP ConfReq id=0x1 <addr 192.0.2.2>]
sent [IPV6CP ConfReq id=0x1 <addr fe80::7446:ff2a:96c8:0a2d>]
rcvd [IPCP ConfReq id=0x1 <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
sent [IPCP ConfRej id=0x1 <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
rcvd [IPV6CP ConfReq id=0x1 <addr fe80::0000:5eff:fe00:5303>]
sent [IPV6CP ConfAck id=0x1 <addr fe80::0000:5eff:fe00:5303>]
rcvd [LCP EchoRep id=0x0 magic=0x0]
rcvd [IPCP ConfAck id=0x1 <addr 192.0.2.2>]
rcvd [IPV6CP ConfAck id=0x1 <addr fe80::7446:ff2a:96c8:0a2d>]
local  LL address fe80::7446:ff2a:96c8:0a2d
remote LL address fe80::0000:5eff:fe00:5303
Script /etc/ppp/ipv6-up started (pid 41117)
rcvd [IPCP ConfReq id=0x2 <addr 0.0.0.0>]
sent [IPCP ConfNak id=0x2 <addr 192.0.2.1>]
Script /etc/ppp/ipv6-up finished (pid 41117), status = 0x0
rcvd [IPCP ConfReq id=0x3 <addr 192.0.2.1>]
sent [IPCP ConfAck id=0x3 <addr 192.0.2.1>]
Script /etc/ppp/ip-pre-up started (pid 41122)
Script /etc/ppp/ip-pre-up finished (pid 41122), status = 0x0
local  IP address 192.0.2.2
remote IP address 192.0.2.1
Script /etc/ppp/ip-up started (pid 41129)
Script /etc/ppp/ip-up finished (pid 41129), status = 0x0
```
So Zephyr sends first ConfReq, and pppd replies ConfReq and ConfAck, and the connection establishment continues from there successfully.
Now if you say that in your case the modem does not send ConfReq to zephyr, just wondering if this is the issue and zephyr just waits for that before continuing.

jukkar, 2024-01-10:
> Also, if I've read your post correctly, the first thing that happens for you is `rcvd [LCP ConfReq id=0x1]`, which is different to my case where it is Zephyr doing the first TX

In my test Zephyr sends the first packet too, the log is from the Linux (modem) side in this case and there pppd just waits data and the first packet is from zephyr LCP ConfReq

jukkar, 2024-01-10:
> Looking at `fsm.c`, when it receives a Configure Acknowledge it does seem to call the FSM to send a Configure Request:

It might be that we are not in the proper state `PPP_ACK_RECEIVED` when we receive the ACK.

jukkar, 2024-01-10:
Would you be able to provide PPP debug log about this? So I would be interested in seeing what zephyr outputs when you have `CONFIG_NET_L2_PPP_LOG_LEVEL_DBG=y` enabled.

RobMeades, 2024-01-10:
I think that is already in the attachment? https://github.com/zephyrproject-rtos/zephyr/files/13564324/lcp_ppp_l2.txt

jukkar, 2024-01-10:
According to the log
```
16:50:47,366 D: (rx_q[0]): [LCP/0x200018d8] LCP Configure-Ack (2) id 1 payload len 0
16:50:47,377 D: (rx_q[0]): [LCP/0x200018d8] Current state REQUEST_SENT (6)
16:50:47,388 D: (rx_q[0]): [LCP/0x200018d8] state REQUEST_SENT (6) => ACK_RECEIVED (7) (fsm_recv_configure_ack():735)
```
we receive the ACK and accept it (because we go to ACK_RECEIVED state)
```
16:50:47,729 D: (sysworkq): [LCP/0x200018d8] Current state ACK_RECEIVED (7)
16:50:47,740 D: (sysworkq): [LCP/0x200018d8] Sending Configure-Req (1) id 1 to peer while in ACK_RECEIVED (7)
16:50:47,751 D: (sysworkq): [LCP/0x200018d8] Sending 6 bytes pkt 0x2002aea0 (options len 0)
16:50:47,762 D: (sysworkq): [LCP/0x200018d8] state ACK_RECEIVED (7) => REQUEST_SENT (6) (ppp_fsm_timeout():127)
16:50:47,773 D: send L2
```
but then we have a timeout as we expect to receive Configure-Req that never comes. 

According to the RFC 1661 chapter 3.4 https://www.rfc-editor.org/rfc/rfc1661#section-3.4:
```
The Link Control Protocol (LCP) is used to establish the connection
through an exchange of Configure packets.  This exchange is complete,
and the LCP Opened state entered, once a Configure-Ack packet
(described later) has been both sent and received.
```
So both parties need to negotiate the options via the ConfReq and ConfAck packets, and that is not happening here.
While looking at your log file, I am wondering have you shortened the `CONFIG_NET_L2_PPP_TIMEOUT` from the default 3 sec, as the timeout happens very fast.

RobMeades, 2024-01-10:
> While looking at your log file, I am wondering have you shortened the CONFIG_NET_L2_PPP_TIMEOUT from the default 3 sec, as the timeout happens very fast.

No, `CONFIG_NET_L2_PPP_TIMEOUT` does not occur on a search of all of the files in my codebase (just to be certain).  You will see in the log that a configure request does arrive at Zephyr from the modem: it is sent at about the same time that the re-send of the configure request from Zephyr goes out:

```
// Configure ack from modem for initial configure request sent by Zephyr
16:50:47,334 D: recv L2
16:50:47,355 D: c0 21 02 01 00 04
16:50:47,388 D: (rx_q[0]): [LCP/0x200018d8] state REQUEST_SENT (6) => ACK_RECEIVED (7) (fsm_recv_configure_ack():735)

// Re-send of configure request from Zephyr
16:50:47,740 D: (sysworkq): [LCP/0x200018d8] Sending Configure-Req (1) id 1 to peer while in ACK_RECEIVED (7)
16:50:47,805 D: send ppp
16:50:47,826 D: c0 21 01 01 00 04

// Configure request from arrives from modem
16:50:48,097 D: Received data 51 bytes
16:50:48,227 D: recv L2
16:50:48,249 D: c0 21 01 01 00 16 02 06
16:50:48,260 D: 00 00 00 00 01 04 05 dc
16:50:48,271 D: 03 04 c0 23 08 02 07 02
16:50:48,282 D: (rx_q[0]): [LCP/0x200018d8] LCP Configure-Req (1) id 1 payload len 18
```

So it does seem likely that this is a timing thing: somehow the code is expecting a configure request to arrive from the modem in the same packet as the configure ack, whereas in this case it arrives in a separate packet.

jukkar, 2024-01-10:
Could you try to set for example `CONFIG_NET_L2_PPP_TIMEOUT=10000` (10 sec). It is really weird that the timeout triggers so fast in your log.

jukkar, 2024-01-10:
> So it does seem likely that this is a timing thing: somehow the code is expecting a configure request to arrive from the modem in the same packet as the configure ack, whereas in this case it arrives in a separate packet.

No, there is no expectation like that. By default there should be 3sec timeout but for some reason the timeout happens almost immediately.

RobMeades, 2024-01-10:
> Could you try to set for example CONFIG_NET_L2_PPP_TIMEOUT=10000 (10 sec). It is really weird that the timeout triggers so fast in your log.

OK, will try that and update.

RobMeades, 2024-01-10:
Nicely done sir, that has certainly moved things forward: the retransmission of the initial configure request no longer occurs!  I searched through my Zephyr codebase for `CONFIG_NET_L2_PPP_TIMEOUT` and I see that quite a lot of the samples set it to 10000, so it seems fair enough that I do so also.

Zephyr is now sending a configure reject in response to the modem's configure request: can you tell what it is objecting to, what I need to switch-on in my Zephyr build to make it accept the configure request?

```
16:01:42,319 D: [0x20001b68] dev 0x943ac
16:01:42,330 D: [0x20001b68] iface 0x20030910
16:01:42,341 D: (main): Initializing PPP L2 0x200018a0 for iface 0x20030910
16:01:42,352 D: (main): proto IPCP (0x8021) fsm 0x200019a8
16:01:42,363 D: (main): proto IPV6CP (0x8057) fsm 0x20001a80
16:01:42,374 D: (main): proto LCP (0xc021) fsm 0x200018d8
16:01:49,800 D: (tx_handler_thread): PPP TX started
16:02:28,522 I: Initializing PPP to use uart-ppp@8000
16:02:28,545 D: RX enabled
16:02:28,556 D: (net_mgmt): [0x200018a0] phase DEAD (0) => ESTABLISH (1) (ppp_open():204)
16:02:28,568 D: (net_mgmt): Starting LCP
16:02:28,579 D: (net_mgmt): [LCP/0x200018d8] Current state INITIAL (0)
16:02:28,590 D: (net_mgmt): [LCP/0x200018d8] state INITIAL (0) => CLOSED (2) (ppp_fsm_lower_up():312)
16:02:28,602 D: (net_mgmt): [LCP/0x200018d8] Current state CLOSED (2)
16:02:28,613 D: (net_mgmt): [LCP/0x200018d8] state CLOSED (2) => REQUEST_SENT (6) (ppp_fsm_open():335)
16:02:28,625 D: (net_mgmt): [LCP/0x200018d8] Sending Configure-Req (1) id 1 to peer while in REQUEST_SENT (6)
16:02:28,636 D: (net_mgmt): [LCP/0x200018d8] Sending 6 bytes pkt 0x2002b120 (options len 0)
16:02:28,648 D: send L2
16:02:28,659 D: 0x2002b120
16:02:28,670 D: c0 21 01 01 00 04       |.!....
16:02:28,682 D: send ppp
16:02:28,693 D: 0x2002b120
16:02:28,704 D: c0 21 01 01 00 04       |.!....
16:02:28,741 D: UART_TX_DONE: sent 17 bytes
16:02:29,340 D: Received data 18 bytes
16:02:29,351 D: Sync byte (0x7e) start
16:02:29,362 D: (ppp_workq): [0x20001b68] state START (0) => ADDRESS (1)
16:02:29,374 D: Address byte (0xff) start
16:02:29,385 D: (ppp_workq): [0x20001b68] state ADDRESS (1) => DATA (2)
16:02:29,398 D: End of pkt (0x7e)
16:02:29,410 D: (ppp_workq): [0x20001b68] state DATA (2) => ADDRESS (1)
16:02:29,421 D: recv ppp
16:02:29,432 D: 0x2002b020
16:02:29,443 D: ff 03 c0 21 02 01 00 04 |...!....
16:02:29,455 D: 1c 90                   |..
16:02:29,466 D: Ringbuf 0x200021bc is empty!
16:02:29,478 D: recv L2
16:02:29,489 D: 0x2002b020
16:02:29,500 D: c0 21 02 01 00 04       |.!....
16:02:29,512 D: (rx_q[0]): [LCP/0x200018d8] LCP Configure-Ack (2) id 1 payload len 0
16:02:29,523 D: (rx_q[0]): [LCP/0x200018d8] Current state REQUEST_SENT (6)
16:02:29,536 D: (rx_q[0]): [LCP/0x200018d8] state REQUEST_SENT (6) => ACK_RECEIVED (7) (fsm_recv_configure_ack():735)
16:02:30,147 D: Received data 51 bytes
16:02:30,158 D: Address byte (0xff) start
16:02:30,170 D: (ppp_workq): [0x20001b68] state ADDRESS (1) => DATA (2)
16:02:30,181 D: End of pkt (0x7e)
16:02:30,192 D: (ppp_workq): [0x20001b68] state DATA (2) => ADDRESS (1)
16:02:30,204 D: recv ppp
16:02:30,215 D: 0x2002b020
16:02:30,226 D: ff 03 c0 21 01 01 00 16 |...!....
16:02:30,237 D: 02 06 00 00 00 00 01 04 |........
16:02:30,249 D: 05 dc 03 04 c0 23 08 02 |.....#..
16:02:30,260 D: 07 02 1c d4             |....
16:02:30,272 D: Ringbuf 0x200021bc is empty!
16:02:30,283 D: recv L2
16:02:30,294 D: 0x2002b020
16:02:30,305 D: c0 21 01 01 00 16 02 06 |.!......
16:02:30,317 D: 00 00 00 00 01 04 05 dc |........
16:02:30,328 D: 03 04 c0 23 08 02 07 02 |...#....
16:02:30,339 D: (rx_q[0]): [LCP/0x200018d8] LCP Configure-Req (1) id 1 payload len 18
16:02:30,351 D: (rx_q[0]): [LCP/0x200018d8] Current state ACK_RECEIVED (7)
16:02:30,362 D: (rx_q[0]): [LCP/0x200018d8] Sending Configure-Rej (4) id 1 to peer while in ACK_RECEIVED (7)
16:02:30,374 D: (rx_q[0]): [LCP/0x200018d8] Sending 24 bytes pkt 0x2002b120 (options len 18)
16:02:30,386 D: send L2
16:02:30,397 D: 0x2002b120
[... 50 lines cut ...]
16:02:37,008 D: UART_TX_DONE: sent 17 bytes
16:02:46,954 D: (sysworkq): [LCP/0x200018d8] Current state REQUEST_SENT (6)
16:02:46,965 D: (sysworkq): [LCP/0x200018d8] Sending Configure-Req (1) id 1 to peer while in REQUEST_SENT (6)
16:02:46,975 D: (sysworkq): [LCP/0x200018d8] Sending 6 bytes pkt 0x2002b120 (options len 0)
16:02:46,986 D: send L2
16:02:46,997 D: 0x2002b120
16:02:47,008 D: c0 21 01 01 00 04       |.!....
16:02:47,018 D: send ppp
16:02:47,029 D: 0x2002b120
16:02:47,040 D: c0 21 01 01 00 04       |.!....
16:02:47,050 D: UART_TX_DONE: sent 17 bytes
16:02:56,897 D: (sysworkq): [LCP/0x200018d8] Current state REQUEST_SENT (6)
16:02:56,910 D: (sysworkq): [LCP/0x200018d8] Sending Configure-Req (1) id 1 to peer while in REQUEST_SENT (6)
16:02:56,923 D: (sysworkq): [LCP/0x200018d8] Sending 6 bytes pkt 0x2002b120 (options len 0)
16:02:56,934 D: send L2
16:02:56,946 D: 0x2002b120
16:02:56,957 D: c0 21 01 01 00 04       |.!....
16:02:56,969 D: send ppp
16:02:56,980 D: 0x2002b120
16:02:56,991 D: c0 21 01 01 00 04       |.!....
16:02:57,003 D: UART_TX_DONE: sent 17 bytes
16:02:57,357 D: (net_mgmt): [0x200018a0] phase ESTABLISH (1) => TERMINATE (5) (lcp_close():166)
16:02:57,369 D: (net_mgmt): [LCP/0x200018d8] Current state REQUEST_SENT (6)
16:02:57,381 D: (net_mgmt): [LCP/0x200018d8] state REQUEST_SENT (6) => CLOSING (4) (terminate():215)
```

jukkar, 2024-01-10:
> Zephyr is now sending a configure reject in response to the modem's configure request: can you tell what it is objecting to, what I need to switch-on in my Zephyr build to make it accept the configure request?
> 

Unfortunately I have forgotten how to read binary PPP frames, so cannot help you directly. You could add some more debugging prints to `subsys/net/l2/ppp/options.c` to print out the options being rejected. Perhaps it might make sense to prepare a PR for that if you have time, it would be useful information in the future.

RobMeades, 2024-01-10:
I think it's `CONFIG_NET_L2_PPP_PAP`, trying that now...

RobMeades, 2024-01-10:
Adding `CONFIG_NET_L2_PPP_PAP` has taken things further but we seem to now be stuck at PAP authentication (log attached, [lcp_ppp_auth.txt](https://github.com/zephyrproject-rtos/zephyr/files/13891353/lcp_ppp_auth.txt)), the relevant part being the following (also pasted-in below):

- the modem sends a request for PAP authentication (`c0 21 01 02 00 08 03 04 c0 23`),
- Zephyr receives and understands this and sends back a configure ack for it (`c0 21 02 02 00 08 03 04 c0 23`),
- Zephyr follows this with a PAP authentication request including the "blank"/"blank" username/password as hard-coded in `pap.c` (`c0 23 01 03 00 10 05 62 6c 61 6e 6b 05 62 6c 61 6e 6b`),
- a second later the modem asks for PAP authentication again,
- and repeat.

I don't think the modem is checking the username/password it receives but in any case I tried removing them by editing `pap.c` to give zero lengths for both parameters (which is what LWIP does, see below); this did not help.

FYI, if I capture up to this far in the PPP negotiation between the same modem and LWIP (manually de-escaped by me) I get the following:

```
21:56:38 LWIP sent:     c0 21 01 01 00 04 02 06 00 00 00 00 05 06 08 3b 50 54 07 02 08 02
21:56:38 LWIP received: c0 21 02 01 00 04 02 06 00 00 00 00 05 06 08 3b 50 54 07 02 08 02
21:56:41 LWIP sent:     c0 21 02 01 00 06 02 06 00 00 00 00 01 04 05 dc 03 04 c0 23 08 02 07 02
21:56:41 LWIP sent:     c0 23 01 01 00 06 00 00 (zero length username and password)
21:56:41 LWIP received: c0 23 02 01 00 16 11 4c 6f 67 69 6e 20 73 75 63 63 65 73 73 66 75 6c 2e ("Login successful.")
```

Any suggestions as to where to go next with this?

```
17:14:27,624 D: Address byte (0xff) start
17:14:27,635 D: (ppp_workq): [0x20001c18] state ADDRESS (1) => DATA (2)
17:14:27,647 D: End of pkt (0x7e)
17:14:27,659 D: (ppp_workq): [0x20001c18] state DATA (2) => ADDRESS (1)
17:14:27,671 D: recv ppp
17:14:27,683 D: 0x2002b080
17:14:27,696 D: ff 03 c0 21 01 02 00 08 |...!....
17:14:27,707 D: 03 04 c0 23 23 eb       |...##.
17:14:27,718 D: Ringbuf 0x2000226c is empty!
17:14:27,729 D: recv L2
17:14:27,740 D: 0x2002b080
17:14:27,751 D: c0 21 01 02 00 08 03 04 |.!......
17:14:27,762 D: c0 23                   |.#
17:14:27,773 D: (rx_q[0]): [LCP/0x200018d8] LCP Configure-Req (1) id 2 payload len 4
17:14:27,784 D: (rx_q[0]): [LCP/0x200018d8] Current state ACK_RECEIVED (7)
17:14:27,796 D: (rx_q[0]): [LCP/0x200018d8] option AUTH_PROTO (3) len 4
17:14:27,807 D: (rx_q[0]): [LCP/0x200018d8] Check option AUTH_PROTO (3) len 2
17:14:27,818 D: (rx_q[0]): [LCP/0x200018d8] option AUTH_PROTO (3) len 4
17:14:27,829 D: (rx_q[0]): [LCP] Received auth protocol c023 (PAP)
17:14:27,840 D: (rx_q[0]): Authentication protocol negotiated: c023 (PAP)
17:14:27,851 D: (rx_q[0]): [LCP/0x200018d8] Sending Configure-Ack (2) id 2 to peer while in ACK_RECEIVED (7)
17:14:27,862 D: (rx_q[0]): [LCP/0x200018d8] Sending 10 bytes pkt 0x2002b180 (options len 4)
17:14:27,873 D: (rx_q[0]): [LCP/0x200018d8] state ACK_RECEIVED (7) => OPENED (9) (fsm_recv_configure_req():653)
17:14:27,884 D: (rx_q[0]): [0x200018a0] Link established
17:14:27,895 D: (rx_q[0]): [0x200018a0] phase ESTABLISH (1) => AUTH (2) (do_auth():74)
17:14:27,906 D: (rx_q[0]): [PAP/0x20001b48] Current state STARTING (1)
17:14:27,917 D: (rx_q[0]): [PAP/0x20001b48] Invalid state STARTING (1)
17:14:27,928 D: (rx_q[0]): [IPCP/0x200019a8] Current state INITIAL (0)
17:14:27,939 D: (rx_q[0]): [IPCP/0x200019a8] state INITIAL (0) => CLOSED (2) (ppp_fsm_lower_up():312)
17:14:27,950 D: (rx_q[0]): [IPV6CP/0x20001a80] Current state INITIAL (0)
17:14:27,960 D: (rx_q[0]): [IPV6CP/0x20001a80] state INITIAL (0) => CLOSED (2) (ppp_fsm_lower_up():312)
17:14:27,971 D: (rx_q[0]): [PAP/0x20001b48] Current state STARTING (1)
17:14:27,982 D: (rx_q[0]): [PAP/0x20001b48] Sending Configure-Req (1) id 3 to peer while in STARTING (1)
17:14:27,993 D: (rx_q[0]): [PAP/0x20001b48] Sending 18 bytes pkt 0x2002b1c0 (options len 12)
17:14:28,004 D: (rx_q[0]): [PAP/0x20001b48] state STARTING (1) => REQUEST_SENT (6) (ppp_fsm_lower_up():317)
17:14:28,015 D: send L2
17:14:28,025 D: 0x2002b180
17:14:28,036 D: c0 21 02 02 00 08       |.!....
17:14:28,047 D: 0x2002b180
17:14:28,058 D: 03 04 c0 23             |...#
17:14:28,068 D: send ppp
17:14:28,080 D: 0x2002b180
17:14:28,090 D: c0 21 02 02 00 08       |.!....
17:14:28,101 D: 0x2002b180
17:14:28,113 D: 03 04 c0 23             |...#
17:14:28,123 D: send L2
17:14:28,135 D: 0x2002b1c0
17:14:28,145 D: c0 23 01 03 00 10       |.#....
17:14:28,156 D: 0x2002b1c0
17:14:28,167 D: 05 62 6c 61 6e 6b 05 62 |.blank.b
17:14:28,178 D: 6c 61 6e 6b             |lank
17:14:28,189 D: send ppp
17:14:28,200 D: 0x2002b1c0
17:14:28,211 D: c0 23 01 03 00 10       |.#....
17:14:28,222 D: 0x2002b1c0
17:14:28,233 D: 05 62 6c 61 6e 6b 05 62 |.blank.b
17:14:28,243 D: 6c 61 6e 6b             |lank
17:14:28,254 D: UART_TX_DONE: sent 23 bytes
17:14:28,363 D: UART_TX_DONE: sent 31 bytes

17:14:29,866 D: Received data 23 bytes
17:14:29,877 D: Address byte (0xff) start
17:14:29,888 D: (ppp_workq): [0x20001c18] state ADDRESS (1) => DATA (2)
17:14:29,899 D: End of pkt (0x7e)
17:14:29,909 D: (ppp_workq): [0x20001c18] state DATA (2) => ADDRESS (1)
17:14:29,920 D: recv ppp
17:14:29,931 D: 0x2002b080
17:14:29,942 D: ff 03 c0 21 01 03 00 08 |...!....
17:14:29,952 D: 03 04 c0 23 f6 74       |...#.t
17:14:29,963 D: Ringbuf 0x2000226c is empty!
17:14:29,974 D: recv L2
17:14:29,985 D: 0x2002b080
17:14:29,995 D: c0 21 01 03 00 08 03 04 |.!......
17:14:30,006 D: c0 23                   |.#
17:14:30,017 D: (rx_q[0]): [LCP/0x200018d8] LCP Configure-Req (1) id 3 payload len 4
...and repeat
```

jukkar, 2024-01-11:
> Any suggestions as to where to go next with this?

Hah, that is a good question :scream:

As you noticed the issue is in PAP handshake. The `pap.c` function `pap_config_info_ack()` has a comment `We only support one way negotiation for now, so move to ACK_SENT phase right away.` Perhaps you could investigate that and add the missing code?

jukkar, 2024-01-11:
If you are able to fix this PAP handshaking issue and submit it to upstream, then the fix will propagate itself to NCS too, I would not be too worried about it.

RobMeades, 2024-01-11:
FYI, I put some prints in `ppp.c` to see which of the requested configuration options turn out to be supported and which don't (see log below): it would seem that _only_ setting the authentication protocol is supported, nothing else is.

Am I missing some configuration steps somewhere?  I've no idea what half of this stuff is, I'd just quite like to connect a standard modem to Zephyr via PPP.  How do people usually do that?

```
D: (rx_q[0]): [LCP/0x200018d8] LCP Configure-Req (1) id 3 payload len 18
D: (rx_q[0]): [LCP/0x200018d8] Current state ACK_RECEIVED (7)
D: (rx_q[0]): [LCP/0x200018d8] option ASYNC_CTRL_CHAR_MAP (2) len 6
D: (rx_q[0]): Looking for option 2 in set of size 1
D: (rx_q[0]): [LCP/0x200018d8] Check option ASYNC_CTRL_CHAR_MAP (2) len 4
D: (rx_q[0]): [LCP/0x200018d8] ASYNC_CTRL_CHAR_MAP not supported
D: (rx_q[0]): [LCP/0x200018d8] option MRU (1) len 4
D: (rx_q[0]): Looking for option 1 in set of size 1
D: (rx_q[0]): [LCP/0x200018d8] Check option MRU (1) len 2
D: (rx_q[0]): [LCP/0x200018d8] MRU not supported
D: (rx_q[0]): [LCP/0x200018d8] option AUTH_PROTO (3) len 4
D: (rx_q[0]): Looking for option 3 in set of size 1
D: (rx_q[0]): Found it
D: (rx_q[0]): [LCP/0x200018d8] Check option AUTH_PROTO (3) len 2
D: (rx_q[0]): [LCP/0x200018d8] option ADDR_CTRL_COMPRESS (8) len 2
D: (rx_q[0]): Looking for option 8 in set of size 1
D: (rx_q[0]): [LCP/0x200018d8] Check option ADDR_CTRL_COMPRESS (8) len 0
D: (rx_q[0]): [LCP/0x200018d8] ADDR_CTRL_COMPRESS not supported
D: (rx_q[0]): [LCP/0x200018d8] option PROTO_COMPRESS (7) len 2
D: (rx_q[0]): Looking for option 7 in set of size 1
D: (rx_q[0]): [LCP/0x200018d8] Check option PROTO_COMPRESS (7) len 0
D: (rx_q[0]): [LCP/0x200018d8] PROTO_COMPRESS not supported
D: (rx_q[0]): [LCP/0x200018d8] Sending Configure-Rej (4) id 3 to peer while in ACK_RECEIVED (7)
D: (rx_q[0]): [LCP/0x200018d8] Sending 20 bytes pkt 0x2002b1c0 (options len 14)
```

jukkar, 2024-01-12:
I managed to test the pap with Linux pppd and everything was working fine with upstream zephyr.
Changes compared to my earlier instructions somewhere above:
```
diff --git a/loop-pppd.sh b/loop-pppd.sh
index 3d39169..6fae209 100755
--- a/loop-pppd.sh
+++ b/loop-pppd.sh
@@ -33,7 +33,7 @@ ctrl_c() {
 
 while [ $STOPPED -eq 0  ]; do
     pppd $PTS_DEV +ipv6 local \
-        noauth novj noccp noaccomp nodefaultroute nomp nodetach \
+        +pap novj noccp noaccomp nodefaultroute nodetach \
         debug silent 192.0.2.2:192.0.2.1
     sleep 2
 done
```

and then adding the secret to `/etc/ppp/pap-secrets`
```
*   *  blank  *
```

The Linux pppd then tells this:
```
Using interface ppp0
Connect: ppp0 <--> /dev/pts/3
rcvd [LCP ConfReq id=0x1]
sent [LCP ConfReq id=0x1 <asyncmap 0x0> <auth pap> <magic 0xb0dcb4ea> <pcomp>]
sent [LCP ConfAck id=0x1]
rcvd [LCP ConfRej id=0x1 <asyncmap 0x0> <magic 0xb0dcb4ea> <pcomp>]
sent [LCP ConfReq id=0x2 <auth pap>]
rcvd [LCP ConfAck id=0x2 <auth pap>]
sent [LCP EchoReq id=0x0 magic=0x0]
rcvd [PAP AuthReq id=0x1 user="blank" password=<hidden>]
sent [PAP AuthAck id=0x1 "Login ok"]
PAP peer authentication succeeded for blank
sent [IPCP ConfReq id=0x1 <addr 192.0.2.2>]
sent [IPV6CP ConfReq id=0x1 <addr fe80::c9af:1e91:ed32:8ff3>]
rcvd [LCP EchoRep id=0x0 magic=0x0]
rcvd [IPCP ConfReq id=0x1 <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
sent [IPCP ConfRej id=0x1 <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
rcvd [IPV6CP ConfReq id=0x1 <addr fe80::0000:5eff:fe00:53ca>]
sent [IPV6CP ConfAck id=0x1 <addr fe80::0000:5eff:fe00:53ca>]
rcvd [IPCP ConfAck id=0x1 <addr 192.0.2.2>]
rcvd [IPV6CP ConfAck id=0x1 <addr fe80::c9af:1e91:ed32:8ff3>]
local  LL address fe80::c9af:1e91:ed32:8ff3
remote LL address fe80::0000:5eff:fe00:53ca
Script /etc/ppp/ipv6-up started (pid 41424)
rcvd [IPCP ConfReq id=0x2 <addr 0.0.0.0>]
sent [IPCP ConfNak id=0x2 <addr 192.0.2.1>]
rcvd [IPCP ConfReq id=0x3 <addr 192.0.2.1>]
sent [IPCP ConfAck id=0x3 <addr 192.0.2.1>]
Script /etc/ppp/ip-pre-up started (pid 41427)
Script /etc/ppp/ip-pre-up finished (pid 41427), status = 0x0
local  IP address 192.0.2.2
remote IP address 192.0.2.1
Script /etc/ppp/ip-up started (pid 41433)
Script /etc/ppp/ipv6-up finished (pid 41424), status = 0x0
Script /etc/ppp/ip-up finished (pid 41433), status = 0x0
Modem hangup
```

jukkar, 2024-01-12:
> I had assumed that `net_if_up()` returning [zero] meant that the network interface was up but it seems that's not the case, I should be waiting for an event of some form?

Yes, you could wait `NET_EVENT_IF_UP` event if you want to know when the interface is up. The reason why this is not tied to the function return value is that the operation might take time and we do not want the function to block.

RobMeades, 2024-01-12:
Lots of progress, you'll be pleased to know that I may nearly be there.  The main issue was that I thought `net_if_up()` returning 0 indicated success, so actually the code was, after a short delay, shutting everything down again, which terminated the link of course.  Also, I had assumed that the TX buffers from PPP wouldn't be overwritten until a TX_DONE event had been returned but that doesn't seem to be the case, I had to introduce TX queueing.  Oh, and I needed to switch off IPV6 (`CONFIG_NET_IPV6=n`) as the modem has no IPV6 address and trying to negotiate that resulted in the link failing.

The last problem I have is that the `NET_EVENT_IF_UP` event arrives _immediately_ I start PPP, rather than when PPP negotiation has completed, you can see this in the log below.

Is there another event I should be waiting for?

```
21:58:39,421 D: [0x20001960] dev 0x90f9c
21:58:39,431 D: [0x20001960] iface 0x2002fab4
21:58:39,442 D: (main): Initializing PPP L2 0x200016b0 for iface 0x2002fab4
21:58:39,452 D: (main): proto IPCP (0x8021) fsm 0x200017b8
21:58:39,463 D: (main): proto LCP (0xc021) fsm 0x200016e8
21:58:39,473 D: (main): proto PAP (0xc023) fsm 0x20001890
21:58:39,484 *** Booting nRF Connect SDK v2.5.0 ***
21:58:47,886 D: (tx_handler_thread): PPP TX started

...other stuff happens; eventually the modem is connected to the cellular network and ready to do PPP

21:59:14,765 I: Initializing PPP to use uart-ppp@8000
21:59:14,776 D: RX enabled

21:59:14,786 #### callback() received netIfEventCallback() received event -805240830 (0xd0010002) (NET_EVENT_IF_UP would be -805240830).

21:59:14,797 D: (net_mgmt): [0x200016b0] phase DEAD (0) => ESTABLISH (1) (ppp_open():204)
21:59:14,808 D: (net_mgmt): Starting LCP
21:59:14,819 D: (net_mgmt): [LCP/0x200016e8] Current state INITIAL (0)
21:59:14,829 D: (net_mgmt): [LCP/0x200016e8] state INITIAL (0) => CLOSED (2) (ppp_fsm_lower_up():312)
21:59:14,840 D: (net_mgmt): [LCP/0x200016e8] Current state CLOSED (2)
21:59:14,851 D: (net_mgmt): [LCP/0x200016e8] state CLOSED (2) => REQUEST_SENT (6) (ppp_fsm_open():335)
21:59:14,862 D: (net_mgmt): [LCP/0x200016e8] Sending Configure-Req (1) id 1 to peer while in REQUEST_SENT (6)
21:59:14,873 D: (net_mgmt): [LCP/0x200016e8] Sending 6 bytes pkt 0x2002ac84 (options len 0)
21:59:14,883 D: send L2
21:59:14,894 D: 0x2002ac84
21:59:14,905 D: c0 21 01 01 00 04       |.!....
21:59:14,915 D: send ppp
21:59:14,926 D: 0x2002ac84
21:59:14,937 D: c0 21 01 01 00 04       |.!....

21:59:14,947 ### WAIT FOR 20 SECONDS TO LET PPP NEGOTIATION COMPLETE

21:59:14,947 D: UART_TX_DONE: sent 17 bytes
21:59:14,958 D: Received data 18 bytes
21:59:14,969 D: Sync byte (0x7e) start
21:59:14,980 D: (ppp_workq): [0x20001960] state START (0) => ADDRESS (1)
21:59:14,991 D: Address byte (0xff) start
21:59:15,001 D: (ppp_workq): [0x20001960] state ADDRESS (1) => DATA (2)
21:59:15,012 D: End of pkt (0x7e)
21:59:15,023 D: (ppp_workq): [0x20001960] state DATA (2) => ADDRESS (1)
21:59:15,034 D: recv ppp
21:59:15,044 D: 0x2002ab94
21:59:15,055 D: ff 03 c0 21 02 01 00 04 |...!....
21:59:15,066 D: 1c 90                   |..
21:59:15,077 D: Ringbuf 0x20001fb4 is empty!
21:59:15,088 D: recv L2
21:59:15,099 D: 0x2002ab94
21:59:15,109 D: c0 21 02 01 00 04       |.!....
21:59:15,120 D: (rx_q[0]): [LCP/0x200016e8] LCP Configure-Ack (2) id 1 payload len 0
21:59:15,131 D: (rx_q[0]): [LCP/0x200016e8] Current state REQUEST_SENT (6)
21:59:15,142 D: (rx_q[0]): [LCP/0x200016e8] state REQUEST_SENT (6) => ACK_RECEIVED (7) (fsm_recv_configure_ack():735)
21:59:17,966 D: Received data 51 bytes
21:59:17,978 D: Address byte (0xff) start
21:59:17,989 D: (ppp_workq): [0x20001960] state ADDRESS (1) => DATA (2)
21:59:18,000 D: End of pkt (0x7e)
21:59:18,011 D: (ppp_workq): [0x20001960] state DATA (2) => ADDRESS (1)
21:59:18,023 D: recv ppp
21:59:18,034 D: 0x2002ab94
21:59:18,045 D: ff 03 c0 21 01 01 00 16 |...!....
21:59:18,056 D: 02 06 00 00 00 00 01 04 |........
21:59:18,068 D: 05 dc 03 04 c0 23 08 02 |.....#..
21:59:18,079 D: 07 02 1c d4             |....
[... 225 lines cut ...]
21:59:20,615 D: (ppp_workq): [0x20001960] state ADDRESS (1) => DATA (2)
21:59:20,627 D: End of pkt (0x7e)
21:59:20,638 D: (ppp_workq): [0x20001960] state DATA (2) => ADDRESS (1)
21:59:20,649 D: recv ppp
21:59:20,660 D: 0x2002ab58
21:59:20,671 D: ff 03 80 21 02 02 00 16 |...!....
21:59:20,682 D: 03 06 c0 a8 c9 01 81 06 |........
21:59:20,694 D: 08 08 08 08 83 06 08 08 |........
21:59:20,705 D: 08 08 51 35             |..Q5
21:59:20,716 D: Ringbuf 0x20001fb4 is empty!
21:59:20,727 D: recv L2
21:59:20,739 D: 0x2002ab58
21:59:20,751 D: 80 21 02 02 00 16 03 06 |.!......
21:59:20,764 D: c0 a8 c9 01 81 06 08 08 |........
21:59:20,777 D: 08 08 83 06 08 08 08 08 |........
21:59:20,788 D: (rx_q[0]): [IPCP/0x200017b8] IPCP Configure-Ack (2) id 2 payload len 18
21:59:20,799 D: (rx_q[0]): [IPCP/0x200017b8] Current state ACK_SENT (8)
21:59:20,810 D: (rx_q[0]): [IPCP/0x200017b8] state ACK_SENT (8) => OPENED (9) (fsm_recv_configure_ack():713)
21:59:20,821 D: (rx_q[0]): PPP up with address 192.168.201.1
21:59:20,833 D: (rx_q[0]): [0x200016b0] phase NETWORK (3) => RUNNING (4) (ppp_network_up():22)
21:59:20,844 D: (rx_q[0]): [0x200016b0] Proto IPv4 (0x0021) up (1)
21:59:20,855 D: (rx_q[0]): [IPCP/0x200017b8] Current state OPENED (9)

21:59:34,957 ### 20 SECONDS IS UP
```

RobMeades, 2024-01-12:
Hmph.  OK, maybe I'm not done.  Waiting for the event `NET_EVENT_IPV4_ADDR_ADD` seemed to do the trick, I get that right at the end of the logged output as above and all seems good.  However, if I then call `zsock_socket(AF_INET, SOCK_STREAM, IPPROTO_IP);` to obtain a socked I get back -1 and `errno` is set to 107 `ENOTCONN`.

Is there something else I need to do to to be able to use the IP address I now have?

RobMeades, 2024-01-13:
Summarizing, for anyone who gets here, the things I had to get right in order to have my own PPP driver to a standard [cellular] modem (i.e. a driver that gets hooked-up as a `zephyr,ppp-uart`) were as follows:

- Know that, when `net_if_up()` returns suvcess, that only means that your PPP interface has _begun_ coming up, you have to monitor the event `NET_EVENT_IPV4_ADDR_ADD` to determine that it has _finished_ coming up before that connection can be used by Zephyr applications.
- If you want your driver to receive more than one byte at a time for transmission (which is what the Zephyr blocking UART interface does), you must use `CONFIG_NET_PPP_ASYNC_UART` and, when you do that, you need to buffer the TX packets you are given as they may be over-written before you return `UART_TX_DONE`.
- If your modem does not have an IPV6 address you may need to set `CONFIG_NET_IPV6=n` in order to complete IPCP negotiation.
- Remember that, currently, any username/password for PAP is hard-coded inside Zephyr `pap.c` and that CHAP authentication is not supported.

Thanks for all your help @jukkar; now to get TCP working...

closed 2024-01-13 by RobMeades