zephyrproject-rtos/zephyr #66185: npcx9m6f_evb: tests/kernel/sleep/kernel.common.timing and tests/kernel/sleep/kernel.common.timing.minimallibc/ failing

pkunieck, 2023-12-05:
**Describe the bug**
Test from tests/kernel/sleep/kernel.common.timing failing with timeout 

**To Reproduce**
Steps to reproduce the behavior:
1. cd <path/to>zephyrproject/zephyr
2. source zephyr-env.sh
3. twister --device-testing --hardware-map <path/to>/map.yaml -T tests/kernel/sleep/ -vv

**Expected behavior**
Test pass

**Impact**
annoyance

**Logs and console output**
START - test_sleep
Kernel objects initialized
Test thread started: id = 0x200c0918
Helper thread started: id = 0x200c0860
Testing normal expiration of k_sleep()

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 16.1
 - tested on the newest, 100% reproduction on the newest commits

pkunieck, 2023-12-12:
I checked and issue also reproduced on 16.4 some time ago, I edited it. We currently can't reproduce this, I'll give an update after we run more tests.

pkunieck, 2023-12-20:
We have reproduction at zephyr version: zephyr-v3.5.0-2951-gbd227e19cff9

ChiHuaL, 2023-12-22:
Hi,  
I test the version bd227e19cff9 and I cannot reproduce the issue. 
The following is my environment setting:

- OS: Linux

- Toolchain Zephyr SDK 16.1

- Version: bd227e19cff9 
- Reproduce steps:
1. cd <path/to>zephyrproject/zephyr
2.  west  build -p auto  -b npcx9m6f_evb tests/kernel/sleep
3. west flash
or 
1. cd <path/to>zephyrproject/zephyr
2. ./scripts/twister -v --device-testing  --device-serial /dev/ttyUSB0 --device-serial-baud 115200 -p npcx9m6f_evb  -T tests/kernel/sleep

Attach the test result log below:
```
*** Booting Zephyr OS build zephyr-v3.5.0-2951-gbd227e19cff9 ***
Running TESTSUITE sleep
===================================================================
START - test_sleep
Kernel objects initialized
Test thread started: id = 0x200c0918
Helper thread started: id = 0x200c0860
Testing normal expiration of k_sleep()
Testing: test thread sleep + helper thread wakeup test
Testing: test thread sleep + isr offload wakeup test
Testing: test thread sleep + main wakeup test thread
Testing kernel k_sleep()
 PASS - test_sleep in 2.032 seconds
===================================================================
START - test_sleep_forever
Kernel objects initialized
 PASS - test_sleep_forever in 0.004 seconds
===================================================================
START - test_usleep
elapsed_ms = 1000
 PASS - test_usleep in 1.002 seconds
===================================================================
TESTSUITE sleep succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [sleep]: pass = 3, fail = 0, skip = 0, total = 3 duration = 3.038 seconds
 - PASS - [sleep.test_sleep] duration = 2.032 seconds
 - PASS - [sleep.test_sleep_forever] duration = 0.004 seconds
 - PASS - [sleep.test_usleep] duration = 1.002 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL

```
Is it 100% reproduced ?
Could you share the test log for us to check?

gbarkadiusz, 2024-02-26:
@ChiHuaL 
ZephyrVersion: zephyr-sdk-0.16.5
OS: Linux
```
tests/kernel/sleep/kernel.common.timing             FAILED Timeout (device: D30CC4MH, 63.707s)
INFO    - <path_to>/twister-out/npcx9m6f_evb/tests/kernel/sleep/kernel.common.timing/handler.log
```
```
*** Booting Zephyr OS build v3.6.0-22-g074152a03db9 (delayed boot 500ms) ***
Running TESTSUITE sleep
===================================================================
START - test_sleep
```

pkunieck, 2024-03-28:
I checked in our test environment and we have about 58% of reproduction rate of these tests. I checked results in a one month time span(28.02-28.03)

pkunieck, 2024-05-08:
Can we have an update? Do we need more information?

ChiHuaL, 2024-05-09:
Hi,  
Sorry for the late reply.
I am unable to reproduce this symptom over 150 iterations with the following shell script:
```
#!/bin/bash                                                                                          
                                                                                                     
for i in {1..150}                                                                                    
do                                                                                                   
        echo "Iteration: $i"                                                                                                                                                                                                                  
        west flash                                                                                   
        sleep 6                                                                                      
done                           
```
The console log is attached [zephyr_test_kernel_sleep_npcx9m6f_evb.log](https://github.com/zephyrproject-rtos/zephyr/files/15259588/zephyr_test_kernel_sleep_npcx9m6f_evb.log).
OS: Linux

Toolchain Zephyr SDK 16.1

Version: https://github.com/zephyrproject-rtos/zephyr/commit/3d76e498e9475b67da824247d1fce665f84d0dbc

I have no idea why it happened at  your side.
Do you have any chance to change another npcx9m6f_evb for the test?

pkunieck, 2024-05-22:
Okay so I checked this on commit that was used here and also on commit i was using, and I still got reproductions. I got one different output, but issue is still visible. Can you send exact model of board you are using? Maybe there are some differences, with older version of npcx9m6f?
[test_3d76e498e947.log](https://github.com/zephyrproject-rtos/zephyr/files/15400283/test_3d76e498e947.log)
[test_cd685bdce82d.log](https://github.com/zephyrproject-rtos/zephyr/files/15400284/test_cd685bdce82d.log)

pkunieck, 2024-06-04:
So i checked this issue on older board and I don't have reproduction on it. We have 2 boards in validation, one in CI one for debug, they are newer models, and we got repro on them. This is numbers from stickers from one that we got repro on: L5533004C142-129 And numbers from chip: nuvoton NPCX997FA0B E121M0092

And these are numbers from board that I run 300 iterations and 100% pass: L5533004C815-78 And chip: NPCX997FA0B E105M0022 (there was also "R3" sticker on it and numbers were a little scratched after removal)

[test_older_npcx2.log](https://github.com/user-attachments/files/15545422/test_older_npcx2.log)
[test_older_npcx.log](https://github.com/user-attachments/files/15545423/test_older_npcx.log)

pkunieck, 2024-06-19:
Hi, can I get an update?

ChiHuaL, 2024-06-25:
> So i checked this issue on older board and I don't have reproduction on it. We have 2 boards in validation, one in CI one for debug, they are newer models, and we got repro on them. This is numbers [...]

May I know where do you get those npcx EVBs, so we might check what is the difference?

SzymonRichert, 2024-07-17:
As I understand npcx EVBs is just normal board. EVB stands for Early Validation Board, Intel is ordering for internal use.

pkunieck, 2024-10-03:
We recently connected and started testing another Nuvoton board: NPCX4M8F, and the same errors are also being reproduced on it. Can we get an update regarding the newer board?

pkunieck, 2024-10-16:
Any updates about that issue on 2 platforms?

ChiHuaL, 2024-10-25:
HI, 
Thanks for the update of the issue that can be reproduced on npcx4 EVB.
I will try to reproduce it  on the npcx4 EVB at our side.
Please note that I am busy on some urgent tasks and might be able to look this issue after couples of weeks.
Sorry for the late reponse.

ChiHuaL, 2025-02-18:
Hi,  
We have checked this issue is related to the deviation of the High Frequency Clock (`APB Clock`) and Low Frequency Clock (`LFCLK`).
Although in normal case this deviation is very small over 1000 ms time window , according the spec in extreme cases it can go up to +/- 1%   (+/- 10 ms) .   
For more information check Core, High Frequency Clock Timing paragraph in the datasheet.

**7.5.3 Clock Timing, "Core, High-Frequency Clock Timing:"**  

| Symbol |  Figure |              Description                 |  Reference Conditions  |  Min                       |  Typ    |           Max                 |  Units  |
| -------- |---------|-----------------------------------|-------------------------|---------------------|--------|-------------------------|--------|
| FCLK     |    -        |  Core clock frequency accuracy |  Set CLK frequency    |  FTYP-LFTOL%-1%|  FTYP  |  FTYP + LFTOL% + 1%|            |

SzymonRichert, 2025-03-20:
@ChiHuaL so please do so. Thank you.

ChiHuaL, 2025-03-21:
@SzymonRichert  Thanks for the suggestion.   
What is the convention to remove a board from the test?  
Is adding the board to `platform_exclude:` in the `testcase.yaml` a correct way?

SzymonRichert, 2025-03-26:
I would just go with platform exclude and wait for the reviewers to decide.

PR #87878 (2025-04-04): tests: kernel/sleep: exclude npcx platforms from the test
According the specification, in extreme cases, the deviation of the APB clock and LFCLK clock can reach up to +/-1% (+/- 10ms). Therefore, exclude npcx platforms from the test because it required 1ms accuracy.

fixes: #66185
commit 5e54a640 (2025-04-04): tests: kernel/sleep: exclude npcx platforms from the test

According the specification, in extreme cases, the deviation of the APB
clock and LFCLK clock can reach up to +/-1% (+/- 10ms).
Therefore, exclude npcx platforms from the test because it required 1ms
accuracy.

fixes: #66185
PR #89001 (2025-04-30): tests: kernel/sleep: exclude npcx platforms from the test
Pr https://github.com/zephyrproject-rtos/zephyr/pull/87878 fixed only the `kernel.common.timing`. The same fix should also be applied to `kernel.common.timing.minimallibc`.

fixes: #66185
commit 515c9813 (2025-04-30): tests: kernel/sleep: exclude npcx platforms from the test

According the specification, in extreme cases, the deviation of the APB
clock and LFCLK clock can reach up to +/-1% (+/- 10ms).
Therefore, exclude npcx platforms from the test because it required 1ms
accuracy.

fixes: #66185
closed 2025-04-04 by kartben