zephyrproject-rtos/zephyr #66189: building sample  net/wifi fails to compile

rabarar, 2023-12-05:
**Describe the bug**

issuing the build command for `xiao_esp32c3` for the `samples/net/wifi` fails to compile - although it build fine for `xiao_esp32s3`

```
In file included from /Users/robert/src/west/xiaoC32/demo/zephyr/include/zephyr/net/net_if.h:29,
                 from /Users/robert/src/west/xiaoC32/demo/zephyr/include/zephyr/net/net_pkt.h:30,
                 from /Users/robert/src/west/xiaoC32/demo/zephyr/include/zephyr/net/ethernet.h:22,
                 from /Users/robert/src/west/xiaoC32/demo/zephyr/drivers/wifi/esp32/src/esp_wifi_drv.c:14:
/Users/robert/src/west/xiaoC32/demo/zephyr/include/zephyr/net/net_l2.h:83:32: error: '_net_l2_ETHERNET_L2' undeclared here (not in a function)
   83 | #define NET_L2_GET_NAME(_name) _net_l2_##_name
      |                                ^~~~~~~~
/Users/robert/src/west/xiaoC32/demo/zephyr/include/zephyr/net/net_if.h:2831:25: note: in expansion of macro 'NET_L2_GET_NAME'
 2831 |                 .l2 = &(NET_L2_GET_NAME(_l2)),                          \
      |                         ^~~~~~~~~~~~~~~
/Users/robert/src/west/xiaoC32/demo/zephyr/include/zephyr/net/net_if.h:2875:9: note: in expansion of macro 'NET_IF_INIT'
 2875 |         NET_IF_INIT(dev_id, 0, l2, mtu, NET_IF_MAX_CONFIGS)
      |         ^~~~~~~~~~~
/Users/robert/src/west/xiaoC32/demo/zephyr/include/zephyr/net/net_if.h:2921:9: note: in expansion of macro 'Z_NET_DEVICE_INIT'
 2921 |         Z_NET_DEVICE_INIT(node_id, Z_DEVICE_DT_DEV_ID(node_id), \
      |         ^~~~~~~~~~~~~~~~~
/Users/robert/src/west/xiaoC32/demo/zephyr/include/zephyr/net/net_if.h:2934:9: note: in expansion of macro 'NET_DEVICE_DT_DEFINE'
 2934 |         NET_DEVICE_DT_DEFINE(DT_DRV_INST(inst), __VA_ARGS__)
      |         ^~~~~~~~~~~~~~~~~~~~
/Users/robert/src/west/xiaoC32/demo/zephyr/drivers/wifi/esp32/src/esp_wifi_drv.c:653:1: note: in expansion of macro 'NET_DEVICE_DT_INST_DEFINE'
  653 | NET_DEVICE_DT_INST_DEFINE(0,
      | ^~~~~~~~~~~~~~~~~~~~~~~~~

```
**Environment**
West version: v1.1.0

* Page: `boards/riscv/xiao_esp32c3/doc/index`
* Version: main
* SHA-1: 7bc822df041b4a7ca316640f3f9b589a08e692c6

kartben, 2023-12-05:
That was fast :) -- and excellent, thanks in advance!

rabarar, 2023-12-05:
Pleasure! PR submitted - thank you for the quick reply!  I just got the boards in this afternoon, so I haven’t had time to play just yet!

rabarar, 2023-12-05:
Can you assist me on the “check-warns” failure on the PR? Let me know what needs to be done to pass this and I’ll update the PR…

kartben, 2023-12-05:
You need to fix the issues listed at https://github.com/zephyrproject-rtos/zephyr/actions/runs/7105827573/job/19343990483#step:9:1
So basically amending your commit message to follow the guidelines from https://docs.zephyrproject.org/latest/contribute/guidelines.html#commit-message-guidelines, making sure you have a proper sign-off entry that matches your "author" email address, and then force pushing the amended commit to your branch. Thanks!