zephyrproject-rtos/zephyr #66200: gPTP(RMII mode):  <err>eth_mcux: ENET_GetRxFrameSize return: 4001

YMont, 2023-12-06:
**Describe the bug**
System version is Zephyr3.5.0. 
SoC is NXP MIMXRT1061CVL5B.
Protocol is RMII.

I build, compile, run SDK samples/net/gptp, and generate binary file. I flash this binary by J-Link. Otherwise, when MCU initiate packet to FPGA. And FPGA totally bypass packet to MCU(RT1061), but I found RX packet within some error about ENET_GetRxFrameSize return:4001.

I use external clock from FPGA which frequency is 50 MHz, so I think I have to change MCU ENET_REF_CLK pin setting from output to input.

I trace #51107 and #60144, but I'm not sure my issue is same with them. How can I do fix this bug?

**As the following is my setting and configuration:**
1.`zephyr\boards\arm\mimxrt1060_evk\mimxrt1060_evk-pinctrl.dtsi`
```
@@ -55,7 +55,7 @@
                group0 {
                        pinmux = <&iomuxc_gpio_b1_10_enet_ref_clk>;
                        bias-disable;
-                       drive-strength = "r0-6";
+                       drive-strength = "disabled";
                        slew-rate = "fast";
                        nxp,speed = "50-mhz";
                        input-enable;
@@ -77,8 +77,8 @@
                                <&iomuxc_gpio_b1_08_enet_tx_data1>,
                                <&iomuxc_gpio_b1_09_enet_tx_en>,
                                <&iomuxc_gpio_b1_11_enet_rx_er>,
-                               <&iomuxc_gpio_emc_40_enet_mdc>,
-                               <&iomuxc_gpio_emc_41_enet_mdio>;
+                               <&iomuxc_gpio_b1_14_enet_mdc>,
+                               <&iomuxc_gpio_b1_15_enet_mdio>;
                        drive-strength = "r0-5";
                        bias-pull-up;
                        bias-pull-up-value = "100k";
``` 
2.`zephyr\samples\net\gptp\boards\mimxrt1060_evk.conf`
```
-CONFIG_NET_GPTP_INIT_LOG_PDELAY_REQ_ITV=-3
+CONFIG_NET_GPTP_INIT_LOG_PDELAY_REQ_ITV=3
+CONFIG_NET_GPTP_LOG_LEVEL_DBG=y
```

**Error log:**
```
uart:~$
JD[00ELAY_REQ TX
:00:00.501,000] <dbg> net_gptp: gptp_thread: (): Starting PTP thread
[00:00:00.501,000] <dbg> net_gptp: gptp_change_port_state_debug: (): [1] state INITIALIZING -> DISABLED (gptp_thread():587)
*** Booting Zephyr OS build zephyr-v3.5.0-1898-g560d06b9e39c ***
[00:00:00.501,000] <inf> net_config: Initializing network
[00:00:00.501,000] <inf> net_config: Waiting interface 1 (0x800012d0) to be up...
[00:00:00.502,000] <dbg> net_gptp: gptp_change_pa_info_state_debug: (gptp): [1] PA info state DISABLED -> POST_DISABLED (gptp_mi_port_announce_information_state_machine():1437)
[00:00:00.502,000] <dbg> net_gptp: gptp_change_port_state_debug: (gptp): [0] state INITIALIZING -> SLAVE (gptp_updt_roles_tree():1868)
[00:00:00.504,000] <dbg> net_gptp: gptp_send_pdelay_req: (gptp): Sending PDELAY_REQ seq 42260 pkt 0x8000e910
[00:00:00.529,000] <wrn> net_if: iface 0x800012d0 is down
[00:00:01.502,000] <inf> net_config: Interface 1 (0x800012d0) coming up
[00:00:01.502,000] <err> eth_mcux: ENET_GetRxFrameSize return: 4001
[00:00:01.503,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:01.503,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:01.503,000] <err> eth_mcux: ENET_GetRxFrameSize return: 4001
uart:H~$ello World!
[00:00:01.603,000] <inf> net_config: IPv6 address: 2001:db8::1
uart:~$ PDELAY_REQ TX
[00:00:08.531,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:08.533,000] <dbg> net_gptp: gptp_send_pdelay_req: (gptp): Sending PDELAY_REQ seq 42261 pkt 0x8000e910
[00:00:08.559,000] <err> eth_mcux: ENET_GetRxFrameSize return: 4001
uart:~$ PDELAY_REQ TX
[00:00:16.560,000] <wrn> net_gptp: Reset Pdelay requests
[00:00:16.562,000] <dbg> net_gptp: gptp_send_pdelay_req: (gptp): Sending PDELAY_REQ seq 42262 pkt 0x8000e910
[00:00:16.587,000] <err> eth_mcux: ENET_GetRxFrameSize return: 4001

```

pdgendt, 2023-12-06:
Hi @YMont, is there an issue besides the error print? We see these errors too if there is a lot of network traffic, so it could be some overrun.

YMont, 2023-12-08:
Hi @pdgendt, 
I trace code, as the following:
zephyr\drivers\ethernet\eth_mcux.c --> ENET_GetRxFrameSize()
modules\hal\nxp\mcux\mcux-sdk\drivers\enet\fsl_enet.c --> `Line 1539: if (0U != (curBuffDescrip->control & ENET_BUFFDESCRIPTOR_RX_EMPTY_MASK))`

Then, I found the relate with this error `ENET_BUFFDESCRIPTOR_RX_ERR_MASK`. And I print error code.
```
ENET_GetRxFrameSize, 1549 curBuffDescrip->control 0x844
ENET_GetRxFrameSize, 1567 curBuffDescrip->control 0x844
``` 
How could I fix   CRC error `ENET_BUFFDESCRIPTOR_RX_CRC_MASK`?

Thanks

pdgendt, 2023-12-08:
Not sure, maybe @decsny or @danieldegrasse can help out.

In the IMXRT1064RM document for **41.7.14.1 Enhanced receive buffer descriptor**, I find **Table 41-36. Receive buffer descriptor field definitions**:
[image]

YMont, 2023-12-08:
Hi pdgendt, thanks a lot. Execute me, my document number is `IMXRT1060RM Rev. 1, 12/2018` that is too old?
Is that means should I define it `ENET_ENHANCEDBUFFERDESCRIPTOR_MODE` ?

Otherwise, And I bypass some error, then I found it can receive RX data from TX.
[image]

YMont, 2023-12-11:
Hi @pdgendt, @decsny, @danieldegrasse,

I think CR field relate with L field by RT1060RM file. Then, how could I check uDMA table value ?
[image]

Is this CRC error cause from MCU or Zephyr OS or lack of some configuration? How could I converge on this problem advance?

Thanks.

YMont, 2023-12-28:
Hi @DerekSnell ,
Apologies for the delay. I already refer to #60073. Then I enable `CONFIG_ETH_MCUX_RMII_EXT_CLK=y` and modify these configuration `IOMUXC_EnableMode(IOMUXC_GPR, kIOMUXC_GPR_ENET1RefClkMode, true);` , `IOMUXC_EnableMode(IOMUXC_GPR, kIOMUXC_GPR_ENET1TxClkOutputDir, false);`.

It seems still receive the return error `4001`. I'm curious that what kind of situation or condition MCU will trigger `ENET_BUFFDESCRIPTOR_RX_CRC_MASK` tag?
Thanks.

DerekSnell, 2024-01-03:
Hi @YMont ,
I know that you are also getting support through NXP's private support option.  And I have seen that the NXP support engineers received your RMII bus captures, and are reviewing your latest data.  Let's see what they suggest to resolve this RMII issue.  Best regards

closed 2024-01-04 by YMont

DerekSnell, 2024-01-15:
Hi @YMont ,
I saw that the NXP support team helped resolve this issue, and I wanted to share the solution here for others.
> this issue has been solved by configuring RX_ER pin to normal GPIO with pull-down resistor enabled.
Previously, the RX_ER pin was floating, and the pinctrl initialized this pin.
The CR bit was getting set because this flag also indicates a PHY error.