zephyrproject-rtos/zephyr #66206: Enable

benner, 2023-12-06:
**Is your enhancement proposal related to a problem? Please describe.**

During development I noticed `read-only` property in `partitions` (`compatible = "fixed-partitions"`) does not do anything. Some talk in Discord revealed it's just copied from Linux DT.

**Describe the solution you'd like**

Disable writes for this region of storage (if hardware allows it) or check arguments of related write/erase functions and disallow this in software.

**Describe alternatives you've considered**

Remove this option from DT  until proper implementation will be implemented (will avoid some confusion and digging).

**Additional context**
N/A

de-nordic, 2023-12-06:
Read-only for who?
Considering DTS as description of hardware you have, something you attach to you technical specification with pcb printout, who is not allowed to write to such partition?
If you have MCUboot and and App running on a give DTS defined hardware, which one is allowed and which one is not allowed to write to partition?
Lets say that for security you set MCUboot as read-only, that is ok because nothing except JTAG should be able to write that.
Now, app should not be able to overwrite itself, right? Lets set to read-only the partition the app resides on. That is the same DTS. Now how does the MCUboot swaps images? Can it override the read-only, with special function?
So now everyone has the special function to override the read-only. Does read-only flag work anymore?

benner, 2023-12-06:
@de-nordic, good questions. I had similar when I was trying to evaluate this (until discovered it's does not not work). Initial Idea (at least for MCUBoot) was to have overlay without read-only.

Laczen, 2023-12-07:
@benner, the read-only property when inserted would only apply to the "users" of fixed partitions (the flash driver is not a user, only the flash_map api is a user). This read-only property would therefore not apply to direct flash writes.
@de-nordic, it would be possible to add the read-only property in a overlay file.

de-nordic, 2023-12-07:
I would rather go with property that would state which other partitions could access the one, for example "write-allowed-by: slot1, mcuboot", or something like this.
DTS defines a product, you start adding overlays you multiply products.
MCUboot with one overlay is build for different device than app without that one.
And I know that MCUboot already uses specific overlay to even work with Zephyr, which annoys me anyway.