zephyrproject-rtos/zephyr #66223: [frdm_kw41z] hello world test failure

hakehuang, 2023-12-06:
**Describe the bug**
hello world test failure.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 FRDM_KW41Z
 - What have you tried to diagnose or workaround this issue?
 NO
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 ```
ubuntu@ubuntu-OptiPlex-7050:/home/shared/disk/zephyr_project/zephyr_test/zephyr$ git bisect bad
f0daf904bb0202c9247cdcf4d1186e0762a9db67 is the first bad commit
commit f0daf904bb0202c9247cdcf4d1186e0762a9db67
Author: Keith Packard <keithp@keithp.com>
Date:   Tue Jan 10 21:17:32 2023 -0800

    libc: Default to picolibc where supported
    
    This switches the default C library to picolibc for all targets which
    support it.
    
    Signed-off-by: Keith Packard <keithp@keithp.com>

 lib/libc/Kconfig | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

```

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/hello_world
2. west build -b frdm_kw41z
3. west flash
4. no console output.

**Expected behavior**
allication run OK

**Impact**
showstopper

**Logs and console output**
NA

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v3.5.0-2687-g3879b0255

dleach02, 2023-12-06:
duplicate #66154

hakehuang, 2023-12-27:
fixing https://github.com/zephyrproject-rtos/zephyr/pull/67022