zephyrproject-rtos/zephyr #66236: build: Incorrect python interpret found by python.cmake on Nix

teburd, 2023-12-06:
**Describe the bug**
I'd expect python.cmake to choose the first python in my path, it instead seems to choose somewhat at random a python interpreter on my path.

I use nix and archlinux and get similar outcomes when I have multiple pythons installed. When multiple pythons are installed my path, and the python that comes up from the command `which python` should in fact the first option chosen.

find_package(Python COMPONENT INTERPRETER) seems to do this just fine, the current find_program() usage does not.

I've added some message() outputs in python.cmake and west.cmake to try and provide some clues, diff attached

output from this...

```sh
(env) tburdick@zefire ~/z/z/doc (main)> make html-fast
make html DT_TURBO_MODE=1
make[1]: Entering directory '/home/tburdick/z/zephyr/doc'
cmake \
	-GNinja \
	-B_build \
	-S. \
	-DDOC_TAG=development \
	-DSPHINXOPTS="-j auto -W --keep-going -T" \
	-DSPHINXOPTS_EXTRA="" \
	-DLATEXMKOPTS="-halt-on-error -no-shell-escape" \
	-DDT_TURBO_MODE=1
Loading Zephyr module(s) (Zephyr base): doc
Python currently , 
python found /nix/store/d9h22m8c7irvn866jmc6ilkk8j460bxg-python3-3.10.13-env/bin/python
python found /nix/store/d9h22m8c7irvn866jmc6ilkk8j460bxg-python3-3.10.13-env/bin/python version 3.10
python3 found /nix/store/d9h22m8c7irvn866jmc6ilkk8j460bxg-python3-3.10.13-env/bin/python
python3 found /nix/store/d9h22m8c7irvn866jmc6ilkk8j460bxg-python3-3.10.13-env/bin/python version 3.10
Calling find_package Python3, /nix/store/d9h22m8c7irvn866jmc6ilkk8j460bxg-python3-3.10.13-env/bin/python
-- Found Python3: /nix/store/d9h22m8c7irvn866jmc6ilkk8j460bxg-python3-3.10.13-env/bin/python (found suitable version "3.10.13", minimum required is "3.8") found components: Interpreter 
Called find_package Python3, /nix/store/d9h22m8c7irvn866jmc6ilkk8j460bxg-python3-3.10.13-env/bin/python
-- Cache files will be written to: /home/tburdick/.cache/zephyr
-- Zephyr base: /home/tburdick/z/zephyr
-- Found Doxygen: /nix/store/xknvdvpqg40cnd10gbyr3q29imgsra4k-doxygen-1.9.6/bin/doxygen (found version "1.9.6") found components: doxygen dot 
CMake Error at CMakeLists.txt:33 (message):
  The 'sphinx-build' command was not found

-- Configuring incomplete, errors occurred!
See also "/home/tburdick/z/zephyr/doc/_build/CMakeFiles/CMakeOutput.log".
make[1]: *** [Makefile:24: configure] Error 1
make[1]: Leaving directory '/home/tburdick/z/zephyr/doc'
make: *** [Makefile:18: html-fast] Error 2
(env) tburdick@zefire ~/z/z/doc (main) [2]> which python
/home/tburdick/z/env/bin/python
(env) tburdick@zefire ~/z/z/doc (main)> ls -alh /home/tburdick/z/env/bin/python
lrwxrwxrwx 1 tburdick users 74 Dec  1 13:03 /home/tburdick/z/env/bin/python -> /nix/store/c3cjxhn73xa5s8fm79w95d0879bijp04-python3-3.10.13/bin/python3.10
(env) tburdick@zefire ~/z/z/doc (main)> ls -alh /nix/store/c3cjxhn73xa5s8fm79w95d0879bijp04-python3-3.10.13/bin/python3.10
-r-xr-xr-x 1 root root 16K Dec 31  1969 /nix/store/c3cjxhn73xa5s8fm79w95d0879bijp04-python3-3.10.13/bin/python3.10
```

I'd note that which python points at my virtual environment, which itself holds a static link to a nix install of python with various python packages (nix will generate a virtualenv like wrapper itself), and that links to a real python interpreter. None of those are actually selected though. Instead it appears that, at perhaps random or some undefined ordering, another python in my PATH var is selected. That's seemingly, to me, the wrong behavior and very unexpected.

If I replace all of python.cmake with a simple find_package(Python COMPONENT INTERPRETER) as seen in #65995 the symlinked python in my virtualenv is found correctly.

Some other useful environment variables perhaps give some clues

nix will set PATH, CMAKE_, and other environment variables when being used to enable builds with its unique filesystem layout.

env | grep python3-3.10.13 shows... (sorry for the large text blob here)

```sh
XDG_DATA_DIRS=/nix/store/d9h22m8c7irvn866jmc6ilkk8j460bxg-python3-3.10.13-env/share:/nix/store/n7nrikml13ipr3gishnbf2x0g3zbia16-minicom-2.8/share:/nix/store/04p9vxaplyd4a1m4q8qyh371dfmny3pl-ninja-1.11.1/share:/nix/store/sj207xd8h848inbwqpqz9lq89yii08sj-gperf-3.1/share:/nix/store/h8q4r7f1j906rlffj96dlbbmwc9zbirk-cmake-3.25.3/share:/nix/store/lqwlz76ns9c46qj9lcihb6vzxp3gmy0q-gnumake-4.4.1/share:/nix/store/ld4wra61gxq0spf16p0a6z3pcklagg6f-gdb-13.1/share:/nix/store/047yg0alvlwlx2k7nf3qilq3hp0rb7z4-segger-jlink-766/share:/nix/store/89af5sqf1rcjd7x149mwbcbzvsr8wcny-nrf-command-line-tools-10.16.0/share:/nix/store/xbm2s7589mswrgngv8ci1zcqfh1l6v2w-xxd-vim-9.0.1441/share:/nix/store/pq8mdg40n1ggphcicx3ymvbxq56vkyx7-graphviz-7.1.0/share:/nix/store/gw2ql840xwh37fp7dv430mrxc1bgyxsz-lldb-16.0.6/share:/nix/store/wnblw3rp0qr0z6wyzhaahch3mkk42gmi-ncurses-6.4/share:/nix/store/8mw6ssjspf8k1ija88cfldmxlbarl1bb-zlib-1.2.13/share:/nix/store/b8gwvz4hiq9227321s3lqbk3c6pv22kz-llvm-16.0.6/share:/nix/store/k6ssc7alckmv6gbympghw365ribdlj31-rustup-1.26.0/share:/nix/store/gwf21xyl0x0bhgapi66avzf6qmddr44s-patchelf-0.15.0/share:/nix/store/1wbc95psbpyzlc8s20qpmbic1706dfhk-gnome-console-45.0/share:/nix/store/n1qj6nna92xldmqxdqfwj8b4naxiznaj-gsettings-desktop-schemas-45.0/share/gsettings-schemas/gsettings-desktop-schemas-45.0:/nix/store/k11r1l1q236cjim5piz6gp8xs1y7ir2k-gtk4-4.12.3/share/gsettings-schemas/gtk4-4.12.3:/nix/store/1wbc95psbpyzlc8s20qpmbic1706dfhk-gnome-console-45.0/share/gsettings-schemas/gnome-console-45.0:/nix/store/cv38v8k6nf1ng6cr9mjfqq1as3vnb43w-gnome-mimeapps/share:/nix/store/cib1rhpk4653hbf8hqwlw9hanf11yhig-desktops/share:/home/tburdick/.nix-profile/share:/nix/profile/share:/home/tburdick/.local/state/nix/profile/share:/etc/profiles/per-user/tburdick/share:/nix/var/nix/profiles/default/share:/run/current-system/sw/share:/nix/store/ja4xq4ra1w9j4g6cnvkkmmhjyh1qxpbh-gnome-shell-45.1/share/gsettings-schemas/gnome-shell-45.1:/nix/store/78yrfskcsbds6876kgwjdja2ws3d2p1d-gnome-shell-extensions-45.1/share/gsettings-schemas/gnome-shell-extensions-45.1:/nix/store/gl3sw9c9hkhkn5mqi7s2aj1y30fy8h2i-gnome-session-45.0/share:/nix/store/n1qj6nna92xldmqxdqfwj8b4naxiznaj-gsettings-desktop-schemas-45.0/share/gsettings-schemas/gsettings-desktop-schemas-45.0:/nix/store/1jkmjwd4d5c9x520kw2wzpw09bfcldcq-gtk+3-3.24.38/share/gsettings-schemas/gtk+3-3.24.38:/nix/store/bf6d3mj74kmffq97fliq4084fvcr5hq0-gnome-settings-daemon-45.0/share/gsettings-schemas/gnome-settings-daemon-45.0:/nix/store/gl3sw9c9hkhkn5mqi7s2aj1y30fy8h2i-gnome-session-45.0/share/gsettings-schemas/gnome-session-45.0:/nix/store/ja4xq4ra1w9j4g6cnvkkmmhjyh1qxpbh-gnome-shell-45.1/share
NIX_CFLAGS_COMPILE= -frandom-seed=nhccmaiznb -isystem /nix/store/d9h22m8c7irvn866jmc6ilkk8j460bxg-python3-3.10.13-env/include -isystem /nix/store/d9h22m8c7irvn866jmc6ilkk8j460bxg-python3-3.10.13-env/include -isystem /nix/store/d9h22m8c7irvn866jmc6ilkk8j460bxg-python3-3.10.13-env/include -isystem /nix/store/kzwn4zngw9h10k2msk3cm5n2yp07qz96-dtc-1.7.0/include -isystem /nix/store/kzwn4zngw9h10k2msk3cm5n2yp07qz96-dtc-1.7.0/include -isystem /nix/store/kzwn4zngw9h10k2msk3cm5n2yp07qz96-dtc-1.7.0/include -isystem /nix/store/lqwlz76ns9c46qj9lcihb6vzxp3gmy0q-gnumake-4.4.1/include -isystem /nix/store/lqwlz76ns9c46qj9lcihb6vzxp3gmy0q-gnumake-4.4.1/include -isystem /nix/store/lqwlz76ns9c46qj9lcihb6vzxp3gmy0q-gnumake-4.4.1/include -isystem /nix/store/ld4wra61gxq0spf16p0a6z3pcklagg6f-gdb-13.1/include -isystem /nix/store/ld4wra61gxq0spf16p0a6z3pcklagg6f-gdb-13.1/include -isystem /nix/store/ld4wra61gxq0spf16p0a6z3pcklagg6f-gdb-13.1/include -isystem /nix/store/89af5sqf1rcjd7x149mwbcbzvsr8wcny-nrf-command-line-tools-10.16.0/include -isystem /nix/store/89af5sqf1rcjd7x149mwbcbzvsr8wcny-nrf-command-line-tools-10.16.0/include -isystem /nix/store/89af5sqf1rcjd7x149mwbcbzvsr8wcny-nrf-command-line-tools-10.16.0/include -isystem /nix/store/pq8mdg40n1ggphcicx3ymvbxq56vkyx7-graphviz-7.1.0/include -isystem /nix/store/pq8mdg40n1ggphcicx3ymvbxq56vkyx7-graphviz-7.1.0/include -isystem /nix/store/pq8mdg40n1ggphcicx3ymvbxq56vkyx7-graphviz-7.1.0/include -isystem /nix/store/hq9gk0b60w877fbmqhq95xv5kxz3p3mp-protobuf-3.21.12/include -isystem /nix/store/hq9gk0b60w877fbmqhq95xv5kxz3p3mp-protobuf-3.21.12/include -isystem /nix/store/hq9gk0b60w877fbmqhq95xv5kxz3p3mp-protobuf-3.21.12/include -isystem /nix/store/xp3d0pv6rzsliy5sizxw3h5qj1c61ny4-glibc-multi-2.37-45-dev/include -isystem /nix/store/xp3d0pv6rzsliy5sizxw3h5qj1c61ny4-glibc-multi-2.37-45-dev/include -isystem /nix/store/xp3d0pv6rzsliy5sizxw3h5qj1c61ny4-glibc-multi-2.37-45-dev/include -isyste [... cut ...]
Avoiding all the cmake environment variables, ignoring all the cmake paths...

So we have a bug in our python.cmake that does not exist in cmake's own FindPython support package.

**To Reproduce**

I suppose try adding a bogus python install to CMAKE_PREFIX_PATH and watch everything fail oddly.

**Expected behavior**
The first python interpret in my PATH var is used.

**Impact**
Building docs is broken, building on nix is broken, building with multiple python interpreters is broken in CMAKE_ environment variables are set as virtualenv's are ignored.

tejlmand, 2023-12-11:
@teburd thanks a lot for this useful information.

I already now have some idea of what might be happening, but need to do some testing on this, but unfortunately not having time atm for further investigation.

Marked as `priority: low`, as this is impacting a very specific setup and a workaround does exist (setting `Python3_EXECUTABLE`)

teburd, 2023-12-11:
See https://gitlab.kitware.com/cmake/cmake/-/blob/master/Modules/FindPython/Support.cmake#L1786 to start, the very first branch for virtualenvs ignores the cmake env vars.

cmake's FindPython module works well for me, but I can understand if there's confusing things that happen when it goes looking around registries and such for the latest python install.

@tejlmand did you try setting the strategy for FindPython? https://gitlab.kitware.com/cmake/cmake/-/blob/master/Modules/FindPython.cmake#L314 seems like using the LOCATION strategy might be just fine?

tejlmand, 2023-12-11:
> @tejlmand did you try setting the strategy for FindPython? https://gitlab.kitware.com/cmake/cmake/-/blob/master/Modules/FindPython.cmake#L314 seems like using the LOCATION strategy might be just fine?

Unfortunately its not. Been there, tried that.

The location still doesn't guarantee the order in which Python installations are tried (those in path which https://github.com/zephyrproject-rtos/zephyr/pull/24364 fixes).

teburd, 2024-02-01:
Setting Python3_EXECUTABLE doesn't seem to be possible when running twister unless I've missed something (very possible!). I'm carrying around a patch to do my work now which is kind of painful. It seems like in retrospect all the issues here are related to windows and python, can we use the cmake python module on all platforms but windows perhaps?

teburd, 2024-02-01:
> @tejlmand Setting Python3_EXECUTABLE doesn't seem to be possible when running twister unless I've missed something (very possible!). I'm carrying around a patch to do my work now which is kind of painful. It seems like in retrospect all the issues here are related to windows and python, can we use the cmake python module on all platforms but windows perhaps?

Ok I figured it out with, hilariously... I do

```-x Python3_EXECUTABLE="`which python`"```

marc-hb, 2024-02-01:
> It seems like in retrospect all the issues here are related to windows and python, can we use the cmake python module on all platforms but windows perhaps?

I think the "bigger" question is: if the real-world issues are (as expected) very different between Windows and Linux, then why is 100% of the workaround code shared between the two?

Zephyr's C code is (correctly) split between "arch/" code for arch-specific problems versus common code for common problems. So why does `python.cmake`'s re-invention of `FindPython` insists on being 100% "portable" when (according to @teburd) most of the problems it tries to solve are not "portable" at all?

PS: we haven't even looked at macOS/brew yet...

teburd, 2024-03-15:
Related, and probably better described issue #70258

closed 2024-03-15 by teburd