zephyrproject-rtos/zephyr #66244: Setting C standard in prj.conf doc: Documentation issue in 'develop/languages/c/index'

chessami92, 2023-12-06:
**Describe the bug**

Is there a way to set the C standard similar to CONFIG_STD_CPP##? I see -std=c99 in the rsp on my build, but I'm not sure where it's coming from and if I can instead configure it to c11. 

**Environment**

* Page: `develop/languages/c/index`
* Version: main
* SHA-1: 22320db5f701451156228c09c9b8d0f7989367ab

aescolar, 2023-12-11:
> There does exists `set_property(GLOBAL PROPERTY CSTD <c-standard>) which can be called in the app's / board's `CMakeLists.txt` file. Ref:
> 
> https://github.com/zephyrproject-rtos/zephyr/blob/ffd716b4a21451e5989c2a10c1f2ade0b43e1c09/CMakeLists.txt#L931
> 
> Unsure why this is not documented (+ why is this done as a property). @aescolar do you recall why this has been made as a "secret" setting ? ref: #8749

I don't think it was intended as a "secret" setting.
Zephyr required for very long at least c99, and in that PR I had given the option to override it as some users were having trouble with less than c11 with host compilers as described in that PR comment.
I guess we could have been more "public" about it :). But this was very long ago, so I guess we were just not so careful with documentation or discoverability of build options at the time.

tejlmand, 2023-12-11:
> Zephyr required for very long at least c99, and in that PR I had given the option to override it as some users were having trouble with less than c11 with host compilers as described in that PR comment.

It sounds like it should be safe to document, but be specific that default is `c99` and that Zephyr itself is only tested and verified with `c99`, so that users know.

> I guess we could have been more "public" about it :). But this was very long ago, so I guess we were just not so careful with documentation or discoverability of build options at the time.

would not be the first time 😄 

Thanks for quick reply.