zephyrproject-rtos/zephyr #66265: net: Make net_buf pool as rodata

LingaoM, 2023-12-07:
## Introduction

Since most of field in the pool should not change at runtime, and same part of field will change at runtime, so make this struct as rodata, and move some part of variables field to runtime structure, this change will more safely than put this to ram.

### Problem description

Many fields should not change at all at runtime.

### Proposed change

The field `lock`, `uninit_count ` and `avail_count` will move to new structure call `struct net_buf_pool_rt_ctx`.
And any access this fields from `pool->lock` change to `pool->rt->lock` etc...

## Detailed RFC

```C
struct net_buf_pool {
	/* Network buffer pool runtime ctx */
	struct net_buf_pool_rt_ctx {
		/** LIFO to place the buffer into when free */
		struct k_lifo free;

		/* to prevent concurrent access/modifications */
		struct k_spinlock lock;

		/** Number of uninitialized buffers */
		uint16_t uninit_count;

#if defined(CONFIG_NET_BUF_POOL_USAGE)
		/** Amount of available buffers in the pool. */
		atomic_t avail_count;
#endif /* CONFIG_NET_BUF_POOL_USAGE */
	} * const rt;
```

rodrigopex, 2023-12-09:
@LingaoM, thanks for your RFC.

Could you please elaborate a little more about the change? Please complete the [RFC template](https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/003_rfc-proposal.md). The **Detailed RFC** section is incomplete. Essential aspects of that are not explained like: 

- What are the impacts of the change?
- What are the benefits of that?

Why not add `cons *const` (or similar) to all the fields that should not change at all at runtime?

carlescufi, 2023-12-12:
Architecture WG:

- There is some reluctance to accept this change given the size of the change vs the gains, which are very small
- @LingaoM could you please do the following:
   1. Document the memory savings for each net_buf_pool
   2. Respond to "An obvious question that can come up is: "Why not add cons *const (or similar) to all the fields that should not change at all at runtime?""
   3. Extend this issue so that it complies with the RFC template

cfriedt, 2023-12-12:
Many of our API structures are broken into `data` (rw) and `config` (ro) components. I can understand the reasoning here, because we typically want to avoid something like a stack overflow breaking function pointers.

I'm not sure if `net_buf_pool_rt_ctx` is as descriptive as it could be (I like the `data` and `config` conventions personally).

I guess the bigger question is that it would be a breaking API change.

closed 2024-07-05 by LingaoM