zephyrproject-rtos/zephyr #66268: Enhancement request and general queries in the GNSS subsystem

mayankmahajan-nxp, 2023-12-07:
@bjarki-trackunit @fabiobaltieri 
1. Why are we not parsing time and navigational data in **gnss_nmea0183_parse_gga**?
2. Is it necessary to enable both GGA and RMC messages on our gnss device?
3. Why data is published only when both GGA and RMC messages are received?

bjarki-trackunit, 2023-12-07:
1. Because we get it in RMC, no need to get it from both :)
2. Yes, although there is overlap, we need speed, bearing and date from RMC, and we need HDOP and number of satellites from GGA
3. Since we need data from both RMC and GGA to fulfill the `struct gnss_data`, we need to parse both before we publish. We also need to ensure the RMC and GGA are related to the same GNSS fix

@fabiobaltieri Hey, I figured it out! both RMC and GGA contain the same UTC timestamp, so we can use it to ensure they are related to the same fix rather than a timeout :) GSV contains its own index to tie it together (message number and total number of messages)

I will update the NMEA0183 match handlers to use the fix instead of a timeout :)

fabiobaltieri, 2023-12-07:
> @fabiobaltieri Hey, I figured it out! both RMC and GGA contain the same UTC timestamp, so we can use it to ensure they are related to the same fix rather than a timeout :) GSV contains its own index to tie it together (message number and total number of messages)

Neat!