zephyrproject-rtos/zephyr #66289: OSDP Samples Minor Bugs

Ryanjwoodward, 2023-12-07:
**Bug Description**
A clear and concise description of what the bug is.
When Building a project for the **nrf52840dk_nrf52840** board I encountered some minor bugs that caused the build to fail.
The first was a missing **'t'** in a variable declaration with type **uint8_t**. The second was a missing opening curly brace for a function.

**Platform & Context**
I am using:
- Zephyr Toolchain 0.16.4
- nrf SDK 2.4.99-dev2
- nrf52840dk_nrf52840 board

**Error Messages**
When building the **peripheral_device** sample I encounter these errors:
`/opt/nordic/ncs/v2.4.99-dev2/zephyr/subsys/mgmt/osdp/src/osdp_sc.c:28:18: error: unknown type name 'uin8_t'
   28 |         volatile uin8_t *p = mem;`
   
  and

`/opt/nordic/ncs/v2.4.99-dev2/zephyr/subsys/mgmt/osdp/src/osdp_phy.c:224:25: error: label 'out_of_space_error' used but not defined
  224 |                         goto out_of_space_error;` 
  
  
  **My Solutions**
  Investigation of the errors led me to these files:
- osdp_sc.c  
- osdp_phy.c

In **osdp_sc.c**, on line _28_  in the _osdp_memzero_ function:
`volatile uin8_t *p = mem;`. was changed to `volatile uint8_t *p = mem;`

In **osdp_phy.c**, on line _197_ in the _osdp_phy_packet_finalize_:
`(pkt->control & PKT_CONTROL_SCB) && pkt->data[1] >= SCS_15)` was changed to `(pkt->control & PKT_CONTROL_SCB) && pkt->data[1] >= SCS_15) {`

Following those minor additions to the files, the OSDP sample application "peripheral-device" built successfully

henrikbrixandersen, 2023-12-07:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

kartben, 2023-12-08:
PR #66258 will be fixing the exact same two issues you are pointing out. Ironically it was submitted just a few hours before you reported this issue :)

closed 2023-12-08 by fabiobaltieri